Koenkk/zigbee2mqtt #21946: MAC_CHANNEL_ACCESS_FAILURE (255) when connecting usb ssd to Rasoberry Pi4

pklim0v, 2024-03-25:
### What happened?

Sonoff Zigbee 3.0 USB Dongle Plus works well on Raspberry Pi4, but once I connect any usb ssd disk to Raspberry zigbee2mqtt stops sending anything via the dongle: I still can see the incoming data (e.g. temperature, humidity) but I can't send any outgoing data (switch lights, relays etc.) It doesn't matter wether the usd drive is inserted directy to Raspberry or via powered hub. WiFi is switched off all the time, but "tail -f /var/log/syslog" shows MAC_CHANNEL_ACCESS_FAILURE (255)

### What did you expect to happen?

I expected that connecting any USB SSD won't affect the zigbee2mqtt work

### How to reproduce it (minimal and precise)

Connect USB SSD to Raspberry PI with zigbee2mqtt configured

### Zigbee2MQTT version

1.36.0

### Adapter firmware version

20230507

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus

### Setup

plain on Pi4

chris-1243, 2024-03-27:
It is known ssd are generating tons of interferance. It may even kill your zigbee signal

```

MAC_CHANNEL_ACCESS_FAILURE: this happens when the wireless spectrum is too occupied. Mostly happens when a microwave is on or when there are WiFi networks on the same channel. See Reduce Wifi interference by changing the Zigbee channel how to fix this.
```

https://www.zigbee2mqtt.io/guide/faq/#common-error-codes

Is your dongle-P directly connected to your Pi ? You should have it connected via an usb cable.

https://www.zigbee2mqtt.io/advanced/zigbee/02_improve_network_range_and_stability.html#usb-based-adapter

pklim0v, 2024-04-01:
Thank you! 1.8m of shielded usb cable perfectly helped.

closed 2024-04-01 by pklim0v