Koenkk/zigbee2mqtt #21950: Memory Leak (1.36.0)

Outsidewall, 2024-03-26:
### What happened?

Over a few hours approx 12 in this case, memory usage goes from 
[image]
to
[image]

No error messages in logs

zigbee2mqtt will not operate any smart switches, if you attempt to turn them on, they do not, again no error message.

A restart of the container, aka docker compose down;docker compose up -d
will allow operation again.

### What did you expect to happen?

No memory leak, and continued operation of the ability to turn smart sockets on/off

### How to reproduce it (minimal and precise)

Run in docker container.

### Zigbee2MQTT version

1.36.0

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Dongle-E (EXSP v8) 6.10.3.0 build 297

### Setup

Linux Rocky 0 - docker 

### Debug log

[image]

Koenkk, 2024-03-26:
I suggest to try out the new [ember adapter](https://github.com/Koenkk/zigbee2mqtt/discussions/21462)

Outsidewall, 2024-03-27:
Thank you, I have had a read on that ember thread, and left a message.

Outsidewall, 2024-04-16:
Update, Have updated dongle-e to 7.4.2, and the zigbee2mqtt memory leak has gone.

[image]

this is what it looked before

[image]

theorlangur, 2024-04-20:
@Outsidewall , I seem to have a similar memory/stability issue with my dongle-e on 7.3 firmware with ezsp driver
How are your impressions after running 7.4 with ember since updating? Is the network stable, any glitches? Did you have to re-pair devices? How big is your network if you don't mind me asking)

Outsidewall, 2024-04-22:
Hello, I updated to 7.4.2 and use the ember driver setting, no repairing needed. Also loaded the latest version of z2m. All working well, no memory leak, and so far reliable. Network is about 130 devices.

closed 2024-12-30 by github-actions[bot]