Koenkk/zigbee2mqtt #21951: [Wrong device]: IKEA Trådfri Driver, new model detects as the old model

Mats-SE, 2024-03-26:
### Link

https://www.ikea.com/se/sv/p/tradfri-driver-foer-tradloes-styrning-smart-gra-50356187/

### Model

ICPSHC24-10EU-IL-2 (IKEA Trådfri Driver 10W)

### Description

Driver

### Vendor

IKEA

### Picture (link)

https://www.ikea.com/se/sv/images/products/tradfri-driver-foer-tradloes-styrning-smart-gra__0587550_pe672682_s5.jpg?f=xl

### Database entry

Where to find this?

### Notes

IKEA Trådfri Driver 10W and 30W has a new Modelnumber. The older version has at the end "IL-1", the new model has "IL-2". See pictures.

Zigbee2MQTT add the driver, but as the earlier model "ICPSHC24-10EU-IL-1". I have added both the new 10W and the new 30W driver to the IKEA Dirigera hub to verify the latest firmware, and for the old model "IL-1", the latest firmware is now 2.3.086, but for the new model "IL-2" the latest firmware are 1.0.002 so the firmware are not the same.

It would be great if Zigbee2MQTT are enable to detect the right model of the driver (ending with the "IL-2") for both the 10W and the 30W driver. Please tell me if you want any more information about this...

[image]

[image]

[image]

[image]

mrmaximas, 2024-03-26:
Please post database.db entry for this device.

Mats-SE, 2024-03-26:
Absolutely, but where can I found it?

mrmaximas, 2024-03-26:
> Absolutely, but where can I found it?

depend of you os in the z2m folder please find /data/ folder then database.db file, then find ieeee addr of this device.

staastas, 2024-03-26:
Hello. I also have the ICPSHC24-10EU-IL-2 device.
Here's my record from the database.
I'm not disclosing the full ieeeAddr.

`{"id":23,"type":"Router","ieeeAddr":"0x****************","nwkAddr":43521,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI Driver 10W","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,4096,64599],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"TRADFRI Driver 10W","manufacturerName":"IKEA of Sweden","powerSource":1,"zclVersion":8,"appVersion":16,"stackVersion":103,"hwVersion":1,"dateCode":"20181203","swBuildId":"1.0.002"}},"genOnOff":{"attributes":{"onOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":254}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":16,"stackVersion":103,"hwVersion":1,"dateCode":"20181203","swBuildId":"1.0.002","zclVersion":8,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1711491431017}`

Mats-SE, 2024-03-27:
Thanks staastas for that! I also have the 30W version. Unfortenately i still cannot find the database info for it. I don´t have an /data-folder under Zigbee2MQTT. I use HA OS. Where can I find the database.db file?

[image]

mrmaximas, 2024-03-27:
> Thanks staastas for that! I also have the 30W version. Unfortenately i still cannot find the database info for it. I don´t have an /data-folder under Zigbee2MQTT. I use HA OS. Where can I find the [...]

i'm not using HAOS but looks like database.db.backup is good too look inside.

mrskycriper, 2024-03-27:
> Thanks staastas for that! I also have the 30W version. Unfortenately i still cannot find the database info for it. I don´t have an /data-folder under Zigbee2MQTT. I use HA OS. Where can I find the database.db file?

In VSCode *.db files are hidden by default. Enable them in VSCode settings or open it in File editor

mrskycriper, 2024-03-27:
> Hello. I also have the ICPSHC24-10EU-IL-2 device. Here's my record from the database. I'm not disclosing the full ieeeAddr. Zigbee2MQTT 1.36.0-dev commit: 1f3d61d

It looks like newer drivers have `TRADFRI Driver 10W` and `TRADFRI Driver 30W`. While older one are named `TRADFRI transformer 10W` and `TRADFRI transformer 30W`. I'm currently working on IKEA refactoring and can fix those models for newer drivers

mrskycriper, 2024-03-29:
UPD

I've checked mine `ICPSHC24-30EU-IL-1` and it also reports `TRADFRI Driver 30W` Zigbee model.

So the model is exactly the same. I can mark it as a dual model for now.

If IKEA knows, how to separate them, then some form of fingerprinting is possible. But right now I don't see any other patterns here.

Mats-SE, 2024-03-31:
Infact it cannot be the exact same model as they have different firmware versions. Both the old and the new version are upgraded with the IKEA Dirigera HUB so I´m sure it has the correct, last firmware on both, but the firmware are different....

StSaens, 2024-05-09:
I have the old model (ICPSHC24-10EU-IL-1, 10W, with firmware 2.3.086) and after upgrading Z2M to 1.37.0 the state is no longer published in HA. 
The driver’s other properties are available in HA though.

Great-Chart, 2024-11-07:
Anyone with the IL-2 variant able to confirm if their units update (last seen) on a 60 second cycle as I've referenced in https://github.com/Koenkk/zigbee2mqtt/issues/24431

closed 2025-01-15 by github-actions[bot]