Koenkk/zigbee2mqtt #21955: SH-SC07 Scene switch sends nothing since 1.36.0

Lovasz-Akos, 2024-03-26:
### What happened?

After updating to latest, my scene switch returns no action, specifically "action": "", while linkquality gets updated on button press so there's definitely new data coming in.
[image]

### What did you expect to happen?

on pressing the button it should report action such as "single" or "double"

### How to reproduce it (minimal and precise)

seemingly SH-SC07 (TS0041), as I've seen other issues with the renamed 1_single bug reports but that's not what's happening here

### Zigbee2MQTT version

1.36.0

### Adapter firmware version

6.10.3.0 build 297

### Adapter

EZSP v8

### Setup

HomeAssistant add-on, on an x86 Odroid H3+

### Debug log

2024-03-26 17:29:44Received Zigbee message from 'Entrance Button', type 'raw', cluster 'genOnOff', data '{"data":[1,23,253,0],"type":"Buffer"}' from endpoint 1 with groupID 0

Koenkk, 2024-03-26:
Check https://www.zigbee2mqtt.io/guide/faq/#why-is-the-action-property-always-empty

Lovasz-Akos, 2024-03-26:
ah! so it is indeed the 1_single converting to single as the others. Thankfully I only have 2 automations associated with this switch so it's a quick fix. thanks!

closed 2024-03-26 by Lovasz-Akos