Koenkk/zigbee2mqtt #21960: Zen Thermostat No Ability to Set Emergency Mode

88fingerslukee, 2024-03-26:
### What happened?

Zen Thermostat works decently but I can't set it to Emergency Heat mode. For some reason the State of the device shows the system_mode as displaying "emergency_heating" when I set it from the thermostat but I can't read the system_mode number from the Dev console.

here is my state

`{
    "fan_mode": "auto",
    "fan_state": "ON",
    "linkquality": 76,
    "local_temperature": 20.8,
    "occupied_cooling_setpoint": 25,
    "occupied_heating_setpoint": 22,
    "running_mode": "heat",
    "system_mode": "emergency_heating",
    "update": {
        "installed_version": 528,
        "latest_version": 528,
        "state": "idle"
    },
    "voltage": 6000,
    "local_temperature_calibration": null,
    "update_available": null
}`

### Zigbee2MQTT version

 1.36.0 commit: 86ed71c

### Adapter firmware version

20230507

### Adapter

Sonoff Zdongle-E

### Setup

docker container

88fingerslukee, 2024-03-26:
Okay, so the systemmode should have another option (5) that should be exposed as Emergency heating or Aux heating.  Right now I can't set it because this mode isn't exposed.

88fingerslukee, 2024-03-30:
@Koenkk is there an easy way to add this functionality? I'm sorry, but I'm a little lost with the external config generation.

Koenkk, 2024-03-31:
Added it.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

88fingerslukee, 2024-04-05:
@Koenkk `emegency_heating` is not a valid HVAC mode per HA. Is there a way to make this a preset instead? 

Apparently custom presets are allowed as per the Climate developer guide.

closed 2024-11-03 by github-actions[bot]