Koenkk/zigbee2mqtt #21963: IKEA E2001/E2002 on actions being reported

roooodcastro, 2024-03-26:
### What happened?

I have 7 IKEA E2001/E2002 Styrbar remotes, all of them paired to Z2M, and 3 of them in the new 2.4.5 firmware version, the other 4 in the older 1.0.024 version. All of them managed to pair with Z2M correctly, and I have them all bound to lights around the house, which works 99% of the time without issues.

However, not a single one of them seem to have correctly paired, as the following things are missing completely:

- In the "Settings (specific)" tab, I don't have any options at all, not even the legacy toggle thing
- When I press any button in any way (single click, double click, long press), I only get some basic logs missing a ton of info
- There's no entity created in HA for the switch itself, just an `update` entity with update info, but no actions.
- When Z2M publishes the topic to MQTT, I can clearly see it's not publishing the action, so I don't have that info in HA

Also, for some reason the debug log lines are repeated exactly 5 times each time a button is pressed, with the exact same data except the timestamp changes a few milliseconds,

Lastly, after pairing, I only added binds to IKEA light bulbs, I have not removed any binds from the coordinator for these remotes.

### What did you expect to happen?

I expected the remotes to expose an extra entity to HA and send the `action` data so I can use that in automations.

### How to reproduce it (minimal and precise)

I just paired fresh IKEA E2001/E2002 remotes with Z2M following the instructions in [the docs page](https://www.zigbee2mqtt.io/devices/E2001_E2002.html).

### Zigbee2MQTT version

1.36.0-1

### Adapter firmware version

20221226

### Adapter

SONOFF Universal Zigbee 3.0 USB Dongle

### Setup

Run as add-on on Supervised HA in a 4gb RPi 4

### Debug log

```
Debug 2024-03-26 23:11:12Received Zigbee message from 'rod_office_lights_dimmer', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Debug 2024-03-26 23:11:12No converter available for 'E2001/E2002' with cluster 'genOnOff' and type 'commandOn' and data '{}'
Info 2024-03-26 23:11:12MQTT publish: topic 'zigbee2mqtt/rod_office_lights_dimmer', payload '{"battery":70,"last_seen":"2024-03-26T23:11:11.047Z","linkquality":102,"update":{"installed_version":65572,"latest_version":581,"state":"available"},"update_available":null}'
Debug 2024-03-26 23:11:12Received Zigbee message from 'rod_office_lights_dimmer', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Debug 2024-03-26 23:11:12No converter available for 'E2001/E2002' with cluster 'genOnOff' and type 'commandOn' and data '{}'
Info 2024-03-26 23:11:12MQTT publish: topic 'zigbee2mqtt/rod_office_lights_dimmer', payload '{"battery":70,"last_seen":"2024-03-26T23:11:11.071Z","linkquality":94,"update":{"installed_version":65572,"latest_version":581,"state":"available"},"update_available":null}'
Debug 2024-03-26 23:11:12Received Zigbee message from 'rod_office_lights_dimmer', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Debug 2024-03-26 23:11:12No converter available for 'E2001/E2002' with cluster 'genOnOff' and type 'commandOn' and data '{}'
Info 2024-03-26 23:11:12MQTT publish: topic 'zigbee2mqtt/rod_office_lights_dimmer', payload '{"battery":70,"last_seen":"2024-03-26T23:11:11.071Z","linkquality":94,"update":{"installed_version":65572,"latest_version":581,"state":"available"},"update_available":null}'
Debug 2024-03-26 23:11:12Received Zigbee message from 'rod_office_lights_dimmer', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Debug 2024-03-26 23:11:12No converter available for 'E2001/E2002' with cluster 'genOnOff' and type 'commandOn' and data '{}'
Info 2024-03-26 23:11:12MQTT publish: topic 'zigbee2mqtt/rod_office_lights_dimmer', payload '{"battery":70,"last_seen":"2024-03-26T23:11:11.071Z","linkquality":94,"update":{"installed_version":65572,"latest_version":581,"state":"available"},"update_available":null}'
Debug 2024-03-26 23:11:12Received Zigbee message from 'rod_office_lights_dimmer', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Debug 2024-03-26 23:11:12No converter available for 'E2001/E2002' with cluster 'genOnOff' and type 'commandOn' and data '{}'
Info 2024-03-26 23:11:12MQTT publish: topic 'zigbee2mqtt/rod_office_lights_dimmer', payload '{"battery":70,"last_seen":"2024-03-26T23:11:11.110Z","linkquality":94,"update":{"installed_version":65572,"latest_version":581,"state":"available"},"update_available":null}'
```

MrFr33man123, 2024-05-27:
I have a ConnBee II Stick and 2 E2002 Styrbar Remotes that have Updated OTA after Force Deleting them from Zigbee2MQTTT they could be paired again. But with no further Change now i gut suddenly other codes as Button Action.
Before it was BUTTON_LEFT BUTTON_RIGHT and so on. Now its brightness_stop, brightness_move_up, null, none and sometimes nothing at all until i press some other button. DEBUG shows nothing but status in the device. Also in Home Assistant i get some of the Click Action Events.

alienatedsec, 2024-05-28:
I had to delete and re-pair to make the right and left buttons work again.

jasonkaisersmith, 2024-06-07:
I have a very similar problem.

The device is connected OK and a strong linkquality.  The "power" doesn't show correctly either.
Pressing buttons does not trigger any messages to be post on the MQTT queue.

I have removed the batteries, reset, repaired multiple times and always the same issue.  Initial paring works, but not buttons presses are recorded.

Payload retained in MQTT broker.
{
  "action": null,
  "battery": null,
  "last_seen": "2024-06-07T14:04:20+02:00",
  "linkquality": 120,
  "update": {
    "installed_version": -1,
    "latest_version": -1,
    "state": null
  },
  "update_available": null
}

thomasaarholt, 2024-12-16:
I have the same issue with all four of my STYRBAR remotes, after migrating from ZHA to Z2M. They worked fine in ZHA. With Z2M I get no actions available.

I am not able to update the devices either (I just get "Device didn't respond to OTA request"). These are the current versions I have:
[image]

alienatedsec, 2024-12-16:
> I am not able to update the devices either (I just get "Device didn't respond to OTA request")

@thomasaarholt keep pressing the Styrbar buttons before pressing Update in Z2M

thomasaarholt, 2024-12-16:
Wow, this actually worked 🤩 I kept pressing the buttons before and then a bit into the update until it was running. 53 minutes remaining of the update 😄

This seems to be working for the remotes at version 1.0.024. Still getting the same update error for the 2.4.5 ones. 

I’ll report back tomorrow as to how the updated remotes work!

thomasaarholt, 2024-12-17:
Okay, I was able to update all remotes! The `1.0.024` ones first updated to `2.4.5`, then I updated again to `2.4.16`.
[image]

Unfortunately, I still have no actions being reported, so the remotes are unusable in Z2M.

I have attempted to [bind the correct actions as per this comment](https://github.com/Koenkk/zigbee2mqtt/issues/18515#issuecomment-2127507461) without success. There i get the followed error:

```
warning 2024-12-17 17:23:41z2m: Failed to setup reporting for 'Coordinator/1' cluster 'genOnOff' (ZCL command 0x287681fffeefcf3e/1 genOnOff.configReport([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":0,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":210},"zclSequence":199,"commandIdentifier":7} timed out after 10000ms))
```

thomasaarholt, 2024-12-17:
@alienatedsec Do you mean "force remove"? I ask since I did regular removing (e.g. trigger pairing again) quite a few times without success. Note that I do not have _any_ actions available currently, not even regular on and off, which I think is different from what your case was.

kribjo, 2025-01-03:
This seems related: https://github.com/Koenkk/zigbee2mqtt/issues/18515#issuecomment-2569749321

RayBan91, 2025-01-04:
I have this same issue after migration from 1.42.0 to 2.0.0 based on preparing section from https://github.com/Koenkk/zigbee2mqtt/discussions/24198. Any action on Z2M is present on Z2M layer but on HA layer I can't see any reaction in logbook.
[image]
[image]
[image]
Any reapiring procedure doesn't change anything: https://www.zigbee2mqtt.io/devices/E2001_E2002.html#ikea-e2001%252Fe2002
 All of my over 70 other devices from Z2M works great.

emmet02, 2025-01-05:
Experiencing exactly the same problem as above. Can see the ACTION updating fine in the UI for Zigbee2MQTT however no "action" entity or similar on the entity in HomeAssistant.

_(following unplanned overnight --watchtower-- upgrade to zigbee2mqtt v2 )_

alienatedsec, 2025-01-05:
> homeassistant:
>   legacy_action_sensor: true

The above is not needed. I recently had to fix it with v2.0.0 using this https://github.com/Koenkk/zigbee2mqtt/issues/25505#issuecomment-2571335350

kribjo, 2025-01-06:
> This seems related: [#18515 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/18515#issuecomment-2569749321)

For those of us with issues on Z2M 1.x - could a way forward on this be upgrading to 2.x (and throwing away old action-based blueprints!), since the "new way" seems to be device triggers or events rather than actions?

I haven't dived into the reasoning behind changes in 2.x, but could the problems with actions that we see here be part of why Z2M is moving away from them? 

Or should I rather focus on fixing problems with my devices, rather than expecting a solution from Z2M 2.x?

mrblondeblonde, 2025-01-07:
I have the same issue. No actions are available anymore. not only for the IKEA remote but also for Moes Smart knob.

mrblondeblonde, 2025-01-07:
> > I have the same issue. No actions are available anymore. not only for the IKEA remote but also for Moes Smart knob.
> 
> Open the Zigbee2MQTT add on - and go to the Settings -> Home Assistant integration -> enable Home Assistant legacy action sensors <img alt="image" width="819" src="https://private-user-images.githubusercontent.com/10803258/400832731-a8d34708-1623-4e72-b4df-3d4e095a5c1f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYyNzQ1MTMsIm5iZiI6MTczNjI3NDIxMywicGF0aCI6Ii8xMDgwMzI1OC80MDA4MzI3MzEtYThkMzQ3MDgtMTYyMy00ZTcyLWI0ZGYtM2Q0ZTA5NWE1YzFmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMDclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTA3VDE4MjMzM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWMzNDMwNDM5YzE2OWE0NDkyMWUzOGVlNDE2NTYxODNlZmZhYzg1ZjI2YTNhNTY1ZGVkNjEyMjJmZDI3MTg0ZmYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.zTZ9jVVfSNq8XeCr4f4RXOATFkb9OimLuIsT7Nqzeg8">

thanks a lot! but this means they eventualy will be removed?

ProsjektX, 2025-01-16:
Unable to get events at all from this remote. It is successfully paired (Z2M 2.0). Also managed to update the firmware to the latest, 2.4.16, but the same. No events show in the log when I press any buttons. Any idea if this can be solved? Reset and re-paired a couple of times, no difference.

kribjo, 2025-01-16:
> This seems related: [#18515 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/18515#issuecomment-2569749321)

OK, I fixed this (for me), and this seems to be the best place to comment on it.

Short recap: This was on Z2M 1.42.0. I updated firmware on my two STYRBAR E2001 from 1.0.024 to 2.4.16 (using the external converter trick). Afterwards they "stopped working" and though they were added to Z2M nothing showed up in the log or on the "Exposes" tab.

Then I tried a trillion things from issue comments, but nothing helped.

Then I tried one more (scary) thing, I updated Z2M to 2.0.0-2. And everything was broken... Z2M wouldn't even start, complaining about "Adapter EZSP protocol version (9) does not match Host [13-16]" (or something VERY similar). 
I was stuck for a while, but I gathered that I should update my SkyConnect / ZBT-1 firmware to fix this. I did this using Home Assistant add-on "Silicon Labs Flasher". 
(I would like to say that the firmware update worked without any issues, but I got stuck there too for a while. It was easily fixed though when I understood that I had to go into the configuration tab of the addon and set Device to my ZBT-1 rather than some default setting pointing to nothing. After that the update had no problems.)
After doing this (probably some restarting too, I don't remember for certain) Z2M started fine and I enabled "Home Assistant legacy action sensors" in "Home Assistant integration" settings before adding my STYRBAR E2001s again, and ... they work great! One of them is still not reporting it's battery level properly, but otherwise it seems all very fine now! All other Z2M devices (that I hadn't removed) just kept working like before.

So in this storm of bashing the 2.0.0 Z2M update for causing all kinds of troubles, I want to say that it was the final thing that actually fixed my STYRBARs after having tried everything else. I really held off with updating Z2M cause I was scared of breaking things with 2.0.0. And I did break things, but I fixed them again and they are better than before now.

ProsjektX, 2025-01-18:
> Unable to get events at all from this remote. It is successfully paired (Z2M 2.0). Also managed to update the firmware to the latest, 2.4.16, but the same. No events show in the log when I press any buttons. Any idea if this can be solved? Reset and re-paired a couple of times, no difference.

Ok, to recap for anyone else who runs into this. What was required for me was to update the firmware of the coordinator. I have a Sonoff P 3.0, which came with firmware version 20210708 (go to Z2M under settings->about to check). For simplicity I installed the ZigStar TI CC2652P/P7 FW Flasher custom addon in Home Assistant, and updated the Sonoff stick from there (remember to stop Z2M first). Updated to the 20240710 version of the coordinator firmware and started z2m again. All actions are now captured and working fine.

JohnBull-1975, 2025-01-19:
I just bought a new STYRBAR and can't get it to work
Updated Z2M to 2.0.0.2 this afternoon and my two old STYRBAR running FW 1.0.024 still works fine as before.

The new STYRBAR came with FW 2.4.5
Inclusion looked to work and it came in as SUPPORTED and I was very eager to try if it worked, so I switched to "Exposes" tab and pressed the buttons. On-Off-On-Off and that worked (was indicated in Z2M Exposes tab).
Tried left-right-left-right but no response.
Got some red errors shooting up in lower left corner and after that no response from On or Off.

Found this in the log:

[image]

Tried to take out and re-insert batteries and then I could see this in the log:

[image]
But still  no response in "Expose" tab

Otherwise everything looks ok...:

[image]

Running the Sonoff dongle E, about 2 years old:

[image]

JohnBull-1975, 2025-01-25:
aha...it was on fw 2.4.5
Updated it to 2.4.16 and now it works!!!
Long click left and right requires a very very long click, like 3 seconds at least, but that's the same with the other older STYRBAR on fw 1.0.0.24 I have.

Dinth, 2025-01-26:
It seems that i have a similar problem. 
I have a bunch of E2001 around my house with 1.0.24 firmware. TBH i never used them with Z2M because they were bound to the lights directly.
Unfortunately recently i have upgraded one (a simple misclick) to 2.4.16. Obviously, the group binding stopped to work, but that's not the main problem.

When i have setting up an automation to turn lights on an off, i have noticed that no actions are being reported at all, not even in Z2M UI. After some further troubleshooting, i have noticed that the device seems to work correctly but only after 2-3 seconds after either reconfiguration or replacing battery, then it dies. 
When trying to downgrade by sending {"id": "0xa49e69fffe874e04"} to zigbee2mqtt/bridge/request/device/ota_update/update/downgrade, im getting "Device didnt respont to OTA request" error, but after tinkering i have found that if i insert the battery and immediatelly send the command, i am getting a different error: `{"data":{},"error":"Update of 'Baby bedroom light remote' failed ([0xa49e69fffe874e04 | Remote Control N2] No image currently available)","status":"error"}`

dev-tb, 2025-03-03:
> but after tinkering i have found that if i insert the battery and immediatelly send the command,

Thanks, this fixed the issue for me. After having the SKYRBAR join my z2b it went Idle and nothing came through again. When finding your post, I tried re-inserting the batteries and pressed the buttons rapidly for a few good minutes. Z2b suddenly displayed to battery level and pushed a firmware update. Keeping to RC awake seems to fix the issue. Now if I can only think of a way to do it with the VALLHORN PIR sensor.

closed 2025-05-10 by github-actions[bot]