Koenkk/zigbee2mqtt #21964: Frient Intelligent Smoke Alarm (SMSZB-120) failed to configure

dajomas, 2024-03-27:
### What happened?

Adding multiple Frient Intelligent Smoke Alarms (SMSZB-120) seems to work succesfully but the following message appears 3 times:
```
Error 2024-03-27 09:44:48 Failed to configure '0x0015bc003101e692', attempt 3 (Error: Bind 0x0015bc003101e692/35 genBasic from '0x00212effff05f5d7/1' failed (Error: status: 140) at DeconzAdapter.bind (/srv/zigbee2mqtt.git/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:776:19) at Endpoint.bind (/srv/zigbee2mqtt.git/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:551:13) at Object.bind (/srv/zigbee2mqtt.git/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:43:9) at Object.configure (/srv/zigbee2mqtt.git/node_modules/zigbee-herdsman-converters/src/devices/develco.ts:495:13) at Configure.configure (/srv/zigbee2mqtt.git/lib/extension/configure.ts:122:13) at EventEmitter.wrappedCallback (/srv/zigbee2mqtt.git/lib/eventBus.ts:174:17))
```

### What did you expect to happen?

Succesful inclusion of the Frient Intelligent Smoke Alarms (SMSZB-120) without the error message

### How to reproduce it (minimal and precise)

- Remove the Frient Intelligent Smoke Alarm (SMSZB-120) via the web interface if it exists
- Allow Zigbee2mqtt to permit joins
- Put the Frient Intelligent Smoke Alarm (SMSZB-120) in inclusion mode

Zigbee2mqtt will find the device, add it to the device list, and will try to configure the device. This is when the error appears 

### Zigbee2MQTT version

1.36.0

### Adapter firmware version

0x26720700

### Adapter

ConBee2/RaspBee2

### Setup

Plain on RPi from git repository via systemd

### Debug log

Here is the debug log starting at the removal of the device:

[debug_log.txt](https://github.com/Koenkk/zigbee2mqtt/files/14770740/debug_log.txt)

Articole, 2024-03-30:
Hello, i have exactly the same error with SMSZB-120.

After that, there are missing values like Battery and Temperature...
Temporarly, for getting theses values working, i followed this [https://github.com/Koenkk/zigbee2mqtt/issues/14997](url)

But the error continue to appear.

Zigbee2MQTT version : 1.36.0 (Last Stable)
Adapter firmware version : 0x26500900 (Last)
Adapter : ConBee3
Raspberry PI 3B+ : Fresh Install to test if it was from my configuration

Rappelkiste98, 2024-04-05:
As far as i know this problem relais to the ConBee Coordinators and the Firmware for this. Other Coordinators doesn't have this problem. It also can be a problem with the driver for ConBee

grasmachien, 2024-10-25:
Same here and I have the HA skyconnect. These devices seem to be trash.

mamrai1, 2024-10-28:
Fixed in latest development z2m

closed 2025-01-06 by github-actions[bot]