Koenkk/zigbee2mqtt #21965: Failed to configure

2proof4u, 2024-03-27:
### What happened?

Hi

I experience many strange problems and it properly is connected some how.
1=Zigbee2mqtt stop responding and require a restart (2-3 times a week)
2=Sometimes it is very slow and takes many secs before a light turn on (every  
3=Devices last seen stops and hang in last seen state

My log is constantly full of errors and new errors comes every 30 sec or so from different devices but they still work.

I have changed from a Sonoff dongle (CC2652P2) coordinator to a zigstar UZG-01 (CC2652P7) because I though the coordinator was to busy some how.

My zigbee channel is 25 and my network is only on channel 1 and 6 (live in a house so no disturbing from others). 

I have 188 devices and 140 of them is routers. I have 3 sonoff (CC2652P2) as router and 5 Ikea range extender so I am sure my zigbee network is ok.

I simply don’t know what to do to get it stable or where to look. It is so difficult when I don’t understand the basic and what to look for.

Right now, I have a sensor last seen for xx hours ago so the problem most be from Zigbee2mqtt and out I think.
The sensor is located 5 m from the coordinator and there are 5 lights between them + a range extender is 1,5m from it.
I get the following from MQTT exploder if it can help 

{"level":"error","message":"Failed to configure 'Bryggers - Motion sensor 1', attempt 3 (Error: Bind 0x0017880108665566/2 genPowerCfg from '0x00124b0024c0c4a7/1' failed (AREQ - ZDO - bindRsp after 10000ms)\n    at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23)\n    at /app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:692:28\n    at processTicksAndRejections (node:internal/process/task_queues:95:5)\n    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)\n    at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:551:13)\n    at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:43:9)\n    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/philips.ts:2235:13)\n    at Configure.configure (/app/lib/extension/configure.ts:122:13)\n    at Configure.onMQTTMessage (/app/lib/extension/configure.ts:55:21)\n    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))"}

### Zigbee2MQTT version

1.36.0

### Adapter firmware version

20230507

### Adapter

Zigstar UZG-01

### Setup

Always on pc with virtual box

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/14771426/log.txt)

mrmaximas, 2024-03-27:
@2proof4u Currently there is no stable firmware for CC2652P7, please read this https://github.com/Koenkk/Z-Stack-firmware/discussions/496#discussion-6382382

2proof4u, 2024-03-27:
My issue has nothing to do with this I am 100% sure because it started for a long time ago and in a desperate attempt I tried the P7 instead since it support 300 devices while p2 only support 200 devices. (upgraded 3 weeks ago +-)

I will gladly downgrade again if you want to see the same issue with a P2.

If I read your link properly it says it might be something with the firmware 20230507 instead and it might have started at me when I upgraded the firmware a year ago +-

Should I downgrade to 20221226 or try the test 20240315 firmware if I believe the P7 is not my problem or should i also downgrade back to P2 and then with what coordinator firmware.

mrmaximas, 2024-04-06:
@2proof4u Now P7 is very stable. Try it. I'm using CC1352P7_coordinator_20240315.hex last three days without any serious problems.

mrmaximas, 2024-04-06:
Soory, i do not use HA, i flashed it via ZigStart GW Multitul without any troubles. I see that you have the latest firmware 0.2.6 (prod-solo) in the UZG-01 like me

2proof4u, 2024-04-06:
Just for info

I got it to work now
Tried on another pc and then there was no problem autodetecting the ip adress.

closed 2024-11-04 by github-actions[bot]