Koenkk/zigbee2mqtt #21967: Disconnected tuya devices TS0601_soil

Destroyer061090, 2024-03-27:
### What happened?

Hello,

i have bought follow tuya products:
[https://it.aliexpress.com/item/1005005210965804.html?spm=a2g0o.order_list.order_list_main.96.385c36961aYPIn&gatewayAdapt=glo2ita 2](https://it.aliexpress.com/item/1005005210965804.html?spm=a2g0o.order_list.order_list_main.96.385c36961aYPIn&gatewayAdapt=glo2ita)
[https://it.aliexpress.com/item/1005005196806986.html?spm=a2g0o.order_list.order_list_main.76.385c36961aYPIn&gatewayAdapt=glo2ita 1](https://it.aliexpress.com/item/1005005196806986.html?spm=a2g0o.order_list.order_list_main.76.385c36961aYPIn&gatewayAdapt=glo2ita)

what is not clear is why only this kind of device are connected after the paring but after some time (i dont know if after days) the connection is lost… 

[image]

### What did you expect to happen?

I expect device connected and updated value sensor.
i guess the last update is before i reconnect manually the device.

[image]

### How to reproduce it (minimal and precise)

no idea

### Zigbee2MQTT version

1.36.0-1 

### Adapter firmware version

unknown

### Adapter

/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20230220210310-if00

### Setup

HA on docker, Z2M addon in HA + MQTT

chris1350, 2024-06-09:
Same here with 1.37.1-1 on Home Assistant.
The TS0601_soil is lost after ~3 to 4 days, if I restart Z2M the device is back and values updated.

engusz, 2024-07-06:
Same issue here :(

ReX1983, 2024-08-17:
Any luck?

Same issue with 5 new devices here. One of them works, the others haven't sent any update since pairing.

Pairing was smooth and the mesh is stable, with repeaters nearby the sensors.

closed 2025-01-06 by github-actions[bot]