Koenkk/zigbee2mqtt #21972: [New device support]: Novagitial TZB-2

rodrigogbs, 2024-03-27:
### Link

https://www.novadigitalsmart.com.br/produtos/interruptor-tomada-zigbee-com-duas-teclas-fisicas-4x2-tzb-2

### Database entry

{"id":66,"type":"Router","ieeeAddr":"0xa4c138fe15321e4a","nwkAddr":6416,"manufId":4417,"manufName":"_TZ3000_yervjnlj","powerSource":"Mains (single phase)","modelId":"TS0003","epList":[1,2,3,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,57344,57345,0],"outClusterList":[25,10],"clusters":{"manuSpecificBosch":{"attributes":{"53251":"AAAAAgAABAAA"}},"genOnOff":{"attributes":{"32768":0,"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":2,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2,"switchType":0}},"genBasic":{"attributes":{"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZ3000_yervjnlj","zclVersion":3,"appVersion":82,"modelId":"TS0003","powerSource":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001cd4ade8","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[4,5,6,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":256,"inClusterList":[4,5,6,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":1,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-68479682},"lastSeen":1711577592130}

### Comments

The product is identified by pattern from Z2M, but without some configuraion: POWER_ON_BEHAVIOR and BACKLIGHT_MODE, so I created an external converter to it.

### External definition

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
fingerprint: [
{
modelID: 'TS0003',
manufacturerName: '_TZ3000_yervjnlj'
}
],
model: 'TZB-2',
vendor: 'Novadigital',
description: 'Smart light switch - 2 gang with neutral 20A wire CUSTOM',
extend: tuya.extend.switch({power_on_behavior: true, backlightModeLowMediumHigh: true, endpoints: ['l1', 'l2', 'l3']}),
endpoint: () => {
return {'l1': 1, 'l2': 2, 'l3': 3};
},
meta: {multiEndpoint: true},
configure: async (device, coordinatorEndpoint) => {
const firstEndpoint = device.getEndpoint(1);
await firstEndpoint.read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
await reporting.bind(firstEndpoint, coordinatorEndpoint, ['genOnOff']);

const secondEndpoint = device.getEndpoint(1);
await secondEndpoint.read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
await reporting.bind(secondEndpoint, coordinatorEndpoint, ['genOnOff']);

const thirdEndpoint = device.getEndpoint(1);
await thirdEndpoint.read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
await reporting.bind(thirdEndpoint, coordinatorEndpoint, ['genOnOff']);
}
};

module.exports = definition;
```

rodrigogbs, 2024-08-12:
News?

rodrigogbs, 2024-09-03:
@Koenkk is anything missing here?

Can you check if it is possible to complement the device's exposes configuration?

Tks!

Koenkk, 2024-09-04:
Since the external converter you created works, could you make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/tuya.ts)?

closed 2025-01-06 by github-actions[bot]