Koenkk/zigbee2mqtt #21973: Z2M fails to start with Z-Wave.Me Z-Station - gets NACK on version negotiation after ASH Reset

PoltoS, 2024-03-28:
### What happened?

Z2M fails to start with [Z-Wave.Me Z-Station](https://z-wave.me/products/z-station/). Right after the reset the the version request returns NACK. At the same time, ZHA and Z-Way work perfectly with the same dongle.

It looks like  amissing ACK or some timing issue, but I can't explain it.

Same happened on the "edge" add-on (both ember and ezsp).

### How to reproduce it (minimal and precise)

Always happens at the Z2M start.

### Zigbee2MQTT version

1.36.0-1

### Adapter firmware version

?

### Adapter

Z-Wave.Me Z-Station

### Setup

Add-on Z2M on Home Assistant Yellow

### Debug log

With the EZSP driver:
```
[02:44:53] INFO: Preparing to start...
[02:44:54] INFO: Socat not enabled
[02:44:54] INFO: Zigbee Herdsman debug logging enabled
[02:44:56] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:debug 2024-03-28 02:45:02: Loaded state from file /config/zigbee2mqtt/state.json
Zigbee2MQTT:info  2024-03-28 02:45:02: Logging to console and directory: '/config/zigbee2mqtt/log/2024-03-28.02-45-02' filename: log.txt
Zigbee2MQTT:debug 2024-03-28 02:45:02: Removing old log directory '/config/zigbee2mqtt/log/2024-03-28.02-07-27'
Zigbee2MQTT:info  2024-03-28 02:45:03: Starting Zigbee2MQTT version 1.36.0 (commit #unknown)
Zigbee2MQTT:info  2024-03-28 02:45:03: Starting zigbee-herdsman (0.35.1)
Zigbee2MQTT:debug 2024-03-28 02:45:03: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[129,124,122,15,163,67,104,98],"networkKey":"HIDDEN","panID":13224},"serialPort":{"adapter":"ezsp","path":"/dev/serial/by-id/usb-Silicon_Labs_CP2105_Dual_USB_to_UART_Bridge_Controller_012B7B38-if01-port0","rtscts":false}}'
2024-03-28T01:45:03.563Z zigbee-herdsman:adapter Failed to validate path: 'Error: spawn udevadm ENOENT'
2024-03-28T01:45:03.566Z zigbee-herdsman:adapter:ezsp:debg Adapter concurrent: 8
2024-03-28T01:45:03.577Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[3,138,62,11,68,226,43,91,72,167,27,223,19,98,33,130],"panID":13224,"extendedPanID":[129,124,122,15,163,67,104,98],"channelList":[11]},"serialPort":{"rtscts":false,"path":"/dev/serial/by-id/usb-Silicon_Labs_CP2105_Dual_USB_to_UART_Bridge_Controller_012B7B38-if01-port0","adapter":"ezsp"},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
2024-03-28T01:45:03.581Z zigbee-herdsman:adapter:ezsp:uart RTS/CTS config is off, enabling software flow control.
2024-03-28T01:45:03.581Z zigbee-herdsman:adapter:ezsp:uart Opening SerialPort with {"path":"/dev/serial/by-id/usb-Silicon_Labs_CP2105_Dual_USB_to_UART_Bridge_Controller_012B7B38-if01-port0","baudRate":115200,"rtscts":false,"autoOpen":false,"parity":"none","stopBits":1,"xon":true,"xoff":true}
2024-03-28T01:45:03.595Z zigbee-herdsman:adapter:ezsp:uart Serialport opened
2024-03-28T01:45:03.596Z zigbee-herdsman:adapter:ezsp:uart Uart reseting
2024-03-28T01:45:03.597Z zigbee-herdsman:adapter:ezsp:uart --> Write reset
2024-03-28T01:45:03.599Z zigbee-herdsman:adapter:ezsp:uart --> [1ac038bc7e]
2024-03-28T01:45:03.602Z zigbee-herdsman:adapter:ezsp:uart -?- waiting reset
2024-03-28T01:45:04.743Z zigbee-herdsman:adapter:ezsp:uart <-- [c1020b0a527e]
2024-03-28T01:45:04.748Z zigbee-herdsman:adapter:ezsp:uart <-- RSTACK c1020b0a527e
2024-03-28T01:45:04.748Z zigbee-herdsman:adapter:ezsp:uart RSTACK Version: 2 Reason: RESET_SOFTWARE frame: c1020b0a527e
2024-03-28T01:45:04.752Z zigbee-herdsman:adapter:ezsp:uart -+- waiting reset success
2024-03-28T01:45:06.758Z zigbee-herdsman:adapter:ezsp:ezsp ==> version: {"desiredProtocolVersion":4}
2024-03-28T01:45:06.761Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"version","_id_":0,"_isRequest_":true,"desiredProtocolVersion":4}
2024-03-28T01:45:06.765Z zigbee-herdsman:adapter:ezsp:uart --> DATA (0,0,0): 00000004
2024-03-28T01:45:06.766Z zigbee-herdsman:adapter:ezsp:uart --> [004221a850ed2c7e]
2024-03-28T01:45:06.767Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (1)
2024-03-28T01:45:06.770Z zigbee-herdsman:adapter:ezsp:uart <-- [a0547d3a7e]
2024-03-28T01:45:06.771Z zigbee-herdsman:adapter:ezsp:uart <-- NAK (0): a0541a7e
2024-03-28T01:45:06.771Z zigbee-herdsman:adapter:ezsp:uart NAK Unexpected packet sequence 0
2024-03-28T01:45:10.769Z zigbee-herdsman:adapter:ezsp:uart --> Error: Error: {"sequence":1} after 4000ms
2024-03-28T01:45:10.769Z zigbee-herdsman:adapter:ezsp:uart -!- break waiting (1)
2024-03-28T01:45:10.769Z zigbee-herdsman:adapter:ezsp:uart Can't send DATA frame (0,0,0): 00000004
2024-03-28T01:45:11.270Z zigbee-herdsman:adapter:ezsp:uart ->> DATA (0,0,1): 00000004
2024-03-28T01:45:11.273Z zigbee-herdsman:adapter:ezsp:uart --> [084221a850ef017e]
2024-03-28T01:45:11.274Z zigbee-herdsman:adapter:ezsp:uart -?- rewaiting (1)
[... 102 lines cut ...]
2024-03-28T02:00:26.355Z zigbee-herdsman:adapter:ember:uart:ash Waiting for RSTACK... 1200/2500
2024-03-28T02:00:26.456Z zigbee-herdsman:adapter:ember:uart:ash Waiting for RSTACK... 1300/2500
2024-03-28T02:00:26.557Z zigbee-herdsman:adapter:ember:uart:ash Waiting for RSTACK... 1400/2500
2024-03-28T02:00:26.656Z zigbee-herdsman:adapter:ember:uart:ash Waiting for RSTACK... 1500/2500
2024-03-28T02:00:26.757Z zigbee-herdsman:adapter:ember:uart:ash Waiting for RSTACK... 1600/2500
2024-03-28T02:00:26.858Z zigbee-herdsman:adapter:ember:uart:ash Waiting for RSTACK... 1700/2500
2024-03-28T02:00:26.959Z zigbee-herdsman:adapter:ember:uart:ash Waiting for RSTACK... 1800/2500
2024-03-28T02:00:27.060Z zigbee-herdsman:adapter:ember:uart:ash Waiting for RSTACK... 1900/2500
2024-03-28T02:00:27.161Z zigbee-herdsman:adapter:ember:uart:ash Waiting for RSTACK... 2000/2500
2024-03-28T02:00:27.260Z zigbee-herdsman:adapter:ember:uart:ash Waiting for RSTACK... 2100/2500
2024-03-28T02:00:27.361Z zigbee-herdsman:adapter:ember:uart:ash Waiting for RSTACK... 2200/2500
2024-03-28T02:00:27.462Z zigbee-herdsman:adapter:ember:uart:ash Waiting for RSTACK... 2300/2500
2024-03-28T02:00:27.563Z zigbee-herdsman:adapter:ember:uart:ash Waiting for RSTACK... 2400/2500
Zigbee2MQTT:error 2024-03-28 03:00:27: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2024-03-28 03:00:27: Failed to start zigbee
Zigbee2MQTT:error 2024-03-28 03:00:27: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2024-03-28 03:00:27: Exiting...
Zigbee2MQTT:error 2024-03-28 03:00:27: Error: Failed to start EZSP layer with status=HOST_FATAL_ERROR.
    at EmberAdapter.initEzsp (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:805:23)
    at EmberAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:2666:24)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:62:27)
    at Controller.start (/app/lib/controller.ts:109:27)
    at start (/app/index.js:107:5)
```

closed 2024-04-06 by PoltoS

SmartHomeForDummies, 2024-04-21:
Same here!
configuration
`serial:
  port: /dev/serial/by-id/usb-Silicon_Labs_CP2105_Dual_USB_to_UART_Bridge_Controller_012B7AE7-if01-port0
  adapter: ezsp
  baudrate: 115200
  rtscts: true`

log
`
info  2024-04-21 02:06:08: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2024-04-21.02-06-08' filename: log.txt
info  2024-04-21 02:06:08: Starting Zigbee2MQTT version 1.36.1 (commit #ffc2ff1)
info  2024-04-21 02:06:08: Starting zigbee-herdsman (0.40.3)
error 2024-04-21 02:08:19: Error while starting zigbee-herdsman
error 2024-04-21 02:08:19: Failed to start zigbee
error 2024-04-21 02:08:19: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
error 2024-04-21 02:08:19: Exiting...
error 2024-04-21 02:08:19: Error: Failure to connect
    at Ezsp.connect (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:373:19)
    at Driver.startup (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:171:13)
    at Controller.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/opt/zigbee2mqtt/lib/zigbee.ts:62:27)
    at Controller.start (/opt/zigbee2mqtt/lib/controller.ts:109:27)
    at start (/opt/zigbee2mqtt/index.js:107:5)
`

PoltoS, 2024-04-21:
I've solved mine by upgrading the firmware. Z-Wave.Me made a firmware that ignores xon/xoff.

But I think it should be fixed in Z2M too

SmartHomeForDummies, 2024-04-23:
@PoltoS where can I get this firmware?
And how I can perform a firmware upset?

Thanks in advance.

PoltoS, 2024-04-26:
The firmware and the guide are available here: https://z-wave.me/support/tools-for-zigbee-openthread-ble/

jasonhollis, 2024-06-15:
Which firmware did you upgrade to?  I upgraded the Zigbee firmware e.g. not multi-protocol and Z2M won't start.

PoltoS, 2024-06-15:
Yes, to the Zigbee firmware and it starts perfectly. The one from site https://z-wave.me/support/tools-for-zigbee-openthread-ble/ (version 22.04.2024)