Koenkk/zigbee2mqtt #21975: Unable to pair Schneider Electric socket S520559

iggukfr, 2024-03-28:
### What happened?

Impossible to pair a Schneider Electric S520559 socket, it connects during pairing then leaves the network and this in a loop.
Identical behavior with a SONOFF CC2652P key

### What did you expect to happen?

Keep the socket paired

### How to reproduce it (minimal and precise)

Simply try a new association

### Zigbee2MQTT version

1.36

### Adapter firmware version

7.1.1.0 build 273

### Adapter

NabuCasa SkyConnect

### Setup

Proxmox VM

### Debug log

    info  2024-03-27 19:55:33: Starting interview of '0x70ac08fffe3da15b'
    info  2024-03-27 19:55:33: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x70ac08fffe3da15b","ieee_address":"0x70ac08fffe3da15b","status":"started"},"type":"device_interview"}'
    info  2024-03-27 19:55:44: Successfully interviewed '0x70ac08fffe3da15b', device has successfully been paired
    info  2024-03-27 19:55:44: Device '0x70ac08fffe3da15b' is supported, identified as: Schneider Electric Zigbee smart socket with power meter (EKO09716)
    info  2024-03-27 19:55:44: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Zigbee smart socket with power meter","exposes":[{"features":[{"access":7,"description":"On/off state of the switch","label":"State","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":1,"category":"diagnostic","description":"Instantaneous measured power","label":"Power","name":"power","property":"power","type":"numeric","unit":"W"},{"access":1,"description":"Sum of consumed energy","label":"Energy","name":"energy","property":"energy","type":"numeric","unit":"kWh"},{"access":7,"description":"Controls the behaviour when the device is powered on","label":"Power on behavior","name":"power_on_behavior","property":"power_on_behavior","type":"enum","values":["off","previous","on"]},{"access":1,"category":"diagnostic","description":"Instantaneous measured electrical current","label":"Current","name":"current","property":"current","type":"numeric","unit":"A"},{"access":1,"category":"diagnostic","description":"Measured electrical potential value","label":"Voltage","name":"voltage","property":"voltage","type":"numeric","unit":"V"},{"access":1,"category":"diagnostic","description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"EKO09716","options":[{"access":2,"description":"Calibrates the power value (percentual offset), takes into effect on next report of device.","label":"Power calibration","name":"power_calibration","property":"power_calibration","type":"numeric"},{"access":2,"description":"Number of digits after decimal point for power, takes into effect on next report of device. This option can only decrease the precision, not increase it.","label":"Power precision","name":"power_precision","property":"power_precision","type":"numeric","value_max":3,"value_min":0},{"access":2,"description":"Calibrates the energy value (percentual offset), takes into effect on next report of device.","label":"Energy calibration","name":"energy_calibration","property":"energy_calibration","type":"numeric"},{"access":2,"description":"Number of digits after decimal point for energy, takes into effect on next report of device. This option can only decrease the precision, not increase it.","label":"Energy precision","name":"energy_precision","property":"energy_precision","type":"numeric","value_max":3,"value_min":0},{"access":2,"description":"Calibrates the current value (percentual offset), takes into effect on next report of device.","label":"Current calibration","name":"current_calibration","property":"current_calibration","type":"numeric"},{"access":2,"description":"Number of digits after decimal point for current, takes into effect on next report of device. This option can only decrease the precision, not increase it.","label":"Current precision","name":"current_precision","property":"current_precision","type":"numeric","value_max":3,"value_min":0},{"access":2,"description":"Calibrates the voltage value (percentual offset), takes into effect on next report of device.","label":"Voltage calibration","name":"voltage_calibration","property":"voltage_calibration","type":"numeric"},{"access":2,"description":"Number of digits after decimal point for voltage, takes into effect on next report of device. This option can only decrease the precision, not increase it.","label":"Voltage precision","name":"voltage_precision","property":"voltage_precision","type":"numeric","value_max":3,"value_min":0},{"access":2,"description":"State actions will also be published as 'action' when true (default false).","label":"State action","name":"state_action","property":"state_action","type":"binary","value_off":false,"value_on":true}],"supports_ota":false,"vendor":"Schneider Electric"},"friendly_name":"0x70ac08fffe3da15b","ieee_address":"0x70ac08fffe3da15b","status":"successful","supported":true},"type":"device_interview"}'
    info  2024-03-27 19:55:44: Configuring '0x70ac08fffe3da15b'
    info  2024-03-27 19:55:44: Successfully configured '0x70ac08fffe3da15b'
    info  2024-03-27 19:55:47: MQTT publish: topic 'zigbee2mqtt/0x70ac08fffe3da15b', payload '{"current":null,"energy":null,"linkquality":192,"power":null,"power_on_behavior":null,"state":"OFF","voltage":232}'
    info  2024-03-27 19:55:47: MQTT publish: topic 'zigbee2mqtt/0x70ac08fffe3da15b', payload '{"current":null,"energy":null,"linkquality":255,"power":null,"power_on_behavior":null,"state":"ON","voltage":232}'
    info  2024-03-27 19:55:47: MQTT publish: topic 'zigbee2mqtt/0x70ac08fffe3da15b', payload '{"current":null,"energy":null,"linkquality":252,"power":null,"power_on_behavior":null,"state":"ON","voltage":232}'
    info  2024-03-27 19:55:47: MQTT publish: topic 'zigbee2mqtt/0x70ac08fffe3da15b', payload '{"current":null,"energy":null,"linkquality":252,"power":null,"power_on_behavior":null,"state":"OFF","voltage":232}'
    info  2024-03-27 19:55:48: MQTT publish: topic 'zigbee2mqtt/0x70ac08fffe3da15b', payload '{"current":null,"energy":null,"linkquality":255,"power":null,"power_on_behavior":null,"state":"ON","voltage":232}'
    info  2024-03-27 19:55:49: MQTT publish: topic 'zigbee2mqtt/0x70ac08fffe3da15b', payload '{"current":null,"energy":null,"linkquality":255,"power":null,"power_on_behavior":null,"state":"ON","voltage":232}'
    info  2024-03-27 19:55:49: MQTT publish: topic 'zigbee2mqtt/0x70ac08fffe3da15b', payload '{"current":null,"energy":null,"linkquality":252,"power":null,"power_on_behavior":null,"state":"OFF","voltage":232}'
    info  2024-03-27 19:55:49: MQTT publish: topic 'zigbee2mqtt/0x70ac08fffe3da15b', payload '{"current":null,"energy":null,"linkquality":252,"power":null,"power_on_behavior":null,"state":"ON","voltage":232}'
    info  2024-03-27 19:55:49: MQTT publish: topic 'zigbee2mqtt/0x70ac08fffe3da15b', payload '{"current":null,"energy":null,"linkquality":248,"power":null,"power_on_behavior":null,"state":"OFF","voltage":232}'
    info  2024-03-27 19:55:50: MQTT publish: topic 'zigbee2mqtt/0x70ac08fffe3da15b', payload '{"current":null,"energy":null,"linkquality":248,"power":null,"power_on_behavior":null,"state":"ON","voltage":232}'
    warn  2024-03-27 19:55:51: Device '0x70ac08fffe3da15b' left the network
    info  2024-03-27 19:55:51: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x70ac08fffe3da15b","ieee_address":"0x70ac08fffe3da15b"},"type":"device_leave"}'
    warn  2024-03-27 19:55:51: Device '0x70ac08fffe3da15b' left the network
    info  2024-03-27 19:55:51: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x70ac08fffe3da15b","ieee_address":"0x70ac08fffe3da15b"},"type":"device_leave"}'
    warn  2024-03-27 19:55:51: Device '0x70ac08fffe3da15b' left the network
    info  2024-03-27 19:55:51: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x70ac08fffe3da15b","ieee_address":"0x70ac08fffe3da15b"},"type":"device_leave"}'
    warn  2024-03-27 19:55:51: Device '0x70ac08fffe3da15b' left the network
    info  2024-03-27 19:55:51: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x70ac08fffe3da15b","ieee_address":"0x70ac08fffe3da15b"},"type":"device_leave"}'
    warn  2024-03-27 19:55:52: Device '0x70ac08fffe3da15b' left the network
    info  2024-03-27 19:55:52: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x70ac08fffe3da15b","ieee_address":"0x70ac08fffe3da15b"},"type":"device_leave"}'
    warn  2024-03-27 19:55:52: Device '0x70ac08fffe3da15b' left the network

remidebette, 2024-04-30:
Hi, I have seen this as well on 1 socket out of five that I own, no idea how to fix it.
I removed the device from z2m and then also factory reset the device to no avail

My ZigBee dongle is a Sonoff P

 ```

Info 2024-04-30 11:52:34Successfully changed options

Info 2024-04-30 11:52:34MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"e3vzg-6"}'

Info 2024-04-30 11:52:36MQTT publish: topic 'zigbee2mqtt/Multiprise Bureau', payload '{"current":0.01,"energy":14.29,"linkquality":248,"power":3.5,"state":"ON","voltage":229.86}'

Info 2024-04-30 11:52:37Successfully interviewed '0x187a3efffe65196a', device has successfully been paired

Info 2024-04-30 11:52:37Device '0x187a3efffe65196a' is supported, identified as: Schneider Electric Zigbee smart socket with power meter (EKO09716)

Info 2024-04-30 11:52:37MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Zigbee smart socket with power meter","exposes":[{"features":[{"access":7,"description":"On/off state of the switch","label":"State","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":1,"category":"diagnostic","description":"Instantaneous measured power","label":"Power","name":"power","property":"power","type":"numeric","unit":"W"},{"access":1,"description":"Sum of consumed energy","label":"Energy","name":"energy","property":"energy","type":"numeric","unit":"kWh"},{"access":7,"description":"Controls the behaviour when the device is powered on","label":"Power on behavior","name":"power_on_behavior","property":"power_on_behavior","type":"enum","values":["off","previous","on"]},{"access":1,"category":"diagnostic","description":"Instantaneous measured electrical current","label":"Current","name":"current","property":"current","type":"numeric","unit":"A"},{"access":1,"category":"diagnostic","description":"Measured electrical potential value","label":"Voltage","name":"voltage","property":"voltage","type":"numeric","unit":"V"},{"access":7,"description":"Set indicator mode","label":"Indicator mode","name":"indicator_mode","property":"indicator_mode","type":"enum","values":["reverse_with_load","consistent_with_load","always_off","always_on"]},{"access":1,"category":"diagnostic","description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"EKO09716","options":[{"access":2,"description":"Calibrates the power value (percentual offset), takes into effect on next report of device.","label":"Power calibration","name":"power_calibration","property":"power_calibration","type":"numeric"},{"access":2,"description":"Number of digits after decimal point for power, takes into effect on next report of device. This option can only decrease the precision, not increase it.","label":"Power precision","name":"power_precision","property":"power_precision","type":"numeric","value_max":3,"value_min":0},{"access":2,"description":"Calibrates the energy value (percentual offset), takes into effect on next report of device.","label":"Energy calibration","name":"energy_calibration","property":"energy_calibration","type":"numeric"},{"access":2,"description":"Number of digits after decimal point for energy, takes into effect on next report of device. This option can only decrease the precision, not increase it.","label":"Energy precision","name":"energy_precision","property":"energy_precision","type":"numeric","value_max":3,"value_min":0},{"access":2,"description":"Calibrates the current value (percentual offset), takes into effect on next report of device.","label":"Current calibration","name":"current_calibration","property":"current_calibration","type":"numeric"},{"access":2,"description":"Number of digits after decimal point for current, takes into effect on next report of device. This option can only decrease the precision, not increase it.","label":"Current precision","name":"current_precision","property":"current_precision","type":"numeric","value_max":3,"value_min":0},{"access":2,"description":"Calibrates the voltage value (percentual offset), takes into effect on next report of device.","label":"Voltage calibration","name":"voltage_calibration","property":"voltage_calibration","type":"numeric"},{"access":2,"description":"Number of digits after decimal point for voltage, takes into effect on next report of device. This option can only decrease the precision, not increase it.","label":"Voltage precision","name":"voltage_precision","property":"voltage_precision","type":"numeric","value_max":3,"value_min":0},{"access":2,"description":"State actions will also be published as 'action' when true (default false).","label":"State action","name":"state_action","property":"state_action","type":"binary","value_off":false,"value_on":true}],"supports_ota":false,"vendor":"Schneider Electric"},"friendly_name":"0x187a3efffe65196a","ieee_address":"0x187a3efffe65196a","status":"successful","supported":true},"type":"device_interview"}'

Info 2024-04-30 11:52:37Configuring '0x187a3efffe65196a'
```

Success logs

```
Warning 2024-04-30 11:53:06Device '0x187a3efffe65196a' left the network
```

What can I do?

remidebette, 2024-05-04:
I just updated to z2m to 1.37.0, upgraded the firmware of the sonoff dongle E to 7.4.2 and switched to ember.

I factory reset the plug and tried again, same result. 
Here are the logs
```
[2024-05-04 14:57:09] info: 	z2m: Zigbee: allowing new devices to join.
[2024-05-04 14:57:09] info: 	zh:ember: [STACK STATUS] Network opened.
[2024-05-04 14:57:28] info: 	zh:controller: Interview for '0x187a3efffe65196a' started
[2024-05-04 14:57:28] info: 	z2m: Device '0x187a3efffe65196a' joined
[2024-05-04 14:57:28] info: 	z2m: Starting interview of '0x187a3efffe65196a'
[2024-05-04 14:57:31] warning: 	zh:ember: [ZDO] Node descriptor for "35306" reports device is only compliant to revision "22" of the ZigBee specification (current revision: 23).
[2024-05-04 14:57:33] info: 	zh:controller: Succesfully interviewed '0x187a3efffe65196a'
[2024-05-04 14:57:33] info: 	z2m: Successfully interviewed '0x187a3efffe65196a', device has successfully been paired
[2024-05-04 14:57:33] info: 	z2m: Device '0x187a3efffe65196a' is supported, identified as: Schneider Electric Zigbee smart socket with power meter (EKO09716)
[2024-05-04 14:57:33] info: 	z2m: Configuring '0x187a3efffe65196a'
[2024-05-04 14:57:34] info: 	z2m: Successfully configured '0x187a3efffe65196a'
[2024-05-04 14:58:08] warning: 	z2m: Device '0x187a3efffe65196a' left the network
```

Especially, I am not sure what this means
```
Node descriptor for "35306" reports device is only compliant to revision "22" of the ZigBee specification (current revision: 23).
```

Nerivec, 2024-05-16:
@remidebette The compliance thing is just a warning, in case you encounter issues with the device, that could be the cause (old Zigbee revision). It's usually not a problem unless you see a device showing "pre-21", that's when things start breaking 😅

Could you post a debug log of the pairing process for this device? Looks like it might need a workaround since this appears to affect both zstack and emberznet.

remidebette, 2024-05-20:
Hi, please see my debug logs on ember, Sonoff dongle E
Core: 2024.5.4
Supervisor: 2024.05.1
Operating System: 12.3 
z2m: 1.37.1-1

(if also contains other logs, I was not sure how to split the part that would be relevant for you)

[log.log](https://github.com/Koenkk/zigbee2mqtt/files/15378435/log.log)

remidebette, 2024-11-23:
Not stale

closed 2025-02-01 by github-actions[bot]