Koenkk/zigbee2mqtt #21979: [New device support]: FORIA F00YK04-18-1

dinamik1488, 2024-03-28:
### Link

https://ledron.ru/catalog/umnyy-dom/titanium/

### Database entry

{"id":30,"type":"EndDevice","ieeeAddr":"0x3425b4fffe416736","nwkAddr":36686,"manufId":4098,"manufName":"_TZE200_dhke3p9w","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"J-�-i�-�-i�-�-i\u000e.�-i\u001a.�-i'.�-iCD�-\u0011","65506":31,"65508":0,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE200_dhke3p9w","appVersion":65,"stackVersion":0,"dateCode":"","zclVersion":3,"powerSource":3}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-708457359},"lastSeen":1711627811433}

### Comments

Please add.
[image]

### External definition

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils')

const definition = {
    fingerprint: [
        {
            modelID: "TS0601",
            manufacturerName: "_TZE200_dhke3p9w",
        }
    ],
    model: 'F00YK04-18-1',
    vendor: 'FORIA',
    description: '18 scenes remote',
    powerSource: 'Mains (single phase)',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints,],
    configure: tuya.configureMagicPacket,
    exposes: [e.battery(), e.battery_voltage(),

        e.action(['scene_1', 'scene_2', 'scene_3', 'scene_4', 'scene_5', 'scene_6', 'scene_7', 'scene_8', 'scene_9', 'scene_10', 'scene_11', 'scene_12', 'scene_13', 'scene_14', 'scene_15', 'scene_16', 'scene_17', 'scene_18'])

    ],
    endpoint: (device) => {
        return {'l1': 1, 'l2': 1, 'l3': 1, 'l4': 1, 'l5': 1, 'l6': 1, 'l7': 1, 'l8': 1,'l9': 1, 'l10': 1, 'l11': 1, 'l12': 1, 'l13': 1, 'l14': 1, 'l15': 1, 'l16': 1, 'l17': 101, 'l18': 102};
    },
    meta: {
        multiEndpoint: true,
        tuyaDatapoints: [
            [1, 'action', tuya.valueConverter.static('scene_1')],
            [2, 'action', tuya.valueConverter.static('scene_2')],
            [3, 'action', tuya.valueConverter.static('scene_3')],
            [4, 'action', tuya.valueConverter.static('scene_4')],
            [5, 'action', tuya.valueConverter.static('scene_5')],
            [6, 'action', tuya.valueConverter.static('scene_6')],
            [7, 'action', tuya.valueConverter.static('scene_7')],
            [8, 'action', tuya.valueConverter.static('scene_8')],
            [9, 'action', tuya.valueConverter.static('scene_9')],
            [10, 'action', tuya.valueConverter.static('scene_10')],
            [11, 'action', tuya.valueConverter.static('scene_11')],
            [12, 'action', tuya.valueConverter.static('scene_12')],
            [13, 'action', tuya.valueConverter.static('scene_13')],
            [14, 'action', tuya.valueConverter.static('scene_14')],
            [15, 'action', tuya.valueConverter.static('scene_15')],
            [16, 'action', tuya.valueConverter.static('scene_16')],
            [101, 'action', tuya.valueConverter.static('scene_17')],
            [102, 'action', tuya.valueConverter.static('scene_18')],

        ],
    },
};

module.exports = definition;
```

Koenkk, 2024-03-29:
does it work with this converter?

dinamik1488, 2024-03-29:
Yes

Koenkk, 2024-03-29:
I guess the battery is another datapoint, enable debug logging, run z2m for 24 hours and see if you have any `Datapoint XX not defined for '_TZE200_dhke3p9w'` in the log.

dinamik1488, 2024-03-29:
Ок

dinamik1488, 2024-03-30:
There is nothing interesting in the logs. the genPowerCfg cluster is missing

Koenkk, 2024-03-31:
Then let's ignore it, could you make a [PR](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts)?

dinamik1488, 2024-03-31:
Unfortunately, I do not know how to do this.

closed 2024-04-09 by dinamik1488