Koenkk/zigbee2mqtt #21982: Deprecated flag `color_mode` used in MQTT JSON light config (HA 2024.4.0b0)

hmmbob, 2024-03-28:
### What happened?

> Deprecated flag `color_mode` used in MQTT JSON light config , the `color_mode` flag is not used anymore and should be removed. Got: {'availability': [{'topic': 'zigbee2mqtt/bridge/state', 'value_template': '{{ value_json.state }}'}], 'brightness': True, 'brightness_scale': 254, 'color_mode': True, 'command_topic': 'zigbee2mqtt/Ledstrip mini/set', 'device': {'configuration_url': 'https://zigbee2mqtt.hmmbob.nl:6184//#/device/0xcc86ecfffe5ba135/info', 'identifiers': ['zigbee2mqtt_0xcc86ecfffe5ba135'], 'manufacturer': 'Lidl', 'model': 'Livarno Lux smart LED light strip 2.5m (HG06104A)', 'name': 'Ledstrip mini', 'via_device': 'zigbee2mqtt_bridge_0x00124b0024c94c43'}, 'effect': True, 'effect_list': ['blink', 'breathe', 'okay', 'channel_change', 'finish_effect', 'stop_effect'], 'max_mireds': 500, 'min_mireds': 150, 'name': None, 'object_id': 'ledstrip_mini', 'origin': {'name': 'Zigbee2MQTT', 'sw_version': '1.36.0', 'support_url': 'https://www.zigbee2mqtt.io'}, 'schema': 'json', 'state_topic': 'zigbee2mqtt/Ledstrip mini', 'supported_color_modes': ['xy', 'color_temp'], 'unique_id': '0xcc86ecfffe5ba135_light_zigbee2mqtt', 'platform': 'mqtt'}. This will stop working in Home Assistant Core 2025.3

### Zigbee2MQTT version

1.36.0

### Adapter firmware version

20220103

### Adapter

Sonoff

### Setup

Docker container

Koenkk, 2024-03-29:
Since it will break in 1 year from now, lets keep it in to preserve compatibility with older HA versions

javiges, 2024-04-04:
and how can it be fixed, thank you

amacomber, 2024-04-06:
Also having this same issue

TermeHansen, 2024-04-06:
Wouldn't it make more sense to have a setting where one can have a old deprecated flags like this posted. I mean if you have latest zb2mqtt, you probably also have latest HA in far most usercases I believe...

danielkihlgren, 2024-04-06:
I agree with @TermeHansen. It's better to make it work with latest versions. Those who choose to stay on old HA are probably those who use old versions of zigbee2mqtt.

pterisaur, 2024-04-06:
Same issue — we no longer have color-changing lights in our house 😞 ...is there a temporary fix?

sdotter, 2024-04-07:
Same here...I wanna always fix deprecated warnings as soon as possible 🥳

hase-00, 2024-04-08:
I have the same issue. Home Assistant is evolving so rapidly that it makes no sense to use old versions. Those who use old HA versions should also use the corresponding old addons and integrations. Therefore, I believe that the problem should be solved promptly. That's my humble opinion on the matter.

Koenkk, 2024-04-08:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-04-08 by Koenkk