Koenkk/zigbee2mqtt #21988: Gledopto GL-C-006P color temperature controls brightness after firmware update

Swampen, 2024-03-28:
### What happened?

Got the new firmware update available over OTA with z2m and updated it (Version: 10276802, Date: 20210719). 
After the update the controller have been acting up and it now controls the brightness when I try to change the color temperature.
Normal brightness control works fine.
I had no issues before the firmware update so perhaps an issue with the firmware?

### What did you expect to happen?

Adjusting the color temperature should should control the led temperature and not the brightness

### How to reproduce it (minimal and precise)

1. Update to latest available firmware (Version: 10276802, Date: 20210719)
2. Control temperature
    1. Lowering the color_temp increases brightness
    2. Increasing the color_temp decreases the brightness

### Zigbee2MQTT version

1.36.0

### Adapter firmware version

20230507

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus-P

### Setup

Add-on on Home Assistant Yellow

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/14795120/log.txt)
[herdsman_log.txt](https://github.com/Koenkk/zigbee2mqtt/files/14795121/herdsman_log.txt)

Loic691, 2024-03-29:
Can you explain ?

Loic691, 2024-03-29:
I will try this week with an other device which had been updated

lux73, 2024-03-30:
try to re-pair the Device as complete new Device - some of my Danfoss TRV has some weird behaviour after OTA, after new re-pairing all is good

Swampen, 2024-03-31:
> try to re-pair the Device as complete new Device - some of my Danfoss TRV has some weird behaviour after OTA, after new re-pairing all is good

Tried that now without any luck. Same behavior.

Loic691, 2024-03-31:
I had test this morning the device wich be updated with the last firmware.
The cool circuit didn't work.
The warm one yes.
So I think it's normal for you to think controling temperature color adjust the brightness.
This because colours between cool and warm are just a sum af brigjtness.

So i asked to the manufacturer a new firmware. M

Swampen, 2024-04-01:
@Loic691 I messaged Gledopto last week and got a new firmware this morning. I will try it later today.

Loic691, 2024-04-01:
I messaged too. They advise to not provide firmware on OTA because they said me same device référence can have several hardware version...

Loic691, 2024-04-01:
I suggest to wait your test and send me the firmware. I will test on my device

Loic691, 2024-04-01:
I think the gledopto are not manage correctly. They should have one firmware for ine reference product. The firmware should manage different hardware version

Loic691, 2024-04-01:
So if this exactly ,i will remove the firmware on ota

Swampen, 2024-04-01:
> I suggest to wait your test and send me the firmware. I will test on my device

Sure. I attached the one I got. 
My device have Q.C. sticker date marked as 01/2021
[GL-C-006P_10476801_2020401.zip](https://github.com/Koenkk/zigbee2mqtt/files/14822780/GL-C-006P_10476801_2020401.zip)

Loic691, 2024-04-01:
Thanks. Let me know about your test

Swampen, 2024-04-01:
> Thanks. Let me know about your test

Just finished the OTA installation from [my fork](https://github.com/Swampen/zigbee-OTA/tree/gl-c-006p-firmware) and the behavior is back to normal again

Loic691, 2024-04-01:
Great. I will try yours firmware.

Swampen, 2024-04-03:
@Loic691 Did you manage to test the new firmware?

Loic691, 2024-04-03:
Not enough time...
I started to gen json update file.
I hope this friday

Loic691, 2024-04-05:
Hi @Swampen 
I just tried your firmware and it is working now for one of my device.
I will push it on a PR !
And try to update other device with real OTA (not locally)

Swampen, 2024-04-05:
@Loic691 I only tested this with OTA from my [fork](https://github.com/Swampen/zigbee-OTA/tree/gl-c-006p-firmware), so I can just create a PR from that 🙂

closed 2024-04-06 by Koenkk

PskNorz, 2024-04-07:
Hey, yestarday i got a notification for update to 10476801, i did the update but now my controller is broken. I dont have warm colour, only natural and when i try to controll the colour temperature its controlling the brightness. Is there any way to roll back?

Swampen, 2024-04-07:
@PskNorz Did it work fine with the previous update I had issue with? Do you have "QC" sticker on the back off your device? Perhaps it is a different hardware revision?

@Loic691 Do your original device work with the new firmware?

PskNorz, 2024-04-07:
@Swampen Hey, it was first time that i used it, didnt make any update yestarday was the 1st and yes it was working fine. QC sticker is on the back yes.

Swampen, 2024-04-07:
@PskNorz What is the date of the year and month on the sticker? Mine has 01/2021

PskNorz, 2024-04-07:
> @PskNorz What is the date of the year and month on the sticker? Mine has 01/2021

[image]

polsup2, 2024-04-08:
The same issue for GL-C-006P. Got 3 down. Will re-pair them today, but expecting the same trouble. Looks like the OTA should be removed from PR.

[image]

Swampen, 2024-04-08:
@polsup2 Are the devices offline or is the color temperature the only issue? What is the year/month on your Q.C. stickers?

polsup2, 2024-04-08:
> @polsup2 Are the devices offline or is the color temperature the only issue? What is the year/month on your Q.C. stickers?

Device became offline. I didn't re-pair them yet. They are installed in electric box , so I need to disassemble something to get the photos of production date. Will provide photos little bit later.

polsup2, 2024-04-08:
> Will re-pair them today, but expecting the same tro

My expectations were wrong. After physical disconnection of power wire on controllers and re-pairing to network, everything works well: Brightness/wwcw.

I think it would be better to warn people that  OTA installation causes need to physically access the controller. In some cases,  this action is not easy.

Attaching the photo with production date.

[image]

PskNorz, 2024-04-09:
After a alot of repairs, nothing changes for me.

PskNorz, 2024-04-11:
@Swampen It is possible to reinstall the zip file that you have [here?](https://github.com/Koenkk/zigbee2mqtt/issues/21988#issuecomment-2029590352)

And if yes, can you explain me how?

Swampen, 2024-04-12:
@Loic691 Did you try to update the device you had tested [here](https://github.com/Koenkk/zigbee-OTA/issues/461#issuecomment-2016828926) to the latest firmware?

I have sent a new email to Gledopto with this new information to see if there is any changes in hardware revision between these QC stickers. Hopefully there is a way to distinguish them.

@monoimoi and @PskNorz If you want to downgrade the firmware, you can do the following:
1. Add a local `index.json` file to your z2m config https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#local-ota-index-and-firmware-files
2. Add the following to your local `index.json`
```json
[
    {
        "fileSize": 281342,
        "manufacturerCode": 4687,
        "imageType": 0,
        "sha512": "ef83e88563221eeac951f7f6d1769891b7eb9f4df44b0acd2cdd5e473cdd9f2772fd4ec4540b5f852d93fcc14637b28567a2869fb6ea3809fcf76ea5a230572b",
        "modelId": "GL-C-006P",
        "url": "https://github.com/Koenkk/zigbee-OTA/raw/master/images/Gledopto/GL-C-006P_V102.ota",
        "path": "images/Gledopto/GL-C-006P_V102.ota",
        "force": true
    }
]
```
3. Update the device

nick-dijkshoorn, 2024-04-12:
After upgrading to the newest firmware, color temperature was also broken.
I fixed the issue by downgrading the firmware with a local OTA index.
`[
     {
         "fileVersion": 10,
         "fileSize": 281342,
         "manufacturerCode": 4687,
         "imageType": 0,
         "sha512": "ef83e88563221eeac951f7f6d1769891b7eb9f4df44b0acd2cdd5e473cdd9f2772fd4ec4540b5f852d93fcc14637b28567a2869fb6ea3809fcf76ea5a230572b",
         "modelId": "GL-C-006P",
         "url": "https://github.com/Koenkk/zigbee-OTA/raw/master/images/Gledopto/GL-C-006P_V102.ota",
         "path": "images/Gledopto/GL-C-006P_V102.ota",
	 "force": true
     }
]`

nyok92, 2024-04-12:
```
{
    "color_mode": "color_temp",
    "linkquality": 109,
    "state": "ON",
    "update": {
        "installed_version": 10,
        "latest_version": 12,
        "state": "available"
    },
    "update_available": null,
    "brightness": 125,
    "color_temp": 370
}
```
QC passed 21/22
FW ; 10276802
FW date : 20210719

OTA update is not working.
Z2MQTT logs :

```
Info 2024-04-12 11:09:43Updating '0xcc86ecfffe187e3e' to latest firmware
Info 2024-04-12 11:09:47Update of '0xcc86ecfffe187e3e' at 0.00%
Info 2024-04-12 11:09:47MQTT publish: topic 'zigbee2mqtt/0xcc86ecfffe187e3e', payload '{"color_mode":"color_temp","color_temp":370,"linkquality":109,"state":"ON","update":{"installed_version":10,"latest_version":12,"progress":0,"state":"updating"},"update_available":null}'
Info 2024-04-12 11:12:17MQTT publish: topic 'zigbee2mqtt/0xcc86ecfffe187e3e', payload '{"color_mode":"color_temp","color_temp":370,"linkquality":109,"state":"ON","update":{"installed_version":10,"latest_version":12,"state":"available"},"update_available":null}'
Info 2024-04-12 11:12:17MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"0xcc86ecfffe187e3e"},"error":"Update of '0xcc86ecfffe187e3e' failed (OTA: Timeout, device did not request any image blocks)","status":"error","transaction":"2mfdd-1"}'
Error 2024-04-12 11:12:17Update of '0xcc86ecfffe187e3e' failed (OTA: Timeout, device did not request any image blocks)Info 2024-04-12 11:09:43Updating '0xcc86ecfffe187e3e' to latest firmware
Info 2024-04-12 11:09:47Update of '0xcc86ecfffe187e3e' at 0.00%
Info 2024-04-12 11:09:47MQTT publish: topic 'zigbee2mqtt/0xcc86ecfffe187e3e', payload '{"color_mode":"color_temp","color_temp":370,"linkquality":109,"state":"ON","update":{"installed_version":10,"latest_version":12,"progress":0,"state":"updating"},"update_available":null}'
Info 2024-04-12 11:12:17MQTT publish: topic 'zigbee2mqtt/0xcc86ecfffe187e3e', payload '{"color_mode":"color_temp","color_temp":370,"linkquality":109,"state":"ON","update":{"installed_version":10,"latest_version":12,"state":"available"},"update_available":null}'
Info 2024-04-12 11:12:17MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"0xcc86ecfffe187e3e"},"error":"Update of '0xcc86ecfffe187e3e' failed (OTA: Timeout, device did not request any image blocks)","status":"error","transaction":"2mfdd-1"}'
Error 2024-04-12 11:12:17Update of '0xcc86ecfffe187e3e' failed (OTA: Timeout, device did not request any image blocks)
```

Then I can't control the device nymore:

`2024-04-12 11:14:54Failed to read state of '0xcc86ecfffe187e3e' after reconnect (ZCL command 0xcc86ecfffe187e3e/11 genLevelCtrl.read(["currentLevel"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 51571 - 11 - 30 - 8 - 1 after 10000ms))`

Solved with re-pairing the device, but OTA update didn't worked.

Loic691, 2024-04-12:
> @Loic691 Did you try to update the device you had tested [here](https://github.com/Koenkk/zigbee-OTA/issues/461#issuecomment-2016828926) to the latest firmware?

Yes localy and it's OK.
I have another device en V1.0.1 which I can't update.
The update fails with this error `failed (OTA: Timeout, device did not request any image blocks)`
And the device hangs. I have to reboot it and repair again in Z2M.

I think GLEDOPTO have different hardware version with different firmware and there is no issues to check which hardware version we have. I had discussion with GLEDOPTO and they advise to not process update if the device works correctly.

They are easily reachable by email.

Loic691, 2024-04-12:
> Scratch that. I was looking at the wrong device.. I also have hwVersion 2..

Where we can find hardware version  ?

Loic691, 2024-04-12:
Sorry I found  !
For information, the device I updated locallty is hwVersion 2

The other one I can't update have no hardware version in interview...

nyok92, 2024-04-12:
Mine is hardware version 2 : 

Read result of 'genBasic': {"hwVersion":2,"appVersion":0,"zclVersion":3,"stackVersion":0,"modelId":"GL-C-006P","appProfileVersion":255,"swBuildId":"10276802","manufacturerName":"GLEDOPTO"}

PskNorz, 2024-04-12:
> > Scratch that. I was looking at the wrong device.. I also have hwVersion 2..
> 
> Where we can find hardware version ?

Can you tell me how?

Loic691, 2024-04-12:
In database.db file of Z2M config directory

Loic691, 2024-04-12:
> Mine is hardware version 2 :
> 
> Read result of 'genBasic': {"hwVersion":2,"appVersion":0,"zclVersion":3,"stackVersion":0,"modelId":"GL-C-006P","appProfileVersion":255,"swBuildId":"10276802","manufacturerName":"GLEDOPTO"}

For mine which is OK for update firmware
`{"genBasic":{"attributes":{"modelId":"GL-C-006P","manufacturerName":"GLEDOPTO","powerSource":0,"zclVersion":3,"appVersion":0,"stackVersion":0,"hwVersion":2,"dateCode":"20240401","swBuildId":"10476801"}}`

For other one which update failed
`{"genBasic":{"attributes":{"appVersion":0,"stackVersion":0,"dateCode":"","modelId":"GL-C-006P","zclVersion":3,"swBuildId":"V1.0.1"}}`

PskNorz, 2024-04-12:
> database.db

I got only this.
[image]

monoimoi, 2024-04-12:
> @Loic691 Did you try to update the device you had tested [here](https://github.com/Koenkk/zigbee-OTA/issues/461#issuecomment-2016828926) to the latest firmware?
>
> I have sent a new email to Gledopto [...]

The update would fail repeatedly until I added "fileVersion": 10 on top of the json file (@nick-dijkshoorn 's example). I have now successfully reverted back to the older firmware and everything works as it used to.
Hardware version 2. Thanks everyone for your input.

monoimoi, 2024-04-12:
> > database.db
> 
> I got only this.
> [image]
> 
> 

You can also find hwVersion by using z2m web interface.
Go to devices tab then click the device you want to query.
Click "dev console" tab
Endpoint: "11" -> Cluster "genBasic" -> attributes "hwVersion" and then click read.

PskNorz, 2024-04-12:
@monoimoi Thank you! I am a newbie and I am trying to learn how Z2M works.

So I think I have found the solution.

My hardware version is `Read result of 'genBasic': {"hwVersion":2} `
**Q.C Sticker 7/22**

I sent an email to Gledopto and I told them which Q.C Sticker I have, and they replied to me with an OTA Update file.

[GL-C-006P_10576802_20240412.zip](https://github.com/Koenkk/zigbee2mqtt/files/14961288/GL-C-006P_10576802_20240412.zip)

Because I am new to Z2M, I didn't know how to update it, so I switched back to Deconz because I knew how to do it. I updated the device and repaired it back to Z2M.

And it's done! Everything works as it should.

Thanks to everyone. I hope this helps others facing the same problem.

fry42, 2024-05-06:
Hi!

My GL-C-006P is also broken after the update!
My Q/C Sticker says 10/22 so the GL-C-006P_10576802_20240412.zip should be the right one for me...

But: I can't get z2m to update/downgrade/force the firmware.
I just get a "Images currently unavailable for device 'GL-C-006P', hardwareVersion '2', manufacturerName GLEDOPTO, {"fieldControl":0,"manufacturerCode":4687,"imageType":5138,"fileVersion":604057601}'" message in the logs.

The solution suggested by @Loic691 / updated by @monoimoi does not work for me.
If I set the imageType to 5138 it "detects" an update but fails on imageType.

I don't have Deconz available (or a coordinator supported by dconz).

Any advice on how to fix this?

Loic691, 2024-05-18:
I think it's better to remove OTA for this device....
There is several hardware version and we can't know how to manage them...