Koenkk/zigbee2mqtt #21991: Aqara devices won't connect to ZBDongle-P as Router

seederp2p, 2024-03-28:
### What happened?

I've a good working environment with Z2M 1.36.0-1 with one ZBDongle-P (CC2652P) as coordinator and multiple aqara devices such as Temperature Sensor WSDCGQ11LM, Water Leak Sensor SJCGQ11LM and Door / Window Sensor MCCGQ11LM. Everything works fine.

Because I've two devices (Water Leak Sensor) that have very weak signal, I added another ZBDongle-P with latest Router Firmware.

It's correctly display in Z2M.

I was expecting that my devices would find this router as a better peer and would connect... truth is, not a single device connected to it (based on the MAP).

I understand that are some issues reported with Aqara. I would like to know if there's something we can do in this environment.

### What did you expect to happen?

Devices should connect to the best Dongle based on the signal.

### How to reproduce it (minimal and precise)

As described in "What happened"

### Zigbee2MQTT version

1.36.0-1

### Adapter firmware version

20230507

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

### Setup

Home Assistant Addon

### Debug log

[debug.txt](https://github.com/Koenkk/zigbee2mqtt/files/14796421/debug.txt)

mrskycriper, 2024-03-31:
Aqara's Zigbee implementation is known for bad mesh rebuilding algorithm. Especially for battery powered devices (1st gen are the worst). So pairing it manually to the router will most likely keep it in place, as they don't charge their routers that much. There isn't much that could be done here otherwise. 1st gen sensors don't support OTA, so the fixed firmware will never come. You can search for alternative firmware, but the flashing will require soldering.

Also keep in mind that if the router dies, Aqara sensor will not seek a new one (Zigbee 3.0 models might, but they don't do this often), so it's not recommended as the only security device.

andreapx, 2024-04-27:
@seederp2p have you solved this problem? I'm in the same boat with an Aqara motion sensor that doesn't want to connect to any router (Sonoff of bTicino).

closed 2024-11-24 by github-actions[bot]