Koenkk/zigbee2mqtt #21999: TO-Q-SY2-163JZT Problem Settings Protections

omarantelmi, 2024-03-30:
### What happened?

I recently purchased this device: TO-Q-SY2-163JZT
I can't set up the various protections.

### What did you expect to happen?

I expected the respective required values to be set.

### Zigbee2MQTT version

1.36.0 

### Adapter firmware version

 7.3.1.0 build 176

### Adapter

EZSP v12

### Setup

Addon on the Home Assistant operating system on Mini PC

### Debug log

[image]
[image]
[image]

schwarzelise, 2024-03-30:
Hello i have exact the same device and configuration and the same PROBLEM!

originalcom, 2024-04-03:
I have 3 of these. I have exactly the same problem. Dangerous to install as the device defaults to max amps i.e no overcurrent protection. Hopefully this can be fixed.

jkl83, 2024-04-12:
> I have 3 of these. I have exactly the same problem. Dangerous to install as the device defaults to max amps i.e no overcurrent protection. Hopefully this can be fixed.

This device is supposed to be mounted WITH overcurrent circuit breaker. It says so in the manual.

originalcom, 2024-04-12:
> > I have 3 of these. I have exactly the same problem. Dangerous to install as the device defaults to max amps i.e no overcurrent protection. Hopefully this can be fixed.
> 
> This device is supposed to be mounted WITH overcurrent circuit breaker. It says so in the manual.

No, this is definitely incorrect. The model I have does have current limiting: https://www.aliexpress.com/item/1005006139125562.html

The previous versions which I guess you are confusing these with were only adjustable on the Wifi models.

I did have further protection as it tripped the main 63A breaker. As a temporary measure until these are fully supported by Zigbee2mqtt I have these connected to a separate Tuya zigbee hub. The device settings in Tuya allow you to set a over current threshold and I can confirm it works as expected. I have tested this in multiple scenarios

Please, if it's possible delete your post as it may cause this issue not to be looked at / resolved.

jkl83, 2024-04-12:
I have the same one.
Look at the manual inside the box.

[image]

https://elcb.net/product/zigbee-smart-circuit-breaker-over-current-under-voltage-protection-to-q-sy2-jzt/

https://elcb.net/wp-content/uploads/2024/02/SY1-SY2-Din-Rail-Smart-Switch-Operating-Manual.pdf

[image]

It's up to you. Matter of safety.

originalcom, 2024-04-12:
> I have the same one. Look at the manual inside the box. [...]

Yes correct hence the extra 63A breaker as described in my reply to you. They still have inbuilt current limiting enabling these to disconnect when the set current is exceeded. I think we should leave this discussion as it is I am only trying to get the full functionality of these supported by Zigbee2mqtt .

originalcom, 2024-04-17:
> I had the same story. In my case the reason was incorrect passing of interview when connected through some router. Then I had selected permit to join to coordinator only and paired device again the interview passed successfully. I have the only problem couldn't set the overvoltage threshold more than 255V it seems some limit of converter.

I was hoping it was that simple but I've tried completely removing it from Zigbee2mqtt both normally and force remove then adding/interviewing it again. I've also tried it on my backup system with a clean Zigbee2mqtt install but still exactly the same problem.

schwarzelise, 2024-04-17:
As seaman-git says
https://github.com/Koenkk/zigbee2mqtt/issues/21999#issuecomment-2053524181

Delete it and join it new to coordinaor ONLY, this worked perfect for me!
Thx Seaman-git

omarantelmi, 2024-04-17:
Removing the device and reassociating it later solves the issue.

closed 2024-04-17 by omarantelmi

originalcom, 2024-04-18:
> As seaman-git says [#21999 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/21999#issuecomment-2053524181)
> 
> Delete it and join it new to coordinaor ONLY, this worked perfect for me! Thx Seaman-git

Ok after doing this a few times this worked. Seems hit & miss but it does work eventually. Delete it and re add it and try changing the settings for amps etc. Repeat until it accepts the changes. I had to delete all 3 from my setup using force delete on the last one. Now all 3 added and working. Thank you

iaxon, 2024-09-16:
> I had the same story. In my case the reason was incorrect passing of interview when connected through some router. Then I had selected permit to join to coordinator only and paired device again the [...]

I just opened [a pr](https://github.com/Koenkk/zigbee-herdsman-converters/pull/7995) to fix the problem with the overvoltage treshold :)