Koenkk/zigbee2mqtt #22002: [New device support]: F00XN00-04-1

dinamik1488, 2024-03-30:
### Link

https://svetomaniya.ru/product/00000018475

### Database entry

{"id":31,"type":"EndDevice","ieeeAddr":"0xa4c1381c65b01ad8","nwkAddr":7305,"manufId":4417,"manufName":"_TZ3000_etufnltx","powerSource":"Battery","modelId":"TS1002","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":261,"inClusterList":[1,3,4,6,4096,57345,0],"outClusterList":[3,4,5,6,8,4096,25,10],"clusters":{"genBasic":{"attributes":{"65503":"XT�-\u0012^T�-\u0018_T�-\u0016","65506":55,"65508":0,"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZ3000_etufnltx","zclVersion":3,"appVersion":70,"modelId":"TS1002","powerSource":3}},"genOnOff":{"attributes":{"tuyaOperationMode":1}},"manuSpecificTuya_2":{"attributes":{"53265":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200,"batteryVoltage":30}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-2021484159},"lastSeen":1711804995069}

### Comments

Все работает. Но при диммирование скачет значения до 254 и обратно
Everything works. But when dimming, the values jump to 254 and back

[image]

### External definition

```shell
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS1002', manufacturerName: '_TZ3000_etufnltx'}],
        model: 'F00XN00-04-1',
        vendor: 'FORIA',
        description: 'Dimmer',
        fromZigbee: [fz.battery, fz.command_on, fz.command_off, fz.command_move_to_level, fz.command_move_to_color_temp, fz.command_step_color_temperature, fz.command_step],
        toZigbee: [],
        exposes: [e.battery(), e.battery_voltage(), e.action(['on', 'off', 'brightness_move_to_level', 'color_temperature_move'])],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await tuya.configureMagicPacket(device, coordinatorEndpoint, logger);
            await endpoint.command('genGroups', 'miboxerSetZones', {zones: [
                {zoneNum: 1, groupId: 101},
                {zoneNum: 2, groupId: 102},
                {zoneNum: 3, groupId: 103},
                {zoneNum: 4, groupId: 104},
            ]});
            await endpoint.command('genBasic', 'tuyaSetup', {}, {disableDefaultResponse: true});
        },
    };

module.exports = definition;
```

Koenkk, 2024-03-31:
> the values jump to 254 and back.

Looks like an issue of the device itself since z2m just passes the value.

Could you make a PR to add out-of-the-box support for this device?

dinamik1488, 2024-03-31:
Unfortunately, I do not know how to do this.

closed 2024-10-29 by github-actions[bot]