Koenkk/zigbee2mqtt #22004: [New device support]:

Krzysiek1979r, 2024-03-30:
### Link

https://aeotec.com/products/Aeotec-Smartthings-Pico-Switch/

### Database entry

{"id":3,"type":"Router","ieeeAddr":"0x6c5cb1fffe1c1eec","nwkAddr":58045,"manufId":4880,"manufName":"AEOTEC","powerSource":"Mains (single phase)","modelId":"ZGA003","epList":[1,2,3,4,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":2,"inClusterList":[0,2,3,4,5,6,9,1794,2820,64769,64770],"outClusterList":[3,10,25],"clusters":{"genBasic":{"attributes":{"stackVersion":0}},"genOnOff":{"attributes":{"onOff":0}},"haElectricalMeasurement":{"attributes":{"acPowerDivisor":10,"acPowerMultiplier":1,"acCurrentDivisor":1000,"acCurrentMultiplier":1,"acVoltageDivisor":10,"acVoltageMultiplier":1,"activePower":0,"rmsCurrent":0,"rmsVoltage":2224}},"seMetering":{"attributes":{"divisor":1000,"multiplier":1,"currentSummDelivered":[0,84]}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe9cab5c","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe9cab5c","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe9cab5c","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null},{"cluster":2820,"attrId":1291,"minRepIntval":10,"maxRepIntval":65000,"repChange":50,"manufacturerCode":null},{"cluster":2820,"attrId":1288,"minRepIntval":10,"maxRepIntval":65000,"repChange":50,"manufacturerCode":null},{"cluster":2820,"attrId":1285,"minRepIntval":10,"maxRepIntval":65000,"repChange":50,"manufacturerCode":null},{"cluster":1794,"attrId":0,"minRepIntval":10,"maxRepIntval":65000,"repChange":[0,100],"manufacturerCode":null}],"meta":{}},"2":{"profId":260,"epId":2,"devId":2,"inClusterList":[0,3,4,5,6,9,1794,2820],"outClusterList":[3,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0}},"genOnOff":{"attributes":{"onOff":0}},"haElectricalMeasurement":{"attributes":{"acPowerDivisor":10,"acPowerMultiplier":1,"acCurrentDivisor":1000,"acCurrentMultiplier":1,"acVoltageDivisor":10,"acVoltageMultiplier":1,"activePower":0,"rmsCurrent":0,"rmsVoltage":2224}},"seMetering":{"attributes":{"divisor":1000,"multiplier":1,"currentSummDelivered":[0,225]}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe9cab5c","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe9cab5c","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe9cab5c","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null},{"cluster":2820,"attrId":1291,"minRepIntval":10,"maxRepIntval":65000,"repChange":50,"manufacturerCode":null},{"cluster":2820,"attrId":1288,"minRepIntval":10,"maxRepIntval":65000,"repChange":50,"manufacturerCode":null},{"cluster":2820,"attrId":1285,"minRepIntval":10,"maxRepIntval":65000,"repChange":50,"manufacturerCode":null},{"cluster":1794,"attrId":0,"minRepIntval":10,"maxRepIntval":65000,"repChange":[0,100],"manufacturerCode":null}],"meta":{}},"3":{"profId":260,"epId":3,"devId":2096,"inClusterList":[0,3,4096,64768],"outClusterList":[3,4,5,6,8,4096],"clusters":{"genBasic":{"attributes":{"stackVersion":0}}},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":260,"epId":4,"devId":2096,"inClusterList":[0,3,64768],"outClusterList":[3,4,5,6,8],"clusters":{"genBasic":{"attributes":{"stackVersion":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"2023092115:16:49","swBuildId":"1.0.6","zclVersion":8,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1711810455425}

### Comments

The device appears as unrecognized. It is possible to turn individual contacts on and off, but more entities do not work

### External definition

```shell
const {deviceEndpoints, onOff, electricityMeter, identify} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['ZGA003'],
    model: 'ZGA003',
    vendor: 'AEOTEC',
    description: 'Automatically generated definition',
    extend: [deviceEndpoints({"endpoints":{"1":1,"2":2,"3":3,"4":4,"242":242}}), deviceEndpoints({"endpoints":{"1":1,"2":2}}), onOff({"powerOnBehavior":false,"endpointNames":["1","2"]}), electricityMeter(), identify()],
    meta: {"multiEndpoint":true},
};

module.exports = definition;
```

Koenkk, 2024-03-31:
> but more entities do not work

What more do you expect?

Krzysiek1979r, 2024-04-01:
Hi It expects correct identification and correct operation of power and energy measurementBRKrzysztof CzernielewskiOd: "Koen Kanters" ***@***.***&gt;Do: "Koenkk/zigbee2mqtt" ***@***.***&gt;; Wysłane: 20:30 Niedziela 2024-03-31Temat: Re: [Koenkk/zigbee2mqtt] [New device support]:  (Issue #22004)

but more entities do not work

What more do you expect?

johnjoemorgan, 2024-04-02:
I believe he's expecting power, voltage, amps and reporting????

Krzysiek1979r, 2024-04-05:
Hi sorry that you had to wait so long for answer.As you see below many of entities have unknown stateNieznany = Unknown

Also the theoretically working power measurement behaves strangely. It
counts incorrectly (too little) and sometimes there are jumps up and down (as
in the graph below).PozdrawiamKrzysztof CzernielewskiOd: "johnjoemorgan" ***@***.***&gt;Do: "Koenkk/zigbee2mqtt" ***@***.***&gt;; Wysłane: 14:46 Wtorek 2024-04-02Temat: Re: [Koenkk/zigbee2mqtt] [New device support]:  (Issue #22004)

I believe he's expecting power, voltage, amps and reporting????

MiAutomations, 2024-05-08:
Hello 

Can some one please confirm that using this device connected over the Z2M, you listening some sound of the relay as working, but there wasn't any thing in the Logbook of the device? 

Thank you

tech2mar, 2024-05-18:
There is also the Pico Shutter, this one is not year compatible but ismt seams to work (Relays clicking) I'm also very interested in this hardware.

Is there someone that needs more information got also the dual relay pico.

🙏

MiAutomations, 2024-11-03:
Hello 

I notice that this device is currently as supported on the Z2M 

But there are some features like Power on behavior, that is available in the other platforms like Samsung Smartthings and in the HA on ZHA, but is not available in the Z2M , can you please check if is possible to have this feature available also into the Z2M? 

Thank you in advance for your support and best regards 

Please let me know if you need any further information from the device

[image]

[image]

MiAutomations, 2025-01-03:
> This issue is stale because it has been open 60 days with no activity. Remove stale label or comment or this will be closed in 7 days

No its not solved, please keep this issue opened thank you

closed 2025-03-12 by github-actions[bot]