Koenkk/zigbee2mqtt #22011: [New device support]:   F00MB00-04-1

dinamik1488, 2024-03-31:
### Link

https://svetomaniya.ru/product/00000018469

### Database entry

{"id":33,"type":"Router","ieeeAddr":"0x187a3efffe301a3f","nwkAddr":6232,"manufId":4098,"manufName":"_TZE200_mfamvsdb","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65534":0,"modelId":"TS0601","manufacturerName":"_TZE200_mfamvsdb","stackVersion":0,"dateCode":"","zclVersion":3,"appVersion":66,"powerSource":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-708457359},"lastSeen":1711885288542}

### Comments

4 scenes switch.
The scene switches are working.
The switches are not working:
105 -  Atmosphere light setting
106 - Light sensivity setting
107 - Human sense setting
108 - Automatic(vibration motor) setting
[image]

### External definition

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils')

const definition = {
    fingerprint: [
        {
            modelID: "TS0601",
            manufacturerName: "_TZE200_mfamvsdb",
        }
    ],
    model: 'F00MB00-04-1',
    vendor: 'FORIA',
    description: '4 scenes remote',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints,],
    configure: tuya.configureMagicPacket,
    exposes: [

        e.action(['scene_1', 'scene_2', 'scene_3', 'scene_4'])

    ],
    endpoint: (device) => {
        return {'l1': 1, 'l2': 1, 'l3': 1, 'l4': 1};
    },
    meta: {
        multiEndpoint: true,
        tuyaDatapoints: [
            [1, 'action', tuya.valueConverter.static('scene_1')],
            [2, 'action', tuya.valueConverter.static('scene_2')],
            [3, 'action', tuya.valueConverter.static('scene_3')],
            [4, 'action', tuya.valueConverter.static('scene_4')],

        ],
    },
};

module.exports = definition;
```

Koenkk, 2024-04-01:
Looks good, please make a pull request again

dinamik1488, 2024-04-01:
After today's update, converter is not working. =(

closed 2024-10-29 by github-actions[bot]

milkas9, 2026-03-06:
how do you put this remote into pairing mode? I have four of these remotes but I can't pair them.