Koenkk/zigbee2mqtt #22016: [New device support]: TS0601 _TZE204_bjzrowv2 curtain motor

ggiannablue, 2024-03-31:
### Link

https://www.alibaba.com/product-detail/2023-Hot-Sale-Tuya-Zigbee-Electric_1601001656619.html?spm=a2700.shop_plgr.41413.22.6b3e17e5u5h7vl

### Database entry

{"id":6,"type":"Router","ieeeAddr":"0xa4c138dc88b5370f","nwkAddr":7084,"manufId":4417,"manufName":"_TZE204_bjzrowv2","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"Xr�-fYr�-\u0012","65506":56,"65508":0,"stackVersion":0,"dateCode":"","appVersion":74}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1711915796654}

### Comments

Hi.
I tried to follow the support page, but the device after many restart is always unsupported. 
I think the file is not considered by the integration in my version of HA (proxmox vm).
I have a second motor but is not the same and is correctly configured in my system ("_TZE200_bjzrowv2").
The type of the motor is : AT5810-ZB
From : Autorail co., LTD

The motor in my external definition is "_TZE204_bjzrowv2".

Thanks in advance for you help

### External definition

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [
        {modelID: 'TS0601', manufacturerName: '_TZE204_bjzrowv2'},
    ],
    model: 'TS0601_cover_1',
    vendor: 'TuYa',
    description: 'Curtain motor/roller blind motor/window pusher/tubular motor',
    fromZigbee: [legacy.fromZigbee.tuya_cover, fz.ignore_basic_report],
    toZigbee: [legacy.toZigbee.tuya_cover_control, legacy.toZigbee.tuya_cover_options],
    exposes: [
        e.cover_position().setAccess('position', ea.STATE_SET),
        e.composite('options', 'options', ea.STATE_SET)
            .withFeature(e.numeric('motor_speed', ea.STATE_SET)
                .withValueMin(0).withValueMax(255).withDescription('Motor speed'))
            .withFeature(e.binary('reverse_direction', ea.STATE_SET, true, false)
                .withDescription('Reverse the motor direction'))
    ],
};

module.exports = definition;
```

Koenkk, 2024-04-01:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) tomorrow.

closed 2024-04-01 by Koenkk

ggiannablue, 2024-04-01:
Many thanks for your work