Koenkk/zigbee2mqtt #22018: Since updating a couple of adapters the Z2M adapter does no longer start

ThorstenDtk, 2024-04-01:
### What happened?

I'm running ioBroker on raspberry OS on a raspi 4, Zigbee2MQTT is on the same machine. Both Z2M and the adapter were running fine until I upgraded 6 adapters through the IOB UI (don't remember which ones). Since the upgrade the Z2M adapter does no longer start:

host.raspberrypi | 2024-04-01 10:29:46.511 | info | Restart adapter system.adapter.zigbee2mqtt.0 because enabled
-- | -- | -- | --
host.raspberrypi | 2024-04-01 10:29:46.511 | error | instance system.adapter.zigbee2mqtt.0 terminated with code 1 (JS_CONTROLLER_STOPPED)
host.raspberrypi | 2024-04-01 10:29:46.510 | error | Caught by controller[0]: Node.js v20.12.0
host.raspberrypi | 2024-04-01 10:29:46.510 | error | Caught by controller[0]:     at Module._extensions..js (node:internal/modules/cjs/loader:1427:10)
host.raspberrypi | 2024-04-01 10:29:46.510 | error | Caught by controller[0]:     at Module._compile (node:internal/modules/cjs/loader:1369:14)
host.raspberrypi | 2024-04-01 10:29:46.510 | error | Caught by controller[0]:     at Object.<anonymous> (/opt/iobroker/node_modules/sharp/lib/constructor.js:10:1)
host.raspberrypi | 2024-04-01 10:29:46.509 | error | Caught by controller[0]:     at require (node:internal/modules/helpers:179:18)
host.raspberrypi | 2024-04-01 10:29:46.509 | error | Caught by controller[0]:     at Module.require (node:internal/modules/cjs/loader:1231:19)
host.raspberrypi | 2024-04-01 10:29:46.509 | error | Caught by controller[0]:     at Module._load (node:internal/modules/cjs/loader:1022:12)
host.raspberrypi | 2024-04-01 10:29:46.509 | error | Caught by controller[0]:     at Module.load (node:internal/modules/cjs/loader:1206:32)
host.raspberrypi | 2024-04-01 10:29:46.509 | error | Caught by controller[0]:     at Module._extensions..js (node:internal/modules/cjs/loader:1427:10)
host.raspberrypi | 2024-04-01 10:29:46.509 | error | Caught by controller[0]:     at Module._compile (node:internal/modules/cjs/loader:1369:14)
host.raspberrypi | 2024-04-01 10:29:46.508 | error | Caught by controller[0]:     at Object.<anonymous> (/opt/iobroker/node_modules/sharp/lib/sharp.js:114:9)
host.raspberrypi | 2024-04-01 10:29:46.508 | error | Caught by controller[0]:     See https://sharp.pixelplumbing.com/install
host.raspberrypi | 2024-04-01 10:29:46.508 | error | Caught by controller[0]: - Consult the installation documentation:
host.raspberrypi | 2024-04-01 10:29:46.508 | error | Caught by controller[0]:     npm install --os=linux --cpu=arm sharp
host.raspberrypi | 2024-04-01 10:29:46.508 | error | Caught by controller[0]: - Add platform-specific dependencies:
host.raspberrypi | 2024-04-01 10:29:46.507 | error | Caught by controller[0]:     See https://sharp.pixelplumbing.com/install#cross-platform
host.raspberrypi | 2024-04-01 10:29:46.507 | error | Caught by controller[0]: - Ensure your package manager supports multi-platform installation:
host.raspberrypi | 2024-04-01 10:29:46.485 | error | Caught by controller[0]:     yarn add sharp --ignore-engines
host.raspberrypi | 2024-04-01 10:29:46.484 | error | Caught by controller[0]:     npm install --include=optional sharp
host.raspberrypi | 2024-04-01 10:29:46.484 | error | Caught by controller[0]: - Ensure optional dependencies can be installed:
host.raspberrypi | 2024-04-01 10:29:46.484 | error | Caught by controller[0]: Possible solutions:
host.raspberrypi | 2024-04-01 10:29:46.484 | error | Caught by controller[0]: Error: Could not load the "sharp" module using the linux-arm runtime
host.raspberrypi | 2024-04-01 10:29:46.484 | error | Caught by controller[0]:   ^
host.raspberrypi | 2024-04-01 10:29:46.484 | error | Caught by controller[0]:   throw new Error(help.join('\n'));
host.raspberrypi | 2024-04-01 10:29:46.473 | error | Caught by controller[0]: /opt/iobroker/node_modules/sharp/lib/sharp.js:114
host.raspberrypi | 2024-04-01 10:29:44.650 | info | instance system.adapter.zigbee2mqtt.0 started with pid 4841
host.raspberrypi | 2024-04-01 10:29:14.450 | info | Restart adapter system.adapter.zigbee2mqtt.0 because enabled

**Zigbee2MQTT itself is working properly, the frontend as well.**

I've tried serveral things, the steps suggested in the log, downgrade of the adapter to some previous versions, downgrade of sharp, uninstalled and reinstalled the adapter, sudo apt-get update&upgrade, reboot, but nothing helped.

Any help will be appreciated.

### What did you expect to happen?

I expect the adapter to start

### How to reproduce it (minimal and precise)

--

### Zigbee2MQTT version

1.35.2

### Adapter firmware version

7.3.1.0 build 176

### Adapter

Sonoff Zigbee dongle E

### Setup

Installed directly on raspberry OS

### Debug log

host.raspberrypi | 2024-04-01 10:29:46.511 | info | Restart adapter system.adapter.zigbee2mqtt.0 because enabled
-- | -- | -- | --
host.raspberrypi | 2024-04-01 10:29:46.511 | error | instance system.adapter.zigbee2mqtt.0 terminated with code 1 (JS_CONTROLLER_STOPPED)
host.raspberrypi | 2024-04-01 10:29:46.510 | error | Caught by controller[0]: Node.js v20.12.0
host.raspberrypi | 2024-04-01 10:29:46.510 | error | Caught by controller[0]:     at Module._extensions..js (node:internal/modules/cjs/loader:1427:10)
host.raspberrypi | 2024-04-01 10:29:46.510 | error | Caught by controller[0]:     at Module._compile (node:internal/modules/cjs/loader:1369:14)
host.raspberrypi | 2024-04-01 10:29:46.510 | error | Caught by controller[0]:     at Object.<anonymous> (/opt/iobroker/node_modules/sharp/lib/constructor.js:10:1)
host.raspberrypi | 2024-04-01 10:29:46.509 | error | Caught by controller[0]:     at require (node:internal/modules/helpers:179:18)
host.raspberrypi | 2024-04-01 10:29:46.509 | error | Caught by controller[0]:     at Module.require (node:internal/modules/cjs/loader:1231:19)
host.raspberrypi | 2024-04-01 10:29:46.509 | error | Caught by controller[0]:     at Module._load (node:internal/modules/cjs/loader:1022:12)
host.raspberrypi | 2024-04-01 10:29:46.509 | error | Caught by controller[0]:     at Module.load (node:internal/modules/cjs/loader:1206:32)
host.raspberrypi | 2024-04-01 10:29:46.509 | error | Caught by controller[0]:     at Module._extensions..js (node:internal/modules/cjs/loader:1427:10)
host.raspberrypi | 2024-04-01 10:29:46.509 | error | Caught by controller[0]:     at Module._compile (node:internal/modules/cjs/loader:1369:14)
host.raspberrypi | 2024-04-01 10:29:46.508 | error | Caught by controller[0]:     at Object.<anonymous> (/opt/iobroker/node_modules/sharp/lib/sharp.js:114:9)
host.raspberrypi | 2024-04-01 10:29:46.508 | error | Caught by controller[0]:     See https://sharp.pixelplumbing.com/install
host.raspberrypi | 2024-04-01 10:29:46.508 | error | Caught by controller[0]: - Consult the installation documentation:
host.raspberrypi | 2024-04-01 10:29:46.508 | error | Caught by controller[0]:     npm install --os=linux --cpu=arm sharp
host.raspberrypi | 2024-04-01 10:29:46.508 | error | Caught by controller[0]: - Add platform-specific dependencies:
host.raspberrypi | 2024-04-01 10:29:46.507 | error | Caught by controller[0]:     See https://sharp.pixelplumbing.com/install#cross-platform
host.raspberrypi | 2024-04-01 10:29:46.507 | error | Caught by controller[0]: - Ensure your package manager supports multi-platform installation:
host.raspberrypi | 2024-04-01 10:29:46.485 | error | Caught by controller[0]:     yarn add sharp --ignore-engines
host.raspberrypi | 2024-04-01 10:29:46.484 | error | Caught by controller[0]:     npm install --include=optional sharp
host.raspberrypi | 2024-04-01 10:29:46.484 | error | Caught by controller[0]: - Ensure optional dependencies can be installed:
host.raspberrypi | 2024-04-01 10:29:46.484 | error | Caught by controller[0]: Possible solutions:
host.raspberrypi | 2024-04-01 10:29:46.484 | error | Caught by controller[0]: Error: Could not load the "sharp" module using the linux-arm runtime
host.raspberrypi | 2024-04-01 10:29:46.484 | error | Caught by controller[0]:   ^
host.raspberrypi | 2024-04-01 10:29:46.484 | error | Caught by controller[0]:   throw new Error(help.join('\n'));
host.raspberrypi | 2024-04-01 10:29:46.473 | error | Caught by controller[0]: /opt/iobroker/node_modules/sharp/lib/sharp.js:114
host.raspberrypi | 2024-04-01 10:29:44.650 | info | instance system.adapter.zigbee2mqtt.0 started with pid 4841
host.raspberrypi | 2024-04-01 10:29:14.450 | info | Restart adapter system.adapter.zigbee2mqtt.0 because enabled

ThorstenDtk, 2024-04-01:
P.S.: Downgrading js-controller from 5.0.19 to 5.0.17 helped mitigate the issue.

ernie-sys, 2024-04-20:
Any solution without downgrading?

closed 2024-11-17 by github-actions[bot]