Koenkk/zigbee2mqtt #22019: AVATTO 2gang switch does not show the backlight option

asherbersh, 2024-04-01:
### What happened?

Avatta 2gang switch does not show the backlight option it is shwoing only on/off switch in the smart life app this option is avilable and working .

### What did you expect to happen?

to allow to close , can do it from the smart things hub

### How to reproduce it (minimal and precise)

happens all the time

### Zigbee2MQTT version

1.36.0

### Adapter firmware version

20210708

### Adapter

snoof zigbee 3.0  dongle plus

### Setup

add on on home assistance

avbdr, 2024-04-08:
you can use following converter. replace _TZ3000_ogpla3lh with your device id
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
        fingerprint: [
            { modelID: 'TS0002', manufacturerName: '_TZ3000_ogpla3lh' },
        ],
        model: 'TS0002_switch_2_gang',
        vendor: 'TuYa',
        description: '2-Gang switch with backlight',
        extend: tuya.extend.switch({ powerOnBehavior2: true, backlightModeOffOn: true, endpoints: ['l1', 'l2'] }),
        endpoint: (device) => {
            return { 'l1': 1, 'l2': 2 };
        },
        meta: { multiEndpoint: true },
        configure: async (device, coordinatorEndpoint, logger) => {
            await tuya.configureMagicPacket(device, coordinatorEndpoint, logger);
            await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
            await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
        },
        whiteLabel: [
            tuya.whitelabel('RoomsAI', '37022463', '2 Gang switch with backlight', ['_TZ3000_ogpla3lh']),
        ],
};
module.exports = definition;
```

asherbersh, 2024-04-08:
Thanks a lot for this .
Do you have maybe a docmentation that explains how i use this convertor ?
Also do you plan to have it built in supported in future versions?

Koenkk, 2024-04-08:
@avbdr could you submit a pull request to include this?

asherbersh, 2024-04-08:
By the way i can see this solution is specific for two gang switch. But i
have the same problem on 1,3,4,5 gang switch the same vendor , same model .
What should i do then to support it ?

avbdr, 2024-04-08:
@Koenkk ready sir. https://github.com/Koenkk/zigbee-herdsman-converters/pull/7364

@asherbersh im sorry, I dont have 4 and 5 gang models, if you will send there device ids, I can make a test converter for them

asherbersh, 2024-04-08:
@avbdr i have created a new convertor from the code you provided saved it under avatto2g.js and uploaded to zigbee2mqtt folder 
created in the zibee2mqtt configuration file a new line with the following :
external_converters:
  - avatto2g.js

then restarted the zibeee2mqttt process but still the backlight switch was not exposed and can't be used :(

any idea why ?

avbdr, 2024-04-08:
@asherbersh send a screenshot of the device status page. Most likely your switches have different devices id. Avatto does lot of whitelabels

asherbersh, 2024-04-08:
@avbdr see attached
[image]

avbdr, 2024-04-08:
@asherbersh it seems that you saved a file in the root directory of the zigbee2mqtt. move it into data/ directory instead and restart. its the same directory where configuration.yaml is stored. 
If thats not the case, see the error you are getting on z2m startup. most likely it would say that something is wrong

asherbersh, 2024-04-08:
@avbdr looks like three is an error loading the file , see below:

[23:30:31] INFO: Preparing to start...
[23:30:32] INFO: Socat not enabled
[23:30:34] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:error 2024-04-08 23:30:40: Failed to load external converter file 'avatto2g.js' (Cannot read properties of undefined (reading 'switch'))
Zigbee2MQTT:error 2024-04-08 23:30:40: Probably there is a syntax error in the file or the external converter is not compatible with the current Zigbee2MQTT version
Zigbee2MQTT:error 2024-04-08 23:30:40: Note that external converters are not meant for long term usage, it's meant for local testing after which a pull request should be created to add out-of-the-box support for the device

also see file
[image]

asherbersh, 2024-04-08:
@avbdr / @Koenkk  - looks like i found why this convertor is not running . when i downgrade the zigbeewqtt to the previous version 1.36.0-1the error you see in my comments is not happening and the plugin works as expected . but when upgrading it to the latest version 1.36.1-1 the error appears and the convertor does not work

avbdr, 2024-04-08:
@asherbersh oh, you have to replace 'tuya.extend.switch' with 'tuya.modernExtend.tuyaOnOff' and it should be fine

asherbersh, 2024-04-08:
@avbdr - looks like it is still not working 

i get now the following error : 

[02:02:40] INFO: Preparing to start...
[02:02:40] INFO: Socat not enabled
[02:02:43] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:error 2024-04-09 02:02:49: Failed to load external converter file 'avatto2g.js' ('TS0002_switch_2_gang' has legacy extend which is not supported anymore)
Zigbee2MQTT:error 2024-04-09 02:02:49: Probably there is a syntax error in the file or the external converter is not compatible with the current Zigbee2MQTT version
Zigbee2MQTT:error 2024-04-09 02:02:49: Note that external converters are not meant for long term usage, it's meant for local testing after which a pull request should be created to add out-of-the-box support for the device
Zigbee2MQTT:info  2024-04-09 02:02:49: Logging to console and directory: '/config/zigbee2mqtt/log/2024-04-09.02-02-49' filename: log.txt

this is the updated code i changed:
[image]

avbdr, 2024-04-08:
@asherbersh sorry, 1st line have escaped from my copypaste . maybe thats why. try like that

```
const {} = require('zigbee-herdsman-converters/lib/modernExtend');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const {} = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [
        {modelID: 'TS0003', manufacturerName: '_TZ3000_ogpla3lh'},
    ],
    zigbeeModel: ['TS0003'],
    model: 'T3',
    vendor: '_TZ3000_ogpla3lh',
    description: '3-Gang switch with backlight',
    extend: [
        tuya.modernExtend.tuyaOnOff({ powerOnBehavior2: true, backlightModeOffOn: true, endpoints: ['l1', 'l2'] }),
    ],
    endpoint: (device) => {
        return { 'l1': 1, 'l2': 2};
    },
    configure: async (device, coordinatorEndpoint, logger) => {
        await tuya.configureMagicPacket(device, coordinatorEndpoint, logger);
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
    },
    meta: { multiEndpoint: true},
    whiteLabel: [
          tuya.whitelabel('RoomsAI', 'T3', '3 Gang switch with backlight', ['_TZ3000_ogpla3lh']),
    ],
};
module.exports = definition;

```

asherbersh, 2024-04-09:
@avbdr -thanks a lot for your help :)
seems that once i change to the new code you provided there is no error now ( see attached file ) but the backlight option is not exposed like it was before the upgrade to 1.36.1 ( see attached screenshot)
[log_9424_309.txt](https://github.com/Koenkk/zigbee2mqtt/files/14911986/log_9424_309.txt)
[image]

avbdr, 2024-04-09:
@asherbersh are you sure that you have included file in external_converters ? log show that it still recognizing switch as tuya

avbdr, 2024-04-09:
sorry, i have missed to replace ts0003 with ts0002 ... use this one :)
```
const {} = require('zigbee-herdsman-converters/lib/modernExtend');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const {} = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [
        {modelID: 'TS0003', manufacturerName: '_TZ3000_ogpla3lh'},
    ],
    zigbeeModel: ['TS0002'],
    model: 'T3',
    vendor: '_TZ3000_ogpla3lh',
    description: '3-Gang switch with backlight',
    extend: [
        tuya.modernExtend.tuyaOnOff({ powerOnBehavior2: true, backlightModeOffOn: true, endpoints: ['l1', 'l2'] }),
    ],
    endpoint: (device) => {
        return { 'l1': 1, 'l2': 2};
    },
    configure: async (device, coordinatorEndpoint, logger) => {
        await tuya.configureMagicPacket(device, coordinatorEndpoint, logger);
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
    },
    meta: { multiEndpoint: true},
    whiteLabel: [
          tuya.whitelabel('RoomsAI', 'T3', '3 Gang switch with backlight', ['_TZ3000_ogpla3lh']),
    ],
};
module.exports = definition;

```

asherbersh, 2024-04-09:
@avbdr - hello my friend , thank you very much now this is working great 
now does it means you need to change something in what you pulled request to the zigbee2mqtt code ?

closed 2024-11-06 by github-actions[bot]