Koenkk/zigbee2mqtt #22020: Fingerbot Plus (Adaprox TS0001_fingerbot_1) Arm Doesn't Retract

notownblues, 2024-04-01:
### What happened?

_This is a repost of issue #18138 as it went stale before other users experienced the same and I've been asked to re-open it._

I've noticed that after a couple of ON/OFF switches the arm gets stuck and doesn't fully retract anymore. 

This currently happens on 2 different units I own, in both CLICK and SWITCH modes.

The only way I found to "unstuck" the arm is to trigger the device again but this issue renders the device useless with automations.

### What did you expect to happen?

For the arm to retract properly and not get stuck

### How to reproduce it (minimal and precise)

1. Pair the device with Zigbee2MQTT
2. Press the switch a couples of times
3. The arm will get stuck

### Zigbee2MQTT version

1.36.0

### Adapter firmware version

20230507

### Adapter

SONOFF Dongle

### Setup

Addon

### Debug log

Debug 2023-06-28 16:34:48Publishing 'set' 'state' to 'Shower Fan Fingerbot' Info 2023-06-28 16:34:49MQTT publish: topic 'zigbee2mqtt/Shower Fan Fingerbot', payload '{"battery":100,"delay":0,"linkquality":43,"lower":60,"mode":"switch","program":{"0":255,"1":255,"2":255,"3":0},"reverse":"OFF","state":"ON","touch":"ON","upper":0}' Debug 2023-06-28 16:34:49Received Zigbee message from 'Shower Fan Fingerbot', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0 Info 2023-06-28 16:34:49MQTT publish: topic 'zigbee2mqtt/Shower Fan Fingerbot', payload '{"battery":100,"delay":0,"linkquality":43,"lower":60,"mode":"switch","program":{"0":255,"1":255,"2":255,"3":0},"reverse":"OFF","state":"ON","touch":"ON","upper":0}' Debug 2023-06-28 16:34:50Received MQTT message on 'zigbee2mqtt/Shower Fan Fingerbot/set' with data '{"state":"OFF"}' Debug 2023-06-28 16:34:50Publishing 'set' 'state' to 'Shower Fan Fingerbot' Debug 2023-06-28 16:34:52Received Zigbee message from 'Shower Fan Fingerbot', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0 Info 2023-06-28 16:34:52MQTT publish: topic 'zigbee2mqtt/Shower Fan Fingerbot', payload '{"battery":100,"delay":0,"linkquality":47,"lower":60,"mode":"switch","program":{"0":255,"1":255,"2":255,"3":0},"reverse":"OFF","state":"OFF","touch":"ON","upper":0}' Info 2023-06-28 16:34:52MQTT publish: topic 'zigbee2mqtt/Shower Fan Fingerbot', payload '{"battery":100,"delay":0,"linkquality":47,"lower":60,"mode":"switch","program":{"0":255,"1":255,"2":255,"3":0},"reverse":"OFF","state":"OFF","touch":"ON","upper":0}' Debug 2023-06-28 16:34:52Received MQTT message on 'zigbee2mqtt/Shower Fan Fingerbot/set' with data '{"state":"ON"}' Debug 2023-06-28 16:34:52Publishing 'set' 'state' to 'Shower Fan Fingerbot' Debug 2023-06-28 16:34:54Received Zigbee message from 'Shower Fan Fingerbot', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0 Info 2023-06-28 16:34:54MQTT publish: topic 'zigbee2mqtt/Shower Fan Fingerbot', payload '{"battery":100,"delay":0,"linkquality":43,"lower":60,"mode":"switch","program":{"0":255,"1":255,"2":255,"3":0},"reverse":"OFF","state":"ON","touch":"ON","upper":0}' Info 2023-06-28 16:34:54MQTT publish: topic 'zigbee2mqtt/Shower Fan Fingerbot', payload '{"battery":100,"delay":0,"linkquality":43,"lower":60,"mode":"switch","program":{"0":255,"1":255,"2":255,"3":0},"reverse":"OFF","state":"ON","touch":"ON","upper":0}' Debug 2023-06-28 16:34:55Received MQTT message on 'zigbee2mqtt/Shower Fan Fingerbot/set' with data '{"state":"OFF"}' Debug 2023-06-28 16:34:55Publishing 'set' 'state' to 'Shower Fan Fingerbot' Debug 2023-06-28 16:34:56Received Zigbee message from 'Shower Fan Fingerbot', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0 Info 2023-06-28 16:34:57MQTT publish: topic 'zigbee2mqtt/Shower Fan Fingerbot', payload '{"battery":100,"delay":0,"linkquality":43,"lower":60,"mode":"switch","program":{"0":255,"1":255,"2":255,"3":0},"reverse":"OFF","state":"OFF","touch":"ON","upper":0}' Info 2023-06-28 16:34:57MQTT publish: topic 'zigbee2mqtt/Shower Fan Fingerbot', payload '{"battery":100,"delay":0,"linkquality":43,"lower":60,"mode":"switch","program":{"0":255,"1":255,"2":255,"3":0},"reverse":"OFF","state":"OFF","touch":"ON","upper":0}' Debug 2023-06-28 16:34:57Received MQTT message on 'zigbee2mqtt/Shower Fan Fingerbot/set' with data '{"state":"ON"}' Debug 2023-06-28 16:34:57Publishing 'set' 'state' to 'Shower Fan Fingerbot' Info 2023-06-28 16:34:59MQTT publish: topic 'zigbee2mqtt/Shower Fan Fingerbot', payload '{"battery":100,"delay":0,"linkquality":40,"lower":60,"mode":"switch","program":{"0":255,"1":255,"2":255,"3":0},"reverse":"OFF","state":"ON","touch":"ON","upper":0}' Debug 2023-06-28 16:34:59Received Zigbee message from 'Shower Fan Fingerbot', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0 Info 2023-06-28 16:34:59MQTT publish: topic 'zigbee2mqtt/Shower Fan Fingerbot', payload '{"battery":100,"delay":0,"linkquality":43,"lower":60,"mode":"switch","program":{"0":255,"1":255,"2":255,"3":0},"reverse":"OFF","state":"ON","touch":"ON","upper":0}' Debug 2023-06-28 16:35:00Received MQTT message on 'zigbee2mqtt/Shower Fan Fingerbot/set' with data '{"state":"OFF"}' Debug 2023-06-28 16:35:00Publishing 'set' 'state' to 'Shower Fan Fingerbot' Info 2023-06-28 16:35:01MQTT publish: topic 'zigbee2mqtt/Shower Fan Fingerbot', payload '{"battery":100,"delay":0,"linkquality":43,"lower":60,"mode":"switch","program":{"0":255,"1":255,"2":255,"3":0},"reverse":"OFF","state":"OFF","touch":"ON","upper":0}' Debug 2023-06-28 16:35:01Received Zigbee message from 'Shower Fan Fingerbot', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0 Info 2023-06-28 16:35:01MQTT publish: topic 'zigbee2mqtt/Shower Fan Fingerbot', payload '{"battery":100,"delay":0,"linkquality":43,"lower":60,"mode":"switch","program":{"0":255,"1":255,"2":255,"3":0},"reverse":"OFF","state":"OFF","touch":"ON","upper":0}' Debug 2023-06-28 16:35:02Received MQTT message on 'zigbee2mqtt/Shower Fan Fingerbot/set' with data '{"state":"ON"}' Debug 2023-06-28 16:35:02Publishing 'set' 'state' to 'Shower Fan Fingerbot' Info 2023-06-28 16:35:04MQTT publish: topic 'zigbee2mqtt/Shower Fan Fingerbot', payload '{"battery":100,"delay":0,"linkquality":43,"lower":60,"mode":"switch","program":{"0":255,"1":255,"2":255,"3":0},"reverse":"OFF","state":"ON","touch":"ON","upper":0}' Debug 2023-06-28 16:35:04Received Zigbee message from 'Shower Fan Fingerbot', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0 Info 2023-06-28 16:35:04MQTT publish: topic 'zigbee2mqtt/Shower Fan Fingerbot', payload '{"battery":100,"delay":0,"linkquality":47,"lower":60,"mode":"switch","program":{"0":255,"1":255,"2":255,"3":0},"reverse":"OFF","state":"ON","touch":"ON","upper":0}' Debug 2023-06-28 16:35:04Received MQTT message on 'zigbee2mqtt/Shower Fan Fingerbot/set' with data '{"state":"OFF"}' Debug 2023-06-28 16:35:04Publishing 'set' 'state' to 'Shower Fan Fingerbot' Debug 2023-06-28 16:35:06Received Zigbee message from 'Shower Fan Fingerbot', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0 Info 2023-06-28 16:35:06MQTT publish: topic 'zigbee2mqtt/Shower Fan Fingerbot', payload '{"battery":100,"delay":0,"linkquality":43,"lower":60,"mode":"switch","program":{"0":255,"1":255,"2":255,"3":0},"reverse":"OFF","state":"OFF","touch":"ON","upper":0}' Info 2023-06-28 16:35:06MQTT publish: topic 'zigbee2mqtt/Shower Fan Fingerbot', payload '{"battery":100,"delay":0,"linkquality":43,"lower":60,"mode":"switch","program":{"0":255,"1":255,"2":255,"3":0},"reverse":"OFF","state":"OFF","touch":"ON","upper":0}'

SimonMayerhofer, 2024-09-29:
Keep open

4rianton, 2024-10-22:
I am having this issue as well.

Venef1x, 2024-12-10:
I am also having this issue. Keep open.

basvdl, 2025-01-01:
Same issue with all 3 off my fingerbots. My work around is currently working for over a week, roughly 40+ clicks so far.

After every state change to 'off', I switch the mode to 'program', you will hear the motor turn on for a short moment.

Than after 5 sec, switch it back to click. Again you hear the motor turn on for a short moment. Optional, I set the down movement to 100.

You can only do this when the state of the device changes to 'off', since switching from 'click' to 'program', will reset the state to 'off'.

In my case when the custom dehumidifier device goes to idle or off, I'll trigger the automation to 'recalibrate' the fingerbot.

[image]

4rianton, 2025-01-01:
My specific issues were fixed by replacing the batteries that the fingerbot shipped with with brand new batteries.

notownblues, 2025-01-10:
Following @4rianton's comment and SwitchBot's unreliability (the string would break once/month in switch mode or bots would be getting weak all of a sudden, even with brand new batteries) I wanted to give the Fingerbot another shot.

I'm happy to report that after swapping the battery by a Duracell one and other "fixes" it's been working great. 

I put the bot on a kitchen switch which has 50+ triggers/day and it hasn't got stuck once in a couple of days now. 

Here's what I did to achieve that
- Put a Duracell battery
- Ensure that the bot has the proper lower/upper settings (I had to play with that for quite a bit to find the right values. The bot should not be pushing too hard or not enough)
- I noticed the double tape was getting week at the top of the bot, where the most pressure is so decided to put hot glue and I think it really helped as it would cause the arm not to push properly before and get stuck
[image]
[image]

I think a combination of all those things really help and encourage anyone having this issue to try.

I also noticed they release a charging version (without batteries) recently so have one of those on the way.

closed 2025-03-20 by github-actions[bot]