Koenkk/zigbee2mqtt #22028: Frontend dashboad switches do not work in last version

SubToaD, 2024-04-01:
### What happened?

Hello,
Socket and bulb switches do not work in the frontend dashboard (they activate but no action, returns to the origin with a refresh), but works in the expose state of each device

### What did you expect to happen?

they worked in the previous version 1.36.0

### Zigbee2MQTT version

1.36.1-dev commit: 5077b03

### Adapter firmware version

7.4.1 [GA]

### Adapter

SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2

### Setup

RPi4 Debian11 Docker container

### Debug log

my configuration.yaml

homeassistant: false
permit_join: false
frontend: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://192.168.1.48:1883
  include_device_information: true
  keepalive: 60
  reject_unauthorized: false
  version: 4
availability:
  active:
    timeout: 10
  passive:
    timeout: 1500
serial:
  port: /dev/ttyACM0
  adapter: ember
advanced:
  homeassistant_legacy_entity_attributes: false
  legacy_api: false
  legacy_availability_payload: false
  log_syslog:
    app_name: Zigbee2MQTT
    eol: /n
    host: localhost
    localhost: localhost
    path: /dev/log
    pid: process.pid
    port: 514
    protocol: udp4
    type: '5424'
  last_seen: ISO_8601_local
  log_level: debug
  output: attribute_and_json
  log_rotation: true
  cache_state: true
  cache_state_persistent: true
  cache_state_send_on_startup: true
  log_symlink_current: true
device_options:
  legacy: false

ivanfrolovmd, 2024-04-02:
There's this error in console when switch is manually triggered:
```
Uncaught TypeError: a is not a function
    onChange http://smarthome.local:8080/assets/Dashboard-cc086107.js:1
    onChange http://smarthome.local:8080/assets/DashboardFeatureWrapper-ef543cb2.js:1
    onChange http://smarthome.local:8080/assets/DashboardFeatureWrapper-ef543cb2.js:1
    onChange http://smarthome.local:8080/assets/DashboardFeatureWrapper-ef543cb2.js:1
    i http://smarthome.local:8080/assets/DashboardFeatureWrapper-ef543cb2.js:1
    xye http://smarthome.local:8080/assets/index-fb4a4614.js:40
    Hye http://smarthome.local:8080/assets/index-fb4a4614.js:40
    jye http://smarthome.local:8080/assets/index-fb4a4614.js:40
    qC http://smarthome.local:8080/assets/index-fb4a4614.js:40
    EH http://smarthome.local:8080/assets/index-fb4a4614.js:40
    Hg http://smarthome.local:8080/assets/index-fb4a4614.js:40
    Lw http://smarthome.local:8080/assets/index-fb4a4614.js:43
    JR http://smarthome.local:8080/assets/index-fb4a4614.js:40
    Hg http://smarthome.local:8080/assets/index-fb4a4614.js:40
    ow http://smarthome.local:8080/assets/index-fb4a4614.js:40
    eke http://smarthome.local:8080/assets/index-fb4a4614.js:40
    DH http://smarthome.local:8080/assets/index-fb4a4614.js:40
    Rg http://smarthome.local:8080/assets/index-fb4a4614.js:40
    l http://smarthome.local:8080/assets/index-fb4a4614.js:40
    $l http://smarthome.local:8080/assets/index-fb4a4614.js:40
    createRoot http://smarthome.local:8080/assets/index-fb4a4614.js:43
    <anonymous> http://smarthome.local:8080/assets/index-fb4a4614.js:217
Dashboard-cc086107.js:1:3113
```

cracyfloyd, 2024-04-02:
Here the same.
with Conbee 3 and Debian 12 VM on MacMini (zigbee2mqtt Docker).

closed 2024-04-04 by SubToaD