Koenkk/zigbee2mqtt #22029: [New device support]: Heiman PIR

aelsharawi, 2024-04-01:
### Link

https://www.alibaba.com/product-detail/Heiman-Ceiling-embedded-installation-tuya-zigbee_1600736227827.html

### Database entry

HS8OS-EF1-3.0

### Comments

HS8OS-EF1-3.0

I was able to make it work (basic functions) need help get the full potentials

here is the data sheet from the manfucutrer

[HS8OS-EF-.pdf](https://github.com/Koenkk/zigbee2mqtt/files/14829559/HS8OS-EF-.pdf)

### External definition

```shell
const {illuminance, occupancy, identify} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['HS8OS-EF1-3.0'],
    model: 'HS8OS-EF1-3.0',
    vendor: 'HEIMAN',
    description: 'Automatically generated definition',
    extend: [illuminance({"0x0400":["1"]}), occupancy({"0x0406":["1"]}), identify({"0x0003":["1"]})],
    meta: {},
};

module.exports = definition;
```

aelsharawi, 2024-04-04:
any advise>?

aelsharawi, 2024-04-05:
Where can I get this help?

mrskycriper, 2024-04-11:
I'll look into this. Data-sheet is a gold mine of information.

mrskycriper, 2024-04-11:
Also, can you please provide `database.db` entry for a good measure?

aelsharawi, 2024-04-11:
how can I get the .db file please?

aelsharawi, 2024-04-11:
thanks a lot for help ...is this enough?

mrskycriper, 2024-04-11:
> thanks a lot for help ...is this enough?

It should be, thanks

mrskycriper, 2024-04-15:
Sorry, I'm a bit overloaded lately and kinda just forgot

aelsharawi, 2024-04-15:
Ok. No worries

aelsharawi, 2024-04-20:
Hi again...just a gentle reminder

mrskycriper, 2024-04-21:
> Hi again...just a gentle reminder

Hi, thanks for the reminder again. Is there any official model name and device name on the box? Heiman is not particularity good at documenting their devices. Also, can you confirm that it looks like this random picture I've found on aliexpress listing?

[image]

aelsharawi, 2024-04-21:
https://www.alibaba.com/product-detail/Heiman-Ceiling-embedded-installation-tuya-zigbee_1600736227827.html

yes it the correct picture above is the link where I bough it

thanks for the help

closed 2024-11-18 by github-actions[bot]