Koenkk/zigbee2mqtt #22030: IKEA Tradfri (LED2103G5) turns on when receiving off command if already off

ianmtaylor1, 2024-04-02:
### What happened?

Sorry for the grammatically confusing title, hopefully it will be clear below.

### What did you expect to happen?

If the IKEA bulb is off, and I publish `{"state": "OFF"}` to `zigbee2mqtt/My-Bulb-Name/set`, the light will remain off.

### How to reproduce it (minimal and precise)

1. Add IKEA Tradfri bulb (LED2103G5) to Zigbee2MQTT.
2. Update firmware to version 16777270 (Not sure if this is necessary or if this bug exists on old firmware.)
3. Turn bulb on.
4. Publish `{"state": "OFF"}` to `zigbee2mqtt/My-Bulb-Name/set`. **Bulb turns off as expected.**
5. Repeat step 4, publish another off message while bulb is already off. **Bulb does not remain off, instead, it turns on at very low brightness.**

### Zigbee2MQTT version

1.36.0 commit: 86ed71c

### Adapter firmware version

20210708

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

### Setup

Official docker image on Raspberry Pi 3B+

efab2022, 2024-04-02:
+1 I came to report the exact same behaviour for this other IKEA device [LED2005R5_LED2106R3](https://www.zigbee2mqtt.io/devices/LED2005R5_LED2106R3.html). 

This issue also occurs when sending `{"brightness": 0}` repeatedly (the bulb toggles between off and on at low power).
This issue also occurs when sending `{"brightness": 1}` repeatedly (the bulb toggles between off and on at low power).
This issue does not occur for brightness 2 or greater (the bulb stays on)

In some circumstances, doing this repeatedly 

tx [ianmtaylor1](https://github.com/ianmtaylor1)

ianmtaylor1, 2024-04-02:
I want to determine whether this is a bug with zigbee2mqtt or the IKEA devices themselves. I could pair the lightbulb with an IKEA hub coordinator, but I don't own one. I do have an [IKEA somfrig button](https://www.ikea.com/us/en/p/somrig-remote-control-white-smart-30560354/). Would [binding](https://www.zigbee2mqtt.io/guide/usage/binding.html) those two devices bypass zigbee2mqtt and allow me to test whether the same problem exists when hitting the off button twice?

ianmtaylor1, 2024-04-05:
I have one non-update and one semi-update to this topic:

**The non-update**: I haven't had any luck binding my button to my bulb so I haven't been able to determine if the issue is with Zigbee2MQTT or with the bulb. I suspect if I had a Tradfri button instead of a Somrig button it would work better. If anyone else could test this kind of bulb outside of Zigbee2MQTT to see if the problem persists, that would be greatly appreciated.

**The semi-update**

While I haven't been able to resolve the issue, I have resolved an associated issue that has helped in my specific circumstances. In my case, these double-off messages were mostly occurring from an automation in Home Assistant triggered by my Somrig button, which for some reason was firing twice when pressed. I found out this was due to my having configured
```
advanced:
  output: 'attribute_and_json'
```
in configuration.yaml. This was colliding *device automations* published to `zigbee2mqtt/my-button/action` with *states*, which, thanks to that configuration, were also being published to the same topic. I have set this option back to `'json'` and my double-firing automation is fixed. I wish this conflict between attribute publishing and device_automations was better documented.

To be clear, though, the underlying problem with the bulb is still there. Following the steps outlined in this issue's first post will still result in the bulb turning back on at low brightness, contrary to expectation. I've just found a mitigation for my particular main use case.

Furegato, 2024-05-11:
I'm having the same problem, but with IKEA's LED2111G6. (E14, RGB light bulb)
For the past month or so whenever I got back home I would found that the two LED2111G6 bulbs I have  were turned on at 1% brightness.
It was hard to track what happened because it was showing that their state were off ... But after some digging I found out that my "Away from Home' automation was sending a "turn off" command which was triggering the problem you mention as well.
Basically, yes, if I send a "turn lights off" command while they are off, they will turn back on at 1%  brightness.
One small finding... It will happen only once on a sequence of repeatedly "turn off" commands. For example: if I send another "turn off" while it is in this weird off-but-on state, it will turn it off and if I then send yet another "turn off" command, it will not present the same problem again. I know that sending 3 times a command to the light is not a viable workaround, but it might be a clue to what is happening.
I hope someone with the skills can figure it out.
It's really annoying not to be sure if you really turned off the lights. It kinda of defeats one of the most important features of a smart light.
Cheers.

TEF2one, 2024-05-28:
Same issue, was wondering why they would randomly turn on yet not show as turned on and finally noticed the relation with the off command waking them up... such a annoying issue after having replace all my light with these, I hope to find a solution.

TEF2one, 2024-05-29:
Yes I did think of that, but that won't work for groups, all my lights are in groups, I even have a global group in HA, so when I leave lights all turn off...

I figured out that this happens if you ever turn a light off twice. On the second time around, it will be set to 1%.

This happens if you use Apple HomeKit scenes as it will send a turn off command even if the device is already off.

I swapped to HomeAssistant scripts (better than scenes) and put in an if condition to check if the light is on before triggering an off.

This solved my issue and I haven't had any issues since.

TEF2one, 2024-06-15:
FYI, I returned mine to IKEA, they are good with that...
Instead I baught back some [LEDVANCE SMART+ Classic A60 E27 Tunable white](https://www.zigbee2mqtt.io/devices/4058075208384.html#ledvance-4058075208384)

Koenkk, 2024-06-15:
@ianmtaylor1 could you provide the debug log of this with z2m 1.38.0?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

ianmtaylor1, 2024-06-15:
@Koenkk Thanks for your reply. I've attached a snip of the debug log file while experiencing the issue. Please let me know if I've omitted anything or if you need any other information.

Z2m Version: 1.38.0 commit  [fe048e6](https://github.com/Koenkk/zigbee2mqtt/commit/fe048e6)
Tradfri firmware version: 1.0.36, build date 20230918
Device name: Ian's Bedside Lamp

Logs start with a command to turn the light on (line 1, [2024-06-15 12:23:55]), followed by a command to turn the light off which works as expected (line 30, [2024-06-15 12:24:07]), then a second command to turn the light off which reproduces the issue (line 81, [2024-06-15 12:24:11]).

Hope this helps!

[debug_log_snip.txt](https://github.com/user-attachments/files/15849387/debug_log_snip.txt)

ianmtaylor1, 2024-06-16:
> One small finding... It will happen only once on a sequence of repeatedly "turn off" commands. For example: if I send another "turn off" while it is in this weird off-but-on state, it will turn it off and if I then send yet another "turn off" command, it will not present the same problem again.

This is interesting, and different from the behavior of my Tradfri bulb. If I send repeated off commands, the bulb will alternate off, 1%, off, 1%, ... etc. repeatedly.

Furegato, 2024-06-17:
> What you're describing is exactly what is described here -
> The fix was to add an if check to detect if the bulb is ON, before sending

Unfortunately, your suggestion is not a real solution....
The problem is, let's say you are going to sleep and want to make sure all
lights are off... You would say to Google Home/Alexa/Siri... "Turn all
lights off"... Then those pesky lights would actually turn on and worst of
all, it does not show on their status... So you would have them on the
whole night. Or in case(like me) you have a presence automation by an
"external" assistant to when you leave home... You would have to treat
those lights independently. Anyway, a complete unnecessary nuisance to have
to deal.

tungmeister, 2024-06-20:
also seeing this behavior, it's really annoying as I've 10 GU10 bulbs that come on differently depending on whats happening in the room so I'll often find half of them on at 1% when they're meant to be all off.

Koenkk, 2024-06-22:
In your `configuration.yaml` (or `devices.yaml` if present), do you have a `transition` set for this device?

Koenkk, 2024-06-22:
Can you try to completely remove it?

ianmtaylor1, 2024-06-22:
> In your `configuration.yaml` (or `devices.yaml` if present), do you have a `transition` set for this device?

> Can you try to completely remove it? (stop z2m, remove transition from the configuration.yaml, start z2m)

I had `transition: 0` set in my `devices.yaml` for this device. I removed it, following your instructions, and the problem is no longer there. Publishing `{"state": "off"}` twice in a row to `zigbee2mqtt/Ian's Bedside Lamp/set` leaves the light off as expected.

Just FYI, now the light transitions for ~1sec between the on and off states (default transition time?). This is very minor and acceptable to me because it fixes the other problem.

Furegato, 2024-06-22:
> Can you try to completely remove it? (stop z2m, remove `transition` from the `configuration.yaml`, start z2m)

That actually worked for me as well!
Thanks for the suggestion!
Just setting the transition to 0 was making it even worse, but by deleting it completely from z2mqtt devices.yaml worked like a charm!

heyiswas, 2024-06-23:
> Can you try to completely remove it? (stop z2m, remove `transition` from the `configuration.yaml`, start z2m)

Works for LED2201G8. Thanks :)

tungmeister, 2024-06-25:
Unfortunately, this isn't a proper fix. I'm using the transition value within my groups.yaml, without defining it here groups hard cut on/off. Looking through the posts this appears to be affecting multiple ikea models and FW versions (I'm on 3.0.10 and I've seen mention of 1.0.36).  I'm guessing this is down to the new variants that have been introduced over the last 6 months or so, I'm seeing this with LED2106R3 but not LED2005R5. I've had the latter for a couple of years now without issue, the problem is only present on LED2106R3, the new version of this bulb. @Koenkk could this be a converter issue?

Koenkk, 2024-06-25:
If you remove the transition, instead of a "move to brightness 0 with transition 0", it will send the native off command

ianmtaylor1, 2024-06-26:
Thanks for clarifying! Do you think a note about this side-effect should be included in the documentation page for each affected device (i.e., [here](https://www.zigbee2mqtt.io/devices/LED2103G5.html#options))?

If so, could you clarify the default value for `transition`? It's stated that it's zero. I don't know if that's true, though, because the behavior with `transition: 0` is different than with `transition` unspecified, and with `transition` unspecified there is a noticeable transition time.

I'd be happy to make a pull request.

As an aside, I performed a few more tests that others are welcome to also replicate:

- The bug also occurs with no `transition` set in `devices.yaml`, but with publishing `{"state":"off", "transition":0}`.
- Nonzero transition times also exhibit the bug, but slightly differently. My bulb will only unexpectedly turn on *once*, with repeated `{"state":"off"}` commands sent. With a 0 second transition, my bulb would repeatedly cycle on and off.

Koenkk, 2024-06-27:
Does it work when leaving the `transition` empty in the frontend? e.g.

[image]

Midifreakz, 2024-06-27:
@Koenkk  leaving it blank in the frontend does not help. there is no entry in config.yaml when i leave it blanc, but the lights still seem to respond to my previously set delay.

Furegato, 2024-06-27:
> Does it work when leaving the `transition` empty in the frontend? e.g.
> 
> <img alt="Screenshot 2024-06-27 at 10 56 39" width="648" src="https://private-user-images.githubusercontent.com/2892853/343703233-99938369-717a-44dd-acff-8e1a5ce69ecb.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTk0ODg5NjIsIm5iZiI6MTcxOTQ4ODY2MiwicGF0aCI6Ii8yODkyODUzLzM0MzcwMzIzMy05OTkzODM2OS03MTdhLTQ0ZGQtYWNmZi04ZTFhNWNlNjllY2IucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDYyNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA2MjdUMTE0NDIyWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZjgzNzk2YmIxODBlYzEyMjRiYzVkY2M1NjcwMzk2ZWNkYjJlZmI4ODNlZWMxNWVlMjk1NWQ2MDczMDA1OGUyZCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.1zhXH76WgXSFKpH_24Mlg9fWn1wgrT889PGhi3GYHTw">

Yes!
I can confirm that leaving the "transition" field empty and confirming it fixes the problem since it completely removes the "transition=X" line from the devices.yaml
Cheers!

Midifreakz, 2024-06-27:
Maybe some extra information. I only seem to have this problem wen utilising the off command in the scope of a larger automation. When I just send the one command twice. The lights just stay off.

Furegato, 2024-06-27:
Just to be sure, I've double-checked it and it seems to fix it on my side if I leave the device's zigbee2mqtt transition field empty. With transition=0 and 1 I had the erroneous behaviour, with the transition field empty the problem is gone either by calling "lights off" individually or in a group. It only causes problems if I have a send "turn off" command with any transition value data, as other people have already mentioned.
Model: LED2111G6

Midifreakz, 2024-06-27:
I am not able to remove the transition even though the field in config.yaml is gone. If anyone is using Z2M in Home Assistant, creating an automation like the screenshot below is a good tem fix that ensures your light is always off until this issue is resolved.

[image]

tungmeister, 2024-07-11:
My bulbs are all showing an update, currently doing he OTA,  will see if it makes any difference.

alex3305, 2024-07-25:
I also experience this as mentioned by @Koenkk in the referenced (and now closed) issue with [IKEA LED1842G3](https://www.zigbee2mqtt.io/devices/LED1842G3.html) after updating to the newest [1.1.4](https://ww8.ikea.com/ikeahomesmart/releasenotes/releasenotes.html) firmware. Ironically I had set a numerical value for `transition` to mitigate or workaround #19211.

stevelee916a, 2024-08-11:
Seems like this has been a small but recurring issue for years? Skimming through the threads, it’s not clear to me whether this is a homebridge, ha, or z2mqtt issue

Homebridge bug thread (closed but not resolved): https://github.com/homebridge/homebridge/issues/3073

homebridge-z2mqtt bug thread (also closed but not resolved): https://github.com/itavero/homebridge-z2m/issues/383

zigbee2mqtt bug thread (closed but not resolved): https://github.com/Koenkk/zigbee2mqtt/issues/10409

Koenkk, 2024-08-12:
@GoodnessJSON 

> If you manage them directly from Z2M you get the issue when transition is set (even if 0).

You should not set any `transition` (with `0` you will get this issue)

Koenkk, 2024-08-13:
I've fixed the issue, the transition will now always be ignored for the following models: LED2103G5, LED1842G3 and LED2201G8 (let me know if this fix should be applied to more)

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

ipha, 2024-08-13:
> I've fixed the issue, the transition will now always be ignored for the following models: LED2103G5, LED1842G3 and LED2201G8 (let me know if this fix should be applied to more)
>
> Changes will be [...]

LED2107C4 and LED1949C5 (on fw 1.1.20 but not 1.1.003.) have the same issue.

Midifreakz, 2024-08-14:
Hi Koen, 

The issue is also happening on the GU10 bulbs.
I can testify for the newer models

LED1923R5
LED2005R5/LED2106R3
LED2104R3

Koenkk, 2024-08-15:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

tungmeister, 2024-08-15:
@Koenkk  will this affect all versions of the LED2005R5/LED2106R3? I have no issues with the transition settings on the original version of these bulbs however it's broken on the newer revision. 
Original model:
[image]
new version:
[image]

tungmeister, 2024-08-30:
@Koenkk sorry to bump this but I want to make sure these changes won't break all the transition settings I've got on the older version of the bulb (I have a lot of them!) in the next release.

Koenkk, 2024-08-31:
@tungmeister thanks for bringing this up. The off transition will now only be suppressed with firmware > 1.0.012 

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

supaeasy, 2024-09-05:
Hello @Koenkk please also add https://www.zigbee2mqtt.io/devices/LED2003G10.html#ikea-led2003g10

On another note, maybe this helps with the issue: it happens MUCH more with ember (multiple times a day) than it did with ezsp (a couple times per week)

Koenkk, 2024-09-05:
@supaeasy the fix is applied to all TRADFRI bulb with `softwareBuildID > 1.0.012`. I think this issue can be closed now.

closed 2024-09-05 by Koenkk

matnoublanche, 2024-11-15:
Hi,

Sorry to reopen this issue.

@Koenkk : I also have the same problem with this Ikea bulb : LED2109G6 (https://www.zigbee2mqtt.io/devices/LED2109G6.html#ikea-led2109g6)

Could you add it please ?

Thank's ! 

Mathieu

Koenkk, 2024-11-17:
@matnoublanche if it does not work with z2m 1.41.0, can you provide the data/database.db entry?

eirikc, 2024-11-19:
@Koenkk I'm having this issue on z2m 1.41.0 with **LED1937T5** on fw **1.0.012**, The issue is resolved by removing the transition I set under global device_options in the config file. If 1.41.0 is supposed use noOffTransition=true for LED1937T5, then maybe the db entries are helpful:

```json
{"id":9,"type":"Router","ieeeAddr":"0x842e14fffe4852e3","nwkAddr":14120,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRIbulbT120E27WSopal470lm","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,4096,64599],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"TRADFRIbulbT120E27WSopal470lm","manufacturerName":"IKEA of Sweden","powerSource":1,"zclVersion":3,"appVersion":16,"stackVersion":103,"hwVersion":1,"dateCode":"20201111","swBuildId":"1.0.012"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":16,"colorTempPhysicalMin":250,"colorTempPhysicalMax":454,"colorMode":2,"colorTemperature":350,"startUpColorTemperature":65535,"currentX":33107,"currentY":27211}},"genOnOff":{"attributes":{"onOff":0}},"genLevelCtrl":{"attributes":{"currentLevel":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b78e4c","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b78e4c","endpointID":1},{"cluster":768,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b78e4c","endpointID":1}],"configuredReportings":[{"cluster":8,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":1},{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0},{"cluster":768,"attrId":7,"minRepIntval":5,"maxRepIntval":3600,"repChange":1}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":16,"stackVersion":103,"hwVersion":1,"dateCode":"20201111","swBuildId":"1.0.012","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1732056686990}
{"id":10,"type":"Router","ieeeAddr":"0x842e14fffe4f7850","nwkAddr":5420,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRIbulbT120E27WSopal470lm","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,4096,64599],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"TRADFRIbulbT120E27WSopal470lm","manufacturerName":"IKEA of Sweden","powerSource":1,"zclVersion":3,"appVersion":16,"stackVersion":103,"hwVersion":1,"dateCode":"20201111","swBuildId":"1.0.012"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":16,"colorTempPhysicalMin":250,"colorTempPhysicalMax":454,"colorMode":2,"colorTemperature":454,"startUpColorTemperature":65535,"currentX":30015,"currentY":26870}},"genOnOff":{"attributes":{"onOff":0,"startUpOnOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":3}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b78e4c","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b78e4c","endpointID":1},{"cluster":768,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b78e4c","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0,"manufacturerCode":null},{"cluster":8,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":1,"manufacturerCode":null},{"cluster":768,"attrId":7,"minRepIntval":5,"maxRepIntval":3600,"repChange":1,"manufacturerCode":null},{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0},{"cluster":8,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":1},{"cluster":768,"attrId":7,"minRepIntval":5,"maxRepIntval":3600,"repChange":1}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":16,"stackVersion":103,"hwVersion":1,"dateCode":"20201111","swBuildId":"1.0.012","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1732056830374}
```

Thanks for your kick ass work!

matnoublanche, 2024-11-20:
> @matnoublanche if it does not work with z2m 1.41.0, can you provide the data/database.db entry?

By bad, I was on version edge 1.36.0. I was supposing that edge version upgrades are done automatically.

I've upgraded to the last version, will see if all is ok for now.

Thank you

Koenkk, 2024-11-21:
@eirikc in this case transitions should be supported as it's running `1.0.012`, at least this is the feedback I got from https://github.com/Koenkk/zigbee2mqtt/issues/22030#issuecomment-2292063140

baskuipers93, 2025-08-05:
The issues might be back with a newer firmware on the IKEA bulbs. I have some LED1935C3 bulbs on latest IKEA firmware 1.1.4 (20240117). Running latest stable versions of Home Assistant and Z2M. 

My bulbs are in a group that have a transition set to 0.5. When the bulb is off and I trigger the lights-off action on the group this bulb is also present in, it turns physically on (with a low brightness) but the reported state in Z2M still says OFF. 

Removing the transition value on the groups fixes the issue, but I do like having a transition of 0.5 to smoothen out every brightness and color temperature changes.