Koenkk/zigbee2mqtt #22033: Brightness control not working properly in 1.36.1-1

dpgh947, 2024-04-02:
### What happened?

I have a smart knob which I can use to control the brightness of an LED strip (MiBoxer FUT035Z+ controller) using, for example, a light turn_on call from mode red with {"brightness_step_pct":-10} when the knob is turned down.

When I installed 1.36.1-1 just now, I noticed it flashed up a message in Z2M saying the strip light was being reconfigured, don't know what that was about. I tested the light and then found that when I tried to turn down the brightness with the knob, it would reduce slightly and then keep flicking back up to maximum brightness. When I restore back to 1.36.0-1, it works again.

### Zigbee2MQTT version

1.36.1-1

### Adapter firmware version

20210708

### Adapter

Sonoff 3.0 dongle-P

### Setup

addon on HA on Pi 4

D3SL, 2024-04-17:
Having this exact same issue. Below are logs of attempting to turn the light on and then off with brightness steps of 25% in HA:

```
debug 2024-04-17 19:24:17Received MQTT message on 'zigbee2mqtt/Kitchen FCOB CCT/set' with data '{"state":"ON","brightness":64}'
info 2024-04-17 19:24:17MQTT publish: topic 'zigbee2mqtt/Kitchen FCOB CCT', payload '{"brightness":64,"color":{"h":0,"hue":0},"color_mode":"color_temp","color_temp":377,"do_not_disturb":null,"last_seen":"2024-04-17T19:24:26+03:00","linkquality":92,"state":"OFF"}'
info 2024-04-17 19:24:17MQTT publish: topic 'zigbee2mqtt/Kitchen FCOB CCT', payload '{"brightness":64,"color":{"h":0,"hue":0},"color_mode":"color_temp","color_temp":377,"do_not_disturb":null,"last_seen":"2024-04-17T19:24:26+03:00","linkquality":92,"state":"ON"}'
info 2024-04-17 19:24:17MQTT publish: topic 'zigbee2mqtt/Kitchen FCOB CCT', payload '{"brightness":64,"color":{"h":0,"hue":0},"color_mode":"color_temp","color_temp":377,"do_not_disturb":null,"last_seen":"2024-04-17T19:24:26+03:00","linkquality":96,"state":"ON"}'
info 2024-04-17 19:24:17MQTT publish: topic 'zigbee2mqtt/Kitchen FCOB CCT', payload '{"brightness":64,"color":{"h":0,"hue":0},"color_mode":"color_temp","color_temp":377,"do_not_disturb":null,"last_seen":"2024-04-17T19:24:26+03:00","linkquality":96,"state":"ON"}'
info 2024-04-17 19:24:17MQTT publish: topic 'zigbee2mqtt/Kitchen FCOB CCT', payload '{"brightness":64,"color":{"h":0,"hue":0},"color_mode":"color_temp","color_temp":377,"do_not_disturb":null,"last_seen":"2024-04-17T19:24:26+03:00","linkquality":96,"state":"ON"}'
info 2024-04-17 19:24:17MQTT publish: topic 'zigbee2mqtt/Kitchen FCOB CCT', payload '{"brightness":64,"color":{"h":0,"hue":0},"color_mode":"color_temp","color_temp":377,"do_not_disturb":null,"last_seen":"2024-04-17T19:24:26+03:00","linkquality":96,"state":"ON"}'
info 2024-04-17 19:24:17MQTT publish: topic 'zigbee2mqtt/Kitchen FCOB CCT', payload '{"brightness":252,"color":{"h":0,"hue":0},"color_mode":"color_temp","color_temp":377,"do_not_disturb":null,"last_seen":"2024-04-17T19:24:26+03:00","linkquality":96,"state":"ON"}'
info 2024-04-17 19:24:17MQTT publish: topic 'zigbee2mqtt/Kitchen FCOB CCT', payload '{"brightness":252,"color":{"h":0,"hue":0},"color_mode":"color_temp","color_temp":377,"do_not_disturb":null,"last_seen":"2024-04-17T19:24:26+03:00","linkquality":92,"state":"ON"}'
debug 2024-04-17 19:24:18Received MQTT message on 'zigbee2mqtt/Kitchen FCOB CCT/set' with data '{"state":"ON","brightness":188}'
info 2024-04-17 19:24:19MQTT publish: topic 'zigbee2mqtt/Kitchen FCOB CCT', payload '{"brightness":252,"color":{"h":0,"hue":0},"color_mode":"color_temp","color_temp":377,"do_not_disturb":null,"last_seen":"2024-04-17T19:24:27+03:00","linkquality":92,"state":"ON"}'
info 2024-04-17 19:24:19MQTT publish: topic 'zigbee2mqtt/Kitchen FCOB CCT', payload '{"brightness":188,"color":{"h":0,"hue":0},"color_mode":"color_temp","color_temp":377,"do_not_disturb":null,"last_seen":"2024-04-17T19:24:27+03:00","linkquality":92,"state":"ON"}'
info 2024-04-17 19:24:19MQTT publish: topic 'zigbee2mqtt/Kitchen FCOB CCT', payload '{"brightness":188,"color":{"h":0,"hue":0},"color_mode":"color_temp","color_temp":377,"do_not_disturb":null,"last_seen":"2024-04-17T19:24:27+03:00","linkquality":96,"state":"ON"}'
info 2024-04-17 19:24:19MQTT publish: topic 'zigbee2mqtt/Kitchen FCOB CCT', payload '{"brightness":740,"color":{"h":0,"hue":0},"color_mode":"color_temp","color_temp":377,"do_not_disturb":null,"last_seen":"2024-04-17T19:24:28+03:00","linkquality":96,"state":"ON"}'
info 2024-04-17 19:24:19MQTT publish: topic 'zigbee2mqtt/Kitchen FCOB CCT', payload '{"brightness":740,"color":{"h":0,"hue":0},"color_mode":"color_temp","color_temp":377,"do_not_disturb":null,"last_seen":"2024-04-17T19:24:28+03:00","linkquality":92,"state":"ON"}'
info 2024-04-17 19:24:19MQTT publish: topic 'zigbee2mqtt/Kitchen FCOB CCT', payload '{"brightness":188,"color":{"h":0,"hue":0},"color_mode":"color_temp","color_temp":377,"do_not_disturb":null,"last_seen":"2024-04-17T19:24:28+03:00","linkquality":92,"state":"ON"}'
```

The value for brightness in the spurious messages seems random, I've seen as high as 1000, as does the number of spurious messages produced. This kind of message spam also explains why it seems so common for my lights to freeze for up to a minute and then rapidly cycle through all the commands sent while it was frozen.

dpgh947, 2024-05-06:
I read elsewhere that upgrading core to at least 2024.4 will affect this, so I have updated to latest core, and updated this, and it now works.

closed 2024-05-06 by dpgh947