Koenkk/zigbee2mqtt #22035: Sengled Bulbs lose RGB after 1.36.1-1 update

iridris, 2024-04-02:
### What happened?

After updating to 1.36.1-1, several Sengled bulbs are no longer able to be set to RGB colors in Home Assistant. They only appear to support white color temperature. Rolling back to 1.36.0 restores RGB functionality. Bulbs are a mix of Sengled E21-N1EA and E11-N1EA.

### What did you expect to happen?

Bulbs should be able to be set to RGB colors

### How to reproduce it (minimal and precise)

Update to 1.36.1-1. Observe that bulbs lose color support.

### Zigbee2MQTT version

1.36.1-1

### Adapter firmware version

20230507

### Adapter

SONOFF ZBDongle-P

### Setup

Add-on on Home Assistant OS in a VM.

nathan-curtis, 2024-04-02:
Same. Conbee II adapter.
All sengled bulbs reconfigured and lost RGB conntrols on update. 
Revert to previous is a current workaround but can't upgrade until repaired.

shad0wca7, 2024-04-02:
Same, Sonoff adapter. All sengled bulbs have lost RGB.

bweise84, 2024-04-02:
Same thing. Sonoff Coordinator, Sengled E21-N1EA bulbs. Works on 1.36.0. Home Assistant only has options to change color temperature if Zigbee2MQTT upgraded to 1.36.1

kkohls, 2024-04-02:
Having the same issue. Sengled E1F-N5E bulbs. Interesting that the Zigbee2Mqqt documentation for the bulbs no longer list the color options as begin exposed (for a color bulb!).

ThreeFN, 2024-04-02:
Can confirm the same issue for the E11-N1EA bulb as well.

EagleLizard, 2024-04-03:
I confirmed that this issue is resolved when downgrading zigbee2mqtt to `1.36.0`, via docker-compose.yml:
```yml
zigbee2mqtt:
    container_name: zigbee2mqtt
    # image: koenkk/zigbee2mqtt
    image: koenkk/zigbee2mqtt:1.36.0 # 1.36.1 broke sengled E21-N1EA x/y/rgb colors
```

Pulling via `docker compose pull zigbee2mqtt` and restarting on `1.36.0` fixed the issue for me.

## `1.36.0`:
[image]

## `1.36.1` (broken):
[image]

mrskycriper, 2024-04-03:
Can somebody please test any affected Sengled bulb with this temporary external converter? I need to now if they support hs, xy or both color modes. Replace `zigbeeModel` for your bulb if necessary.

```js
const {light} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['E21-N1EA'],
    model: 'E21-N1EA',
    description: 'Smart LED multicolor A19 bulb',
    vendor: 'Sengled',
    extend: [light({effect: false, powerOnBehavior: false, colorTemp: {range: [154, 500]}, color: true})],
}

module.exports = definition;
```

iridris, 2024-04-03:
@mrskycriper I've never used external converters before but hopefully I did this right. Added it, restarted, and it looks like both hs and xy are working. I'm able to click the colors on either one and it changes the bulb as expected. Note I am doing this test from 1.36.0 still.

[image]
```
{
    "brightness": 38,
    "color": {
        "h": 123,
        "hue": 123,
        "s": 100,
        "saturation": 100,
        "x": 0.1704,
        "y": 0.709
    },
    "color_mode": "xy",
    "color_temp": 154,
    "energy": 2.46,
    "last_seen": "2024-04-03T07:39:57-04:00",
    "linkquality": 43,
    "power": 0.2,
    "state": "ON",
    "update": {
        "installed_version": 36,
        "latest_version": 36,
        "state": "idle"
    },
    "update_available": null
}
```

mrskycriper, 2024-04-03:
> @mrskycriper I've never used external converters before but hopefully I did this right. Added it, restarted, and it looks like both hs and xy are working. I'm able to click the colors on either one [...]

Excellent news, this means that my fix will work. Thanks!

kkohls, 2024-04-03:
I too have never used an external converter so take this for what it is...

I copy/pasted the code and changed the model to reflect my bulb and submitted. Restarted Z2M and also HA. No errors in the logs, but still no color option for the E1F-N5E bulb.

mrskycriper, 2024-04-03:
> I too have never used an external converter so take this for what it is...
> 
> 
> 
> I copy/pasted the code and changed the model to reflect my bulb and submitted. Restarted Z2M and also HA. No errors in the logs, but still no color option for the E1F-N5E bulb. 

Were there any other visible changes? Like, for example, description? Could you try changing the text there and restarting Z2M?

mrskycriper, 2024-04-03:
> I changed the description and it did not reflect the change after a restart. Odd thing..

How is the external converter set up? A there any errors in the begging of log file after reboot?

kkohls, 2024-04-03:
> > I changed the description and it did not reflect the change after a restart. Odd thing..
> 
> How is the external converter set up? A there any errors in the begging of log file after reboot?

The screen shot doesn't show the entire line, but it's a copy/paste of the provided code with the model and description changed. Not seeing any errors after restart.

[image]

kkohls, 2024-04-03:
> @kkohls You need to put that code into a file in your zigbee2mqtt folder, and then put the name of the file into that field in the UI.

Ahhh.. ok. Like I said, never used them before. I'll do that and report back. Thank you for the teaching moment.

kkohls, 2024-04-03:
The above code [when properly applied ;-)] works fine with the E1F-N5E bulb. Apologies for complicating the situation. Thank you for your patience and allowing me to learn something new. Appreciate the time, effort, and energy you all put into this.

mrskycriper, 2024-04-03:
I've decided to revert all affected bulbs to XY color space, as was before the bug

lthown, 2024-04-03:
you need to create a file with the name ending in .js - probably something like /root/config/zigbee2mqtt/E11-N1EA.js and then you put the E11-N1EA.js in the the text field in settings > external converters 
[image]