Koenkk/zigbee2mqtt #22039: [New device support]: Imou door/windows Sensor or contact sensor Zigbee

MiAutomations, 2024-04-02:
### Link

https://www.imoulife.com/en/product/detail/imou-home-security-system

### Database entry

{"id":5,"type":"EndDevice","ieeeAddr":"0x8cf681fffef52ff9","nwkAddr":54657,"manufId":21845,"manufName":"MultIR","powerSource":"Battery","modelId":"ZD1-EN","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,1280,2821],"outClusterList":[3],"clusters":{"genPowerCfg":{"attributes":{"batteryPercentageRemaining":31}},"genBasic":{"attributes":{}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b0024c2772c","zoneState":1}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c2772c","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":65000,"repChange":10,"manufacturerCode":null}],"meta":{}}},"appVersion":4,"hwVersion":1,"dateCode":"20220615","swBuildId":"MC100_v1.2.1","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1712078932465}

### Comments

I tried to connect this device but Still not supported 

Can you please add this device to the supported devices list on Z2M devices ? 

Thank you for your support 

### External definition

```shell
const {battery, identify, iasZoneAlarm} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['ZD1-EN'],
    model: 'ZD1-EN',
    vendor: 'MultIR',
    description: 'Automatically generated definition',
    extend: [battery(), identify(), iasZoneAlarm({"zoneType":"generic","zoneAttributes":["alarm_1","alarm_2","tamper","battery_low"]})],
    meta: {},
};

module.exports = definition;
```

Koenkk, 2024-04-03:
Does it work with the generated external definition?

MiAutomations, 2024-04-03:
> Does it work with the generated external definition?

No, I try and I don't received any status change reported

ByAnz12, 2024-04-04:
hi im use this sensor, imou zd1.
before update, this sensor work with generated external definition.
but after update Zigbee2MQTT version
 1.36.1 this device Unsupported again.

is bug?
im see at update Zigbee2MQTT version 1.36.1 list zigbee new device with vendor IMOU, but sensor ZD1 not in list.

so, temporary im use restore old version Zigbee2MQTT from my backup.

ByAnz12, 2024-04-04:
oh yeah im fix for this problem.
i see from this post [https://github.com/Koenkk/zigbee2mqtt/issues/22026#issuecomment-2030622945](url)

so im change js file to this code.

`const {} = require('zigbee-herdsman-converters/lib/modernExtend');
// Add the lines below
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const {} = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['ZD1-EN'],
    model: 'ZD1-EN',
    vendor: 'Imou',
    description: 'Door/window sensor', 
    extend: [],
    fromZigbee: [fz.ias_contact_alarm_1, fz.battery, fz.ignore_iaszone_attreport],
    toZigbee: [],
    exposes: [e.battery(), e.contact(), e.battery_low(), e.tamper()], 
};
module.exports = definition;`

mrskycriper, 2024-04-12:
Can anyone test it with this external converter?

```js
const {battery, identify, iasZoneAlarm} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['ZD1-EN'],
    model: 'ZD1-EN',
    vendor: 'IMOU',
    description: 'Door & window sensor',
    extend: [
        iasZoneAlarm({zoneType: 'contact', zoneAttributes: ['alarm_1', 'tamper', 'battery_low']}),
        identify({isSleepy: true}),
        battery(),
    ],
};

module.exports = definition;
```

MiAutomations, 2024-04-16:
Hello 

I try use as a external converter the code shared by the @mrskycriper and @ByAnz12 without success...

mrskycriper, 2024-04-16:
Can you capture some debug logs of the device operating? At least for opening/closing the contact and pressing the pairing button once (if it doesn't reset of course)

MiAutomations, 2024-04-16:
> Can you capture some debug logs of the device operating? At least for opening/closing the contact and pressing the pairing button once (if it doesn't reset of course)

I try to get some logs in debug but nothing is appearing in the log related to this device ...

closed 2024-11-13 by github-actions[bot]

MiAutomations, 2024-11-13:
For your Information the device is now working as expected

Only the Battery % still as null, but the contact and the tamper are reporting correctly 

Thank you for your support

PawLowo, 2025-01-21:
> For your Information the device is now working as expected
> 
> Only the Battery % still as null, but the contact and the tamper are reporting correctly
> 
> Thank you for your support

Hello. At this time does still shows the battery as null? Do you have any updates please?

MiAutomations, 2025-01-22:
> > For your Information the device is now working as expected
> > Only the Battery % still as null, but the contact and the tamper are reporting correctly
> > Thank you for your support
> 
> Hello. At this time does still shows the battery as null? Do you have any updates please?

No, the battery still not reported

gabrielcolceriu, 2025-03-09:
`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const definition = {
    zigbeeModel: ['ZD1-EN'],
    model: 'ZD1-EN',
    vendor: 'IMOU',
    icon: 'mdi:motion-sensor',
    description: 'Door & window sensor',
    fromZigbee: [fz.ias_contact_alarm_1, fz.battery, fz.ias_enroll],
    toZigbee: [],
    exposes: [
        e.contact(),
        e.battery(),
        e.tamper(),
    ],
};
module.exports = definition;`
Using this code and the sensor works. Tried to add it to the database but the owner wanted me to add a PR. I am not a coder, or do know how to do this. I got the help of chatgpt for the coding of the external converter. If someone knows how to do this please help. Thank you!

MiAutomations, 2025-03-10:
> `const fz = require('zigbee-herdsman-converters/converters/fromZigbee'); const tz = require('zigbee-herdsman-converters/converters/toZigbee'); const exposes = [...]

Thank you 

Can you please share the converter file? 

Best regards

gabrielcolceriu, 2025-03-11:
[https://github.com/Koenkk/zigbee2mqtt/issues/26652](url)

The Sensor was added by the owner of the repo.