Koenkk/zigbee2mqtt #22040: TS011F_plug_1 (_TZ3000_w0qqde0g) randomly powers off

jocel1, 2024-04-02:
### What happened?

My TS011F_plug_1 has just powered off by itself.
When the event occurs, I had the following in the log : 

```
debug 2024-04-02 19:01:31: Received Zigbee message from 'Prise frigo', type 'attributeReport', cluster 'manuSpecificTuya_3', data '{"53249":16}' from endpoint 1 with groupID 0
debug 2024-04-02 19:01:31: No converter available for 'TS011F_plug_1' with cluster 'manuSpecificTuya_3' and type 'attributeReport' and data '{"53249":16}'
debug 2024-04-02 19:01:31: Received Zigbee message from 'Prise frigo', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info  2024-04-02 19:01:31: MQTT publish: topic 'zigbee2mqtt/Prise frigo', payload '{"child_lock":"UNLOCK","current":0.4,"energy":60.85,"indicator_mode":"off/on","linkquality":18,"power":66,"power_outage_memory":"on","state":"OFF","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":false,"voltage":226}'
debug 2024-04-02 19:01:31: Received Zigbee message from 'Prise frigo', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsVoltage":0}' from endpoint 1 with groupID 0
info  2024-04-02 19:01:31: MQTT publish: topic 'zigbee2mqtt/Prise frigo', payload '{"child_lock":"UNLOCK","current":0.4,"energy":60.85,"indicator_mode":"off/on","linkquality":18,"power":0,"power_outage_memory":"on","state":"OFF","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":false,"voltage":226}'
```

```
debug 2024-04-02 19:01:32: Received Zigbee message from 'Prise frigo', type 'attributeReport', cluster 'manuSpecificTuya_3', data '{"53249":0}' from endpoint 1 with groupID 0
debug 2024-04-02 19:01:32: No converter available for 'TS011F_plug_1' with cluster 'manuSpecificTuya_3' and type 'attributeReport' and data '{"53249":0}'
```

Any idea what's this manuSpecificTuya_3 (this is the first time I have this issue).

[image]

FYI I upgraded to Z2M 1.36.1-1 yesterday

### What did you expect to happen?

No autopower off for my fridge :) 

### How to reproduce it (minimal and precise)

Nothing specific

### Zigbee2MQTT version

1.36.1-1

### Adapter firmware version

20240315

### Adapter

 SONOFF ZigBee 3.0 USB Dongle Plus (CC2652P)

### Setup

Add-on on homeassistant with Pi 5

### Debug log

```
debug 2024-04-02 19:01:31: Received Zigbee message from 'Prise frigo', type 'attributeReport', cluster 'manuSpecificTuya_3', data '{"53249":16}' from endpoint 1 with groupID 0
debug 2024-04-02 19:01:31: No converter available for 'TS011F_plug_1' with cluster 'manuSpecificTuya_3' and type 'attributeReport' and data '{"53249":16}'
debug 2024-04-02 19:01:31: Received Zigbee message from 'Prise frigo', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info  2024-04-02 19:01:31: MQTT publish: topic 'zigbee2mqtt/Prise frigo', payload '{"child_lock":"UNLOCK","current":0.4,"energy":60.85,"indicator_mode":"off/on","linkquality":18,"power":66,"power_outage_memory":"on","state":"OFF","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":false,"voltage":226}'
debug 2024-04-02 19:01:31: Received Zigbee message from 'Prise frigo', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsVoltage":0}' from endpoint 1 with groupID 0
info  2024-04-02 19:01:31: MQTT publish: topic 'zigbee2mqtt/Prise frigo', payload '{"child_lock":"UNLOCK","current":0.4,"energy":60.85,"indicator_mode":"off/on","linkquality":18,"power":0,"power_outage_memory":"on","state":"OFF","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":false,"voltage":226}'
```
[...]
```
debug 2024-04-02 19:01:32: Received Zigbee message from 'Prise frigo', type 'attributeReport', cluster 'manuSpecificTuya_3', data '{"53249":0}' from endpoint 1 with groupID 0
debug 2024-04-02 19:01:32: No converter available for 'TS011F_plug_1' with cluster 'manuSpecificTuya_3' and type 'attributeReport' and data '{"53249":0}'
```

lux73, 2024-04-02:
AFAIK there are some faulty Plugs that trigger these Problems ^^

nothing that Z2M can do...   Hardware issue

jocel1, 2024-04-02:
@lux73 ok, I guess it's somehow related to the current spike when the compressor turns on. I'll consider another plug for my fridge then (or do an automation to keep it on)

lux73, 2024-04-02:
ah okay, make sense. then it's overcurrent Protection ^^ - the starting current from your Compressor seems to high

if i switch on both my wash disher & wash dryer my `TS011F_plug_1` will hard shut off due to overcurrent protection too

jocel1, 2024-04-02:
> ah okay, make sense. then it's overcurrent Protection ^^ - the starting current from your Compressor seems to high
> 
> if i switch on both my wash disher & wash dryer my `TS011F_plug_1` will hard shut off due to overcurrent protection too

What's strange is it's the first time in several month it occurs. Anyway the overcurrent protection should at least respect the "Power outage memory" setting, but that's not a Z2M issue

jocel1, 2024-04-02:
And this is covered by discussion https://github.com/Koenkk/zigbee2mqtt/discussions/13343 so I'll close this issue :)

closed 2024-04-02 by jocel1

mrmaximas, 2024-06-17:
@jocel1 This plug has, among other things, voltage protection - above 260 volts the load is switched off and will not be switched on automatically if the voltage turn to below 260V. This is very unpleasant behavior for me