Koenkk/zigbee2mqtt #22043: Enbrighten 43082 goes offline around ~10 minutes after restarting Z2M on 1.36.1

pyrodex, 2024-04-02:
### What happened?

I have two Enbrighten 43082 and after upgrading to 1.36.1 they seem to go offline after about 10 minutes and restarting Z2M fixes it but again they go offline after 10 minutes until Z2M is restarted. 

### What did you expect to happen?

The devices should stay online all the time like it did in 1.36.0

### How to reproduce it (minimal and precise)

Upgrade from 1.36.0 to 1.36.1, restart Z2M and devices are on-line wait 10 minutes and they show up offline. 

### Zigbee2MQTT version

1.36.1

### Adapter firmware version

20230507

### Adapter

TubesZB CC2652 P7 Based Zigbee to PoE Coordinator 2023

### Setup

Z2M Docker container on Debian 12 x86-64 with latest stable Docker version

### Debug log

```
info  2024-04-02 12:21:35: Family Room Dimmer (0xa49e69fffe62fd02): 43082 - Enbrighten Zigbee in-wall smart dimmer (Router)
info  2024-04-02 12:21:35: MQTT publish: topic 'zigbee2mqtt/Family Room Dimmer/availability', payload '{"state":"online"}'
info  2024-04-02 12:21:36: MQTT publish: topic 'homeassistant/light/0xa49e69fffe62fd02/light/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/Family Room Dimmer/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","brightness":true,"brightness_scale":254,"command_topic":"zigbee2mqtt/Family Room Dimmer/set","device":{"identifiers":["zigbee2mqtt_0xa49e69fffe62fd02"],"manufacturer":"Enbrighten","model":"Zigbee in-wall smart dimmer (43082)","name":"Family Room Dimmer","via_device":"zigbee2mqtt_bridge_0x00124b002a7b57cc"},"name":null,"object_id":"family_room_dimmer","origin":{"name":"Zigbee2MQTT","sw":"1.36.1","url":"https://www.zigbee2mqtt.io"},"schema":"json","state_topic":"zigbee2mqtt/Family Room Dimmer","unique_id":"0xa49e69fffe62fd02_light_zigbee2mqtt"}'
info  2024-04-02 12:21:36: MQTT publish: topic 'homeassistant/sensor/0xa49e69fffe62fd02/power/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/Family Room Dimmer/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa49e69fffe62fd02"],"manufacturer":"Enbrighten","model":"Zigbee in-wall smart dimmer (43082)","name":"Family Room Dimmer","via_device":"zigbee2mqtt_bridge_0x00124b002a7b57cc"},"device_class":"power","enabled_by_default":true,"entity_category":"diagnostic","object_id":"family_room_dimmer_power","origin":{"name":"Zigbee2MQTT","sw":"1.36.1","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/Family Room Dimmer","unique_id":"0xa49e69fffe62fd02_power_zigbee2mqtt","unit_of_measurement":"W","value_template":"{{ value_json.power }}"}'
info  2024-04-02 12:21:36: MQTT publish: topic 'homeassistant/sensor/0xa49e69fffe62fd02/energy/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/Family Room Dimmer/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa49e69fffe62fd02"],"manufacturer":"Enbrighten","model":"Zigbee in-wall smart dimmer (43082)","name":"Family Room Dimmer","via_device":"zigbee2mqtt_bridge_0x00124b002a7b57cc"},"device_class":"energy","enabled_by_default":true,"object_id":"family_room_dimmer_energy","origin":{"name":"Zigbee2MQTT","sw":"1.36.1","url":"https://www.zigbee2mqtt.io"},"state_class":"total_increasing","state_topic":"zigbee2mqtt/Family Room Dimmer","unique_id":"0xa49e69fffe62fd02_energy_zigbee2mqtt","unit_of_measurement":"kWh","value_template":"{{ value_json.energy }}"}'
info  2024-04-02 12:21:36: MQTT publish: topic 'homeassistant/sensor/0xa49e69fffe62fd02/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/Family Room Dimmer/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa49e69fffe62fd02"],"manufacturer":"Enbrighten","model":"Zigbee in-wall smart dimmer (43082)","name":"Family Room Dimmer","via_device":"zigbee2mqtt_bridge_0x00124b002a7b57cc"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"Linkquality","object_id":"family_room_dimmer_linkquality","origin":{"name":"Zigbee2MQTT","sw":"1.36.1","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/Family Room Dimmer","unique_id":"0xa49e69fffe62fd02_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
info  2024-04-02 12:21:36: MQTT publish: topic 'homeassistant/sensor/0xa49e69fffe62fd02/last_seen/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/Family Room Dimmer/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa49e69fffe62fd02"],"manufacturer":"Enbrighten","model":"Zigbee in-wall smart dimmer (43082)","name":"Family Room Dimmer","via_device":"zigbee2mqtt_bridge_0x00124b002a7b57cc"},"device_class":"timestamp","enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:clock","name":"Last seen","object_id":"family_room_dimmer_last_seen","origin":{"name":"Zigbee2MQTT","sw":"1.36.1","url":"https://www.zigbee2mqtt.io"},"state_topic":"zigbee2mqtt/Family Room Dimmer","unique_id":"0xa49e69fffe62fd02_last_seen_zigbee2mqtt","value_template":"{{ value_json.last_seen }}"}'
info  2024-04-02 12:21:36: MQTT publish: topic 'zigbee2mqtt/Family Room Dimmer', payload '{"brightness":254,"device":{"applicationVersion":18,"friendlyName":"Family Room Dimmer","hardwareVersion":2,"ieeeAddr":"0xa49e69fffe62fd02","manufacturerID":4388,"manufacturerName":"Jasco Products","model":"43082","networkAddress":46631,"powerSource":"Mains (single phase)","stackVersion":101,"type":"Router","zclVersion":3},"energy":0.91,"last_seen":"2024-04-02T12:20:55-04:00","linkquality":112,"power":0,"state":"OFF"}'
info  2024-04-02 12:22:01: MQTT publish: topic 'zigbee2mqtt/Family Room Dimmer', payload '{"brightness":254,"device":{"applicationVersion":18,"friendlyName":"Family Room Dimmer","hardwareVersion":2,"ieeeAddr":"0xa49e69fffe62fd02","manufacturerID":4388,"manufacturerName":"Jasco Products","model":"43082","networkAddress":46631,"powerSource":"Mains (single phase)","stackVersion":101,"type":"Router","zclVersion":3},"energy":0.91,"last_seen":"2024-04-02T12:20:55-04:00","linkquality":112,"power":0,"state":"OFF"}'
info  2024-04-02 12:22:24: MQTT publish: topic 'zigbee2mqtt/Family Room Dimmer', payload '{"brightness":254,"device":{"applicationVersion":18,"friendlyName":"Family Room Dimmer","hardwareVersion":2,"ieeeAddr":"0xa49e69fffe62fd02","manufacturerID":4388,"manufacturerName":"Jasco Products","model":"43082","networkAddress":46631,"powerSource":"Mains (single phase)","stackVersion":101,"type":"Router","zclVersion":3},"energy":0.91,"last_seen":"2024-04-02T12:22:24-04:00","linkquality":112,"power":0,"state":"OFF"}'
info  2024-04-02 12:22:26: MQTT publish: topic 'zigbee2mqtt/Family Room Dimmer', payload '{"brightness":254,"device":{"applicationVersion":18,"friendlyName":"Family Room Dimmer","hardwareVersion":2,"ieeeAddr":"0xa49e69fffe62fd02","manufacturerID":4388,"manufacturerName":"Jasco Products","model":"43082","networkAddress":46631,"powerSource":"Mains (single phase)","stackVersion":101,"type":"Router","zclVersion":3},"energy":0.91,"last_seen":"2024-04-02T12:22:26-04:00","linkquality":112,"power":0,"state":"OFF"}'
info  2024-04-02 12:22:26: MQTT publish: topic 'zigbee2mqtt/Family Room Dimmer', payload '{"brightness":254,"device":{"applicationVersion":18,"friendlyName":"Family Room Dimmer","hardwareVersion":2,"ieeeAddr":"0xa49e69fffe62fd02","manufacturerID":4388,"manufacturerName":"Jasco Products","model":"43082","networkAddress":46631,"powerSource":"Mains (single phase)","stackVersion":101,"type":"Router","zclVersion":3},"energy":0.91,"last_seen":"2024-04-02T12:22:26-04:00","linkquality":112,"power":0,"state":"ON"}'
info  2024-04-02 12:22:26: MQTT publish: topic 'zigbee2mqtt/Family Room Dimmer', payload '{"brightness":254,"device":{"applicationVersion":18,"friendlyName":"Family Room Dimmer","hardwareVersion":2,"ieeeAddr":"0xa49e69fffe62fd02","manufacturerID":4388,"manufacturerNam [... cut ...]
info  2024-04-02 12:30:17: MQTT publish: topic 'zigbee2mqtt/Family Room Dimmer', payload '{"brightness":254,"device":{"applicationVersion":18,"friendlyName":"Family Room Dimmer","hardwareVersion":2,"ieeeAddr":"0xa49e69fffe62fd02","manufacturerID":4388,"manufacturerName":"Jasco Products","model":"43082","networkAddress":46631,"powerSource":"Mains (single phase)","stackVersion":101,"type":"Router","zclVersion":3},"energy":0.91,"last_seen":"2024-04-02T12:23:04-04:00","linkquality":112,"power":0,"state":"OFF"}'
```

smarthomelawyer, 2024-04-02:
I have the exact same issue with all of my Enbrighten switches (43076, 43080). Rolling back to 1.36 resolved the issue.

pyrodex, 2024-04-02:
> I have the exact same issue with all of my Enbrighten switches (43076, 43080). Rolling back to 1.36 resolved the issue.

Yea I just rolled back for stability and can test since it isn't a huge impact as I only have two in my ecosystem but since I am away from home I wanted to make sure I can control the lights until we return home.

smarthomelawyer, 2024-04-02:
> Yea I just rolled back for stability and can test since it isn't a huge impact as I only have two in my ecosystem but since I am away from home I wanted to make sure I can control the lights until [...]

Makes sense! Nearly all of my switches are Enbrighten so as soon as I noticed the issue I rolled back. If finished only a few minutes before my wife got home from work. She has very little patience when my smart home gadgets go haywire! Unfortunately, maintaining a happy wife and therefore happy life hinders my ability help troubleshoot

woodartworx, 2024-04-02:
I have about 20 of the Enbrighten switches (43082/43078) and upon install of 1.36.1 all of them go offline after a short while.  I can reconfigure them and they come back online, but again after 10 min or so they all go offline.  Rolled back to 1.36.0 and the issue goes away and those devices are once again stable.  All of my other devices both wired and wireless are working ok...

Emilef22, 2024-04-02:
same here, with sonoff dongle plus

Great-Chart, 2024-04-02:
I've got one device (Samotech dimmer) suffering this; reported under #22041.
Since noted that one goes offline for 10 mins, comes back on-line for 10 mins and repeats but skips a beat oncexan hour (one stint is 20 mins).

Letting it run overnight to see how consistent it is.

deviantintegral, 2024-04-02:
I've noticed the same, but it's happening after 3 hours. I have the availability timeout set to 180 minutes, and I'm guessing that the default is 10?

I think this is caused by the change at https://github.com/Koenkk/zigbee-herdsman/pull/992 - I reverted it locally and devices remained available. Now that I realize I can drop the timeout check I can have a quicker debug cycle on this!

As a workaround, you can disable availability checks for those devices. The devices are all totally functional, it's just something in zigbee2mqtt's availability checking that's marking them offline. I'm not sure yet where the issue is (we _added_ an additional event notification for address changes), but if we're lucky I'll have a PR up in herdsman tonight for testing.

deviantintegral, 2024-04-03:
PR is up at https://github.com/Koenkk/zigbee-herdsman/pull/1004 for anyone interested.

Loric76, 2024-04-03:
I’m also running into this problem. Z2M 1.36.1, Sonoff Dongle Plus, Enbrighten 43078 and 43080 devices.

I’ll try disabling availability for these devices in the interim until this gets fixed.

drizztguen77, 2024-04-03:
I also have the exact same issue with 5 Embrighten 43078 switches. All of them go offline after about 10-15 minutes. If I flip the switch off and on it comes back online and then goes offline again. I restored the previous version of zigbee2mqtt and all are working fine again.

onedr0p, 2024-04-03:
Thanks for the update @deviantintegral I've rolled back to 1.35.x until that is released.

closed 2024-04-03 by Koenkk

pknaz, 2024-04-04:
I'm also having this issue :(

pknaz, 2024-04-04:
I'm using HA OS, I know next to nothing about docker.  Is there an easy to follow noob worthy set of instructions for how to "roll back" to 1.35.x?  I have Z2M installed in HA using the addon feature.

rehanson, 2024-04-07:
I'm chiming in to say I had the same issue with my Jasco/Enbrighten 43080 dimmers after upgrading to 1.36.1.

@pknaz I was able to roll back my Z2M version in Home Assistant by going to Settings > System > Backups and finding the backup that was made for Z2M when I updated to 1.36.1. You can restore the backed up version from there.

pyrodex, 2024-04-07:
> When will this patch be pushed out? It keeps automatically running the update every night and my automations fail because the switches are all offline. Then I have to get up and restore it again every morning. It's getting very frustrating having this happen every night. Will the patch be pushed out to us soon?

You have a few options...

- Roll back to 1.36.0, the quickest fix for your needs
- Upgrade to dev, this can introduce more issues but fix your issue
- Wait like everyone else for the next release and realize this is an open source project.

drizztguen77, 2024-04-07:
I understand those are the options. I have rolled it back. It's just that I have to keep rolling it back over and over. I also understand it's open source but that being said the problem has been fixed so why not push it out? I was just asking when it will be pushed out.

pyrodex, 2024-04-07:
> I understand those are the options. I have rolled it back. It's just that I have to keep rolling it back over and over. I also understand it's open source but that being said the problem has been fixed so why not push it out? I was just asking when it will be pushed out.

I rolled back once, you may need to see why you are being required to roll back over and over again. 

If you go and look at the releases page you will see the timeline between releases and can better judge.

drizztguen77, 2024-04-07:
Home Assistant automatically installs updates and there is no way to turn it off that I'm aware of. So I roll it back and then the next day Home Assistant installs the update again.

pyrodex, 2024-04-07:
> Home Assistant automatically installs updates and there is no way to turn it off that I'm aware of. So I roll it back and then the next day Home Assistant installs the update again.

You should disable the auto update feature on the addon itself in Home Assistant.

drizztguen77, 2024-04-07:
Then how do you know when the bug fix is actually pushed out?

pyrodex, 2024-04-07:
> Then how do you know when the bug fix is actually pushed out?

Let me teach you about open source...

You can subscribe to GitHub repositories and get notifications if there is a new release.

drizztguen77, 2024-04-07:
Fair enough. I forgot about that. I don't use them much.

jwittlincohen, 2024-04-11:
I am also seeing the same behavior with all of my (six) Embrighten 43084 smart light switches on Z2M 1.3.6.1.  It is not impacting my Leviton DG6HD smart dimmers, or the dozens of Aqara motion, moisture, temperature/humidity, door/window, or vibration sensors. Rolling back the Z2M add-on in Home Assistant to 1.35.2 resolved the issue.  However, I did have to disable auto-updates to prevent the add-on from automatically being upgraded.