Koenkk/zigbee2mqtt #22047: since update 1.36.1 all sliders and switches on the dashboard no longer work

Jeepmg, 2024-04-02:
### What happened?

Before the last update, I could use the dashboard to turn lights on or off using the switches. I could also adjust the height of the shutters with a slider.

Now it doesn't work anymore.

### How to reproduce it (minimal and precise)

Use a lamp switch in the dashboard

### Zigbee2MQTT version

1.36.1

### Adapter firmware version

20230507

### Adapter

zStack3x0

### Setup

RPI

epaulsen, 2024-04-03:
Confirm.  Same issue here.  Setting state from HomeAssistant or directly via MQTT works, but not the dashboard.
No entry in the logs when you do something in the dashboard.
I'm attempting to add a screen recording GIF repro to this message, never done that on github before, so not sure how it will work out.
[image]

bo0tzz, 2024-04-03:
I can confirm this issue as well. Using the sliders in devices' individual pages does work as expected.

hack2spider, 2024-04-03:
I have several problems as well with z2m .. it used to work fine for like 2-3 years but now I see a lot of RAM building up in the add-on and I have to restart the add-on to recover .. eventually it goes bad again where the sliders and other motion sensors from aqara won't work properly ..

kzajac83, 2024-04-04:
Have the same problem, on dashboard sliders not work after update.  

@bo0tzz 
I see attached fix, need wait to next month for prod version or switch to edge version,correct?

JayJay-Master, 2024-04-05:
Same problem here. All switches and sliders on the dashboard do not work. 
However, all switches and sliders on the side of the individual devices work in detail tab.

Zigbee2MQTT-Version
    1.36.1 commit: ffc2ff1
Coordinator-Typ
    ConBee2/RaspBee2
Coordinator-Version
    0x26720700
Coordinator IEEE Adresse
    0x00212effff0781ef
Frontend-Version
    0.6.161
zigbee_herdsman_converters_version
    19.11.2
zigbee_herdsman_version
    0.40.3

tested browser: Firefox (Windows 10), Edge (Windows 10), Safari (iOS 17.4.1, iPadOS 17.4.1)

Koenkk, 2024-04-07:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/22028 and fixed in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) already.

closed 2024-04-07 by Koenkk