Koenkk/zigbee2mqtt #22050: TuYa ZG-204ZM being identified as ZG-204ZL

amitmalik4, 2024-04-03:
### What happened?

Just got a new TuYa ZG-204ZM (  Presence sensor). But zigbee2mqtt is identifying it a ZG-204ZL (Motion Sensor). Running zigbee2Mqtt version 1.36.1  

### Zigbee2MQTT version

1.36.1

### Adapter firmware version

20210708

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus

### Setup

Add-on Home Assistant OS running on Pi4

MV1707, 2024-04-19:
@amitmalik4 I just ordered one. Not yet received and not yet confirmed to be exactly ZM-204ZL as with AliExpress you never know 100% what iteration you'll get... But I ordered from one vendor explaining how to make it work with Z2M and a link to some documentation + js file here :  [https://drive.google.com/file/d/1uJxiw4Pe9BFhjCxT5fStgXbsseyq_ywL/view?spm=a2g0o.detail.1000023.1.64d2sDE7sDE7sZ](url)

Hopefully that will help.

amitmalik4, 2024-04-19:
> @amitmalik4 I just ordered one. Not yet received and not yet confirmed to be exactly ZM-204ZM as with AliExpress you never know 100% what iteration you'll get... But I ordered from one vendor explaining how to make it work with Z2M and a link to some documentation + js file here : [https://drive.google.com/file/d/1uJxiw4Pe9BFhjCxT5fStgXbsseyq_ywL/view?spm=a2g0o.detail.1000023.1.64d2sDE7sDE7sZ](url)
> 
> Hopefully that will help.
Thanks @MV1707. Unfortunately the above solution did not help. So looks like I got a bad (or old model?) device from AliExpress. Hopefully you will have better luck

Koterak, 2024-04-23:
I got sensors from mentioned seller above but found that for me it won't connect to zigbee -> Keeps blinking when pairing and never stops. In zigbee2mqtt it kepps joining/leaving zigbee network. When sniffing got timeout error from device. I'm still in contact with this seller trying to find solution. Maybe you will have better luck.

closed 2024-11-20 by github-actions[bot]