Koenkk/zigbee2mqtt #22053: Zigbee2Mqtt sometimes does not send any commands but recieves them

u20p17, 2024-04-03:
### What happened?

I have a quite big zigbee2mqtt network with lot of main power devices and a Sonoff C2652 usb stick as the coordinator. Most of the time everything is working perfectly. In the last weeks two times I had a complete outage, where the zigbee2mqtt docker was running fine, but no devices where sending their state anymore nor could i turn on any lights. The last time it seems to happen about 5am today. In the log I can not see any error at this time happening, but what you can see is, that at around 5am no more messages of my plugs where send... 

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/14847302/log.txt)

After restarting zigbee2mqtt docker container everything is again up and running.

### Zigbee2MQTT version

1.36.1

### Adapter firmware version

20230507

### Adapter

Sonoff C2652P

### Setup

Intel Nuc with Unraid, zigbee2mqtt in docker 

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/14847308/log.txt)

nobody1365, 2024-04-03:
I probably have the same issue. Also quite large zigbee installation (~40 devices). Runs fine for days, suddenly I cannot send something to a device not do I get anything back.  Restarting zigbee2mqtt did not help. Disconnecting the Sonoff ZBDongle-E and restarting zigbee2mqtt did help.  Anything I can do to help troubleshooting ?
Don't know exactly which update did bring this problem .. ~3 weeks now ...

GSzabados, 2024-04-04:
Does any of you using the coordinator with the multiprotocol (Zigbee and Thread(Matter)) firmware? If yes, then you should change it to a Zigbee only firmware.

u20p17, 2024-04-04:
I am using the P version with the CC2652 chip... so I do not use a multiprotocol firmware ...

nobody1365, 2024-04-04:
How can I get that information ?

Brignoud, 2024-04-05:
Similar (?) Issue - Using Front End trace any activity imposed by MQTT Client and managed any command like changing configuration or setting up different log
Noone of the command sent to any device is performed.

gio82c, 2024-04-06:
same here..about 2/3 times per day

nobody1365, 2024-04-07:
Is there a maximum number of devices the system can handle  ?

vwtune, 2024-04-09:
Same issue on 1.36.1 with a Skyconnect dongle and 65 zigbee devices.
I restored to zigbee2mqtt 1.35.3-1 and all is running fine again. so it i think there is something in version 1.36 which cause the problem.

nobody1365, 2024-04-12:
Also reverted to 1.35.3-1 and no issues so far ..

umtauscher, 2024-04-29:
I have exactly the same problem since yesterday. Out of the blue, I cannot send anything. Sensors are working fine.
Unfortunately I don‘t have a backup before 1.36.1-1
Is any suggestions?

umtauscher, 2024-04-30:
It happened to a neighbour of mine today as well. So I can rule out a hardware problem.
No lights can be switched any more.
It seems more an more devices are going offline over time.
For me it's Aqara LED Strip and all Ikea bulbs at the moment.

umtauscher, 2024-04-30:
I just tested the edge version in HA and it doesn't work as well.

umtauscher, 2024-04-30:
Does anyone have an idea how I could get 1.35 installed not having a backup?

vwtune, 2024-04-30:
> Does anyone have an idea how I could get 1.35 installed not having a backup?

https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.35.3

umtauscher, 2024-04-30:
Thanks, but how would I get that into my homeassistant installation?

vwtune, 2024-05-02:
There is an update out (1.37) but i don't see any about this issue. So i don't know if it will fix the issue with not sending commands.

I'm still staying on 1.35 and will not update to 1.37 (i also read new problems with the newer version)

umtauscher, 2024-05-02:
Well it doesn‘t fix my problem. Its the same. Only receive states, not send commands. But I see many different issues posted, that could be the same symptom.

umtauscher, 2024-05-02:
[image]

vwtune, 2024-05-02:
Please send this information on this topic: https://github.com/Koenkk/zigbee2mqtt/discussions/22379

maybay they will pick it up quickly

It is trange this problem still exists in version 1.37 :/

umtauscher, 2024-05-02:
I don‘t want to hijack topic #22379 since this is a complete different problem.

umtauscher, 2024-05-02:
Ok, my issue is solved (sort of). It must be interference from one of my neighbours.
I changed everything from channel 11 to channel 20 and all is working again (for now)

closed 2024-11-29 by github-actions[bot]