Koenkk/zigbee2mqtt #22059: IKEA Styrbar remote stopped by new Zigbee2mqtt update 1.36.1-1

bedi67, 2024-04-03:
### What happened?

I updated Zigbee2mqtt to 1.36.1-1, and IKEA Styrbar does not work. And i reload the 1.35 firmware and works again.

### Zigbee2MQTT version

1.35.2-1

### Adapter firmware version

V1.0

### Adapter

SkyConnect

### Setup

Mini PC, Debian 12

McFlyPartages, 2024-04-03:
Thanks for the information, I've been trying to pair my IKEA Styrbar for two days now.

At the moment the updates are not great, I've already had a problem with Philips devices.

Aren't there any checks before publication?

WilkoKetelaar, 2024-04-03:
I seem to have the same problem, some of my Styrbar remotes and als some of my Parrasoll door sensors stopped working.

aervig, 2024-04-04:
Same problem with Philips Hue dimmer switch RWL021

aervig, 2024-04-05:
Seems like this is related to issue #22073

Koenkk, 2024-04-06:
- Could you provide the debug log with both 1.36.0 and 1.36.1 when triggering actions on the device?
- Could you check if this has already been solved in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)?

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable debug logging.

aurelmarius, 2024-04-07:
Maybe related to https://github.com/Koenkk/zigbee2mqtt/issues/22084

phirecc, 2024-04-21:
I was having the same issue, couldn't get my new remote to work. It paired fine, LQI was being updated from time to time, but nothing appeared in the logs on button press.

So I was about to generate those logs for @Koenkk. I enabled debug logs, restarted zigbee2mqtt, pressed a button, and surprisingly the remote started working 😅. Maybe others can verify that it works after restarting?

This is what came up in the logs when I started pressing buttons.  Note that the device had already been paired, why was it being configured only now?
```
...
Zigbee2MQTT:info  2024-04-21 08:32:09: Configuring '0x842712fffe2593df'
Zigbee2MQTT:debug 2024-04-21 08:32:13: Received Zigbee message from '0x842712fffe2593df', type 'commandOn', cluster 'genOnOff', data '{}' from e
ndpoint 1 with groupID 0
Zigbee2MQTT:info  2024-04-21 08:32:13: MQTT publish: topic 'zigbee2mqtt/0x842712fffe2593df', payload '{"action":"on","linkquality":120}'
Zigbee2MQTT:debug 2024-04-21 08:32:17: Received Zigbee message from '0x842712fffe2593df', type 'readResponse', cluster 'genPowerCfg', data '{"ba
tteryPercentageRemaining":255}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2024-04-21 08:32:17: Successfully configured '0x842712fffe2593df'
Zigbee2MQTT:debug 2024-04-21 08:32:20: Received Zigbee message from '0x842712fffe2593df', type 'commandTradfriArrowSingle', cluster 'genScenes',
 data '{"value":257,"value2":13}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2024-04-21 08:32:20: MQTT publish: topic 'zigbee2mqtt/0x842712fffe2593df', payload '{"action":"arrow_left_click","linkquality"
:164}'
Zigbee2MQTT:debug 2024-04-21 08:32:25: Received Zigbee message from '0x842712fffe2593df', type 'commandTradfriArrowSingle', cluster 'genScenes',
 data '{"value":256,"value2":13}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2024-04-21 08:32:25: MQTT publish: topic 'zigbee2mqtt/0x842712fffe2593df', payload '{"action":"arrow_right_click","linkquality
":200}'
```

Running `Zigbee2MQTT version 1.36.1 (commit #ffc2ff1)`
The remote is a:
```
- Zigbee Model: Remote Control N2
- Firmware build date: 20221028
- Firmware version: 2.4.5
- Manufacturer: IKEA
- Model: E2001/E2002
```

closed 2024-12-10 by github-actions[bot]