Koenkk/zigbee2mqtt #22061: [New device support]: PHILIPS Hue Dymera could not be integrated // crash after adding this device (ERR_ASSERTION: LCW004 has multiple 'ota', this is not allowed)

pixolio, 2024-04-03:
### Link

https://www.philips-hue.com/de-de/p/hue-white---color-ambiance-hue-dymera-wandleuchte-fur-den-innen--und-aussenbereich/8720169234505

### Database entry

{"id":29,"type":"Router","ieeeAddr":"0x001788010de23e6e","nwkAddr":8781,"manufId":4107,"manufName":"Signify Netherlands B.V.","powerSource":"Mains (single phase)","modelId":"LCW004","epList":[11,12,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":269,"inClusterList":[0,3,4,5,6,8,4096,64515,768,64513],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"LCW004","manufacturerName":"Signify Netherlands B.V.","powerSource":1,"zclVersion":8,"appVersion":2,"stackVersion":1,"hwVersion":0,"dateCode":"20230609","swBuildId":"1.109.1"}}},"binds":[],"configuredReportings":[],"meta":{}},"12":{"profId":260,"epId":12,"devId":269,"inClusterList":[0,3,4,5,6,8,64515,768,64513],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"hwVersion":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":0,"dateCode":"20230609","swBuildId":"1.109.1","zclVersion":8,"interviewCompleted":true,"meta":{},"lastSeen":1712159953599}

### Comments

My Z2M integration (inside from Home Assistant) detected the device and crashed after that. Every restart after that failed. So I've removed manually the entries inside the configuration.yaml and database.db. So now my installation starts again without error - but the is no chance to add the new outdoor wall lamp "Hue Dymera". 

The error is probably due to the fact that this luminaire contains two individual lamps for the first time and that this is not yet supported by Z2M.

I had no chance to create the "external definition"

[image]

The log file after adding the lamp brings the following error messages in log:

```
info  2024-04-03 18:04:52: Logging to console and directory: '/config/zigbee2mqtt/log/2024-04-03.18-04-52' filename: log.txt
info  2024-04-03 18:04:52: Starting Zigbee2MQTT version 1.36.0 (commit #unknown)
info  2024-04-03 18:04:52: Starting zigbee-herdsman (0.35.1)
error 2024-04-03 18:05:59: Failed to start zigbee
error 2024-04-03 18:05:59: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
error 2024-04-03 18:05:59: Exiting...
error 2024-04-03 18:05:59: AssertionError [ERR_ASSERTION]: 'LCW004' has multiple 'ota', this is not allowed
    at processExtensions (/app/node_modules/zigbee-herdsman-converters/src/index.ts:119:32)
    at prepareDefinition (/app/node_modules/zigbee-herdsman-converters/src/index.ts:182:18)
    at findDefinition (/app/node_modules/zigbee-herdsman-converters/src/index.ts:294:16)
    at runNextTicks (node:internal/process/task_queues:60:5)
    at processTimers (node:internal/timers:509:9)
    at Object.findByDevice (/app/node_modules/zigbee-herdsman-converters/src/index.ts:266:22)
    at Device.resolveDefinition (/app/lib/model/device.ts:36:31)
    at Zigbee.start (/app/lib/zigbee.ts:69:13)
    at Controller.start (/app/lib/controller.ts:109:27)
    at start (/app/index.js:107:5)

```

### External definition

```shell
not possible
```

pixolio, 2024-04-03:
UPDATE to z2m 1.36.1 brings no change to that behavior

```
[22:31:21] INFO: Preparing to start...
[22:31:22] INFO: Socat not enabled
[22:31:24] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:info  2024-04-03 22:31:30: Logging to console and directory: '/config/zigbee2mqtt/log/2024-04-03.22-31-30' filename: log.txt
Zigbee2MQTT:info  2024-04-03 22:31:30: Starting Zigbee2MQTT version 1.36.1 (commit #unknown)
Zigbee2MQTT:info  2024-04-03 22:31:30: Starting zigbee-herdsman (0.40.3)
Zigbee2MQTT:error 2024-04-03 22:32:36: Failed to start zigbee
Zigbee2MQTT:error 2024-04-03 22:32:36: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2024-04-03 22:32:36: Exiting...
Zigbee2MQTT:error 2024-04-03 22:32:36: AssertionError [ERR_ASSERTION]: 'LCW004' has multiple 'ota', this is not allowed
    at processExtensions (/app/node_modules/zigbee-herdsman-converters/src/index.ts:124:28)
    at prepareDefinition (/app/node_modules/zigbee-herdsman-converters/src/index.ts:173:18)
    at findDefinition (/app/node_modules/zigbee-herdsman-converters/src/index.ts:285:16)
    at runNextTicks (node:internal/process/task_queues:60:5)
    at processTimers (node:internal/timers:509:9)
    at Object.findByDevice (/app/node_modules/zigbee-herdsman-converters/src/index.ts:257:22)
    at Device.resolveDefinition (/app/lib/model/device.ts:36:31)
    at Zigbee.start (/app/lib/zigbee.ts:69:13)
    at Controller.start (/app/lib/controller.ts:109:27)
    at start (/app/index.js:107:5)
```

pixolio, 2024-04-05:
hi @Koenkk ,

i have tried it sometimes, but it doesn't work. today I switched the logging to "debug" and this are the messages from the log file (only the entries, when trying to connect the PHILIPS Hue Dymera wall lamp):

after the last entry the zigbee2mqtt instance crashed (error message down in this posing)

```
Info 2024-04-05 17:00:17Device '0x001788010de23e6e' joined
Info 2024-04-05 17:00:18MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x001788010de23e6e","ieee_address":"0x001788010de23e6e"},"type":"device_joined"}'
Info 2024-04-05 17:00:18MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x001788010de23e6e"},"type":"device_connected"}'
Info 2024-04-05 17:00:18Starting interview of '0x001788010de23e6e'
Info 2024-04-05 17:00:18MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x001788010de23e6e","ieee_address":"0x001788010de23e6e","status":"started"},"type":"device_interview"}'
Info 2024-04-05 17:00:18MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x001788010de23e6e"},"type":"pairing"}'
Debug 2024-04-05 17:00:18Received MQTT message on 'homeassistant/sensor/solarleistung/state' with data '582.13'
Debug 2024-04-05 17:00:18Received Zigbee message from '0x001788010de23e6e', type 'readResponse', cluster 'genBasic', data '{"modelId":"LCW004"}' from endpoint 11 with groupID null
Debug 2024-04-05 17:00:18Skipping message, still interviewing
Debug 2024-04-05 17:00:18Received Zigbee message from '0x001788010de23e6e', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"Signify Netherlands B.V."}' from endpoint 11 with groupID null
Debug 2024-04-05 17:00:18Skipping message, still interviewing
Debug 2024-04-05 17:00:18Received Zigbee message from '0x001788010de23e6e', type 'readResponse', cluster 'genBasic', data '{"powerSource":1}' from endpoint 11 with groupID null
Debug 2024-04-05 17:00:18Skipping message, still interviewing
Debug 2024-04-05 17:00:19Received Zigbee message from '0x001788010de23e6e', type 'readResponse', cluster 'genBasic', data '{"zclVersion":8}' from endpoint 11 with groupID null
Debug 2024-04-05 17:00:19Skipping message, still interviewing
Debug 2024-04-05 17:00:19Received Zigbee message from '0x001788010de23e6e', type 'readResponse', cluster 'genBasic', data '{"appVersion":2}' from endpoint 11 with groupID null
Debug 2024-04-05 17:00:19Skipping message, still interviewing
Debug 2024-04-05 17:00:19Received Zigbee message from '0x001788010de23e6e', type 'readResponse', cluster 'genBasic', data '{"stackVersion":1}' from endpoint 11 with groupID null
Debug 2024-04-05 17:00:19Skipping message, still interviewing
Debug 2024-04-05 17:00:19Received Zigbee message from '0x001788010de23e6e', type 'readResponse', cluster 'genBasic', data '{"hwVersion":0}' from endpoint 11 with groupID null
Debug 2024-04-05 17:00:19Skipping message, still interviewing
Debug 2024-04-05 17:00:19Received Zigbee message from '0x001788010de23e6e', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20230609"}' from endpoint 11 with groupID null
Debug 2024-04-05 17:00:19Skipping message, still interviewing
Debug 2024-04-05 17:00:19Received Zigbee message from '0x001788010de23e6e', type 'readResponse', cluster 'genBasic', data '{"swBuildId":"1.109.1"}' from endpoint 11 with groupID null
Debug 2024-04-05 17:00:19Skipping message, still interviewing
Debug 2024-04-05 17:00:19Received Zigbee message from '0x001788010de23e6e', type 'readResponse', cluster 'genBasic', data '{"hwVersion":0}' from endpoint 12 with groupID null
Debug 2024-04-05 17:00:19Skipping message, still interviewing
```

here the error log from crash /  automatic restart from z2m:

```
debug 2024-04-05 17:00:33: Loaded state from file /config/zigbee2mqtt/state.json
info  2024-04-05 17:00:33: Logging to console and directory: '/config/zigbee2mqtt/log/2024-04-05.17-00-32' filename: log.txt
debug 2024-04-05 17:00:33: Removing old log directory '/config/zigbee2mqtt/log/2024-04-03.22-30-01'
info  2024-04-05 17:00:33: Starting Zigbee2MQTT version 1.36.1 (commit #unknown)
info  2024-04-05 17:00:33: Starting zigbee-herdsman (0.40.3)
debug 2024-04-05 17:00:33: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[149,160,25,51,72,182,67,88],"networkKey":"HIDDEN","panID":27609},"serialPort":{"adapter":"deconz","path":"/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2408727-if00"}}'
error 2024-04-05 17:00:34: Failed to start zigbee
error 2024-04-05 17:00:34: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
error 2024-04-05 17:00:34: Exiting...
error 2024-04-05 17:00:34: AssertionError [ERR_ASSERTION]: 'LCW004' has multiple 'ota', this is not allowed
    at processExtensions (/app/node_modules/zigbee-herdsman-converters/src/index.ts:124:28)
    at prepareDefinition (/app/node_modules/zigbee-herdsman-converters/src/index.ts:173:18)
    at findDefinition (/app/node_modules/zigbee-herdsman-converters/src/index.ts:285:16)
    at Object.findByDevice (/app/node_modules/zigbee-herdsman-converters/src/index.ts:257:22)
    at Device.resolveDefinition (/app/lib/model/device.ts:36:31)
    at Zigbee.start (/app/lib/zigbee.ts:69:13)
    at Controller.start (/app/lib/controller.ts:109:27)
    at start (/app/index.js:107:5)

```

for using the z2m instance, I had manually remove the entries from the "new, faulty" lamp.

Could you please help me?

thx and greetings from Germany,
Stefan

Koenkk, 2024-04-06:
Managed to reproduce the issue and fixed it!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-04-06 by Koenkk