Koenkk/zigbee2mqtt #22062: RGB bulbs only allow white balance after last update

Whsitler, 2024-04-03:
### What happened?

All the Sengled bulbs in the house now only have brightness and color temp after the last update.

it seems like the changes in 
https://github.com/Koenkk/zigbee-herdsman-converters/commit/ed7ed68b9d08a50ce0507183b7daa27b031cc07b
need to be rolled back

### What did you expect to happen?

I expected the new update not to take features away

### Zigbee2MQTT version

1.36.1

### Adapter firmware version

20221226

### Adapter

smlight slzb-06

### Setup

HAOS on a hp sff pc

mrskycriper, 2024-04-03:
Fixed in https://github.com/Koenkk/zigbee-herdsman-converters/pull/7317

labanex, 2024-04-06:
I use over 60 Innr bulbs (E26 bulb RGBW (AE 280 C)) for all of my regular bulbs, works great. I do have four E12 bulbs by Sengled for certain lamps around the home (E1F-N5E). Just one day, a few days ago, all of them loss color control. The brightness, color temp works just fine. The color is not working. It appears that this is a known issue. Anything I can do to help, if you need log files, let me know. I do not know which logs to get, so please point me in the right direction. I am up to date on HAOS and plug-ins. I look forward to the update on this issue to fix my E12 bulbs. 
Zigbee2MQTT: 1.36.1-1
Core: 2024.4.1
Supervisor: 2024.03.1
Operating System: 12.1
Frontend: 20240404.1

Thanks to Zigbee2MQTT and HA team for all you do. Amazing work here.
BTW, I will be sending 5 euro for the cause. When I get paid, get ready to go to Starbucks because I will send 25. Love the work you do. It is an amazing project. Keep up the good work.

closed 2024-11-03 by github-actions[bot]