Koenkk/zigbee2mqtt #22068: Nodon SIN-4-FP-21 ends up with missing genBasic

MichaelMure, 2024-04-03:
### What happened?

Other people having this issue at the bottom of https://github.com/Koenkk/zigbee2mqtt/issues/19169

I have 3 `SIN-4-FP-21` from Nodon. Two works perfectly, but with the third I had timeout error when trying to change the pilot wire mode. I don't have the exact error anymore, but similar to `{"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 32138 - 242 - 165 - 6 - 11 after 10000ms)'` (taken from that other issue).

I had a look at the `database.db` file (third device is the broken one):
```

{"id":32,"type":"Router","ieeeAddr":"0xf4b3b1fffee51b9f","nwkAddr":34985,"manufId":4747,"manufName":"NodOn","powerSource":"Mains (single phase)","modelId":"SIN-4-FP-21","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,3,4,5,6,1794,4096,64512],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"SIN-4-FP-21","manufacturerName":"NodOn","powerSource":1,"zclVersion":8,"appVersion":3,"stackVersion":103,"hwVersion":52,"dateCode":"2021","swBuildId":"3.0.0-1.4.4"}},"seMetering":{"attributes":{"multiplier":1,"divisor":1000,"instantaneousDemand":1,"currentSummDelivered":[0,186577]}},"genIdentify":{"attributes":{"identifyTime":0}},"manuSpecificUbisysDeviceSetup":{"attributes":{"mode":5}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b001caa8446","endpointID":1},{"cluster":3,"type":"endpoint","deviceIeeeAddress":"0x00124b001caa8446","endpointID":1},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001caa8446","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b001caa8446","endpointID":1},{"cluster":64512,"type":"endpoint","deviceIeeeAddress":"0x00124b001caa8446","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":1,"maxRepIntval":3600,"repChange":0,"manufacturerCode":null},{"cluster":1794,"attrId":1024,"minRepIntval":5,"maxRepIntval":3600,"repChange":1,"manufacturerCode":null},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1],"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":102,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":3,"stackVersion":103,"hwVersion":52,"dateCode":"2021","swBuildId":"3.0.0-1.4.4","zclVersion":8,"interviewCompleted":true,"meta":{"configured":1520543266},"lastSeen":1712177167846}
{"id":33,"type":"Router","ieeeAddr":"0xf082c0fffe2071e4","nwkAddr":38222,"manufId":4747,"manufName":"NodOn","powerSource":"Mains (single phase)","modelId":"SIN-4-FP-21","epList":[242,1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,3,4,5,6,1794,4096,64512],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"SIN-4-FP-21","manufacturerName":"NodOn","powerSource":1,"zclVersion":8,"appVersion":3,"stackVersion":103,"hwVersion":52,"dateCode":"2021","swBuildId":"3.0.0-1.4.4"}},"seMetering":{"attributes":{"multiplier":1,"divisor":1000,"instantaneousDemand":1,"currentSummDelivered":[0,411]}},"manuSpecificUbisysDeviceSetup":{"attributes":{"mode":3}},"genIdentify":{"attributes":{"identifyTime":0}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b001caa8446","endpointID":1},{"cluster":3,"type":"endpoint","deviceIeeeAddress":"0x00124b001caa8446","endpointID":1},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001caa8446","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b001caa8446","endpointID":1},{"cluster":64512,"type":"endpoint","deviceIeeeAddress":"0x00124b001caa8446","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":1,"maxRepIntval":3600,"repChange":0,"manufacturerCode":null},{"cluster":1794,"attrId":1024,"minRepIntval":5,"maxRepIntval":3600,"repChange":1,"manufacturerCode":null},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1],"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":102,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":3,"stackVersion":103,"hwVersion":52,"dateCode":"2021","swBuildId":"3.0.0-1.4.4","zclVersion":8,"interviewCompleted":true,"meta":{"configured":1520543266},"lastSeen":1712177165864}
{"id":34,"type":"Router","ieeeAddr":"0xf082c0fffe180179","nwkAddr":45975,"manufId":4747,"manufName":"NodOn","powerSource":"Mains (single phase)","modelId":"SIN-4-FP-21","epList":[242,1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,3,4,5,6,1794,4096,64512],"outClusterList":[25],"clusters":{"genIdentify":{"attributes":{"identifyTime":0}},"seMetering":{"attributes":{"multiplier":1,"divisor":1000,"currentSummDelivered":[0,39415],"instantaneousDemand":0}},"manuSpecificUbisysDeviceSetup":{"attributes":{"mode":3}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b001caa8446","endpointID":1},{"cluster":3,"type":"endpoint","deviceIeeeAddress":"0x00124b001caa8446","endpointID":1},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001caa8446","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b001caa8446","endpointID":1},{"cluster":64512,"type":"endpoint","deviceIeeeAddress":"0x00124b001caa8446","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":1,"maxRepIntval":3600,"repChange":0,"manufacturerCode":null},{"cluster":1794,"attrId":1024,"minRepIntval":5,"maxRepIntval":3600,"repChange":1,"manufacturerCode":null},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1],"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":102,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":3,"stackVersion":103,"hwVersion":52,"dateCode":"2021","swBuildId":"3.0.0-1.4.4","zclVersion":8,"interviewCompleted":true,"meta":{"configured":1520543266},"lastSeen":1712177141699}
```

As you can see, `genBasic` is missing in the third device. After copying that section from other and restarting Zigbee2MQTT, the device worked flawlessly.

I've no idea what let to that misconfiguration.

### What did you expect to happen?

All three device fully working.

### Zigbee2MQTT version

1.36.0 commit: 86ed71c

### Adapter firmware version

20220219

### Adapter

zzh!

### Setup

docker on a sinology nas

MichaelMure, 2024-04-04:
Side note, but the `"epList":[1,242]` vs `"epList":[242,1]` difference is also suspicious. It's very similar to https://github.com/Koenkk/zigbee2mqtt/issues/15962 which got fixed with https://github.com/Koenkk/zigbee-herdsman-converters/commit/961fa6f47016b06e604202afe2e869c2f3548abd.

Chartreusito, 2024-04-26:
I have the same problem on one recently purchased module, here is the error log :

`error 2024-04-26 13:13:22Publish 'set' 'pilot_wire_mode' to 'XXXXXXXXXXXX' failed: 'Error: ZCL command 0xbc026efffeae314a/242 manuSpecificNodOnPilotWire.setMode({"mode":1}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4747,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 42272 - 242 - 33 - 64512 - 11 after 10000ms)'`

closed 2024-11-24 by github-actions[bot]

Mortimerp9, 2025-10-19:
I had this exact same issue, updated z2m to 2.0.0 and had to repair my devices, my NoDon SIN-4-FP-21 wouldn't work properly, always timing out, even if they had paired and had good connection. After editing the db file to add the genBasic as per above, they work properly.