Koenkk/zigbee2mqtt #22071: OTA: Firmware on (SNZB-06P)

Mahalov24, 2024-04-04:
### What happened?

SNZB-06P
Firmware version = 1.0.6

is:open OTA: Firmware "0x0ceff6fffe5e086d" (SNZB-06P) is newer than the latest firmware available online.

The latest firmware available online = 1.0.5

### What did you expect to happen?

Update firmware SNZB-06P

### How to reproduce it (minimal and precise)

zigbee2mqtt  > OTA > SNZB-06P Check for updates 

### Zigbee2MQTT version

 1.36.1

### Adapter firmware version

 7.3.1.0 build 176

### Adapter

 EZSP v12

### Setup

home asisstant intel mini pc

### Debug log

Info 2024-04-04 12:58:56Checking if update available for 'SNZB-06P-K'
Warning 2024-04-04 12:58:59OTA: Firmware on '0x0ceff6fffe5e086d' (SNZB-06P) is newer than latest firmware online.
Info 2024-04-04 12:58:59No update available for 'SNZB-06P-K'
Info 2024-04-04 12:58:59MQTT publish: topic 'zigbee2mqtt/SNZB-06P-K', payload '{"illumination":"bright","linkquality":180,"occupancy":false,"occupancy_sensitivity":"high","occupancy_timeout":30,"update":{"installed_version":4102,"latest_version":4101,"state":"idle"},"update_available":null}'
Info 2024-04-04 12:58:59MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"SNZB-06P-K","updateAvailable":false},"status":"ok","transaction":"r2muf-3"}'

oneevil, 2024-04-04:
I'm seeing the same problem myself

gstag, 2024-04-27:
I am experiencing the same problem as well.

i-m-d, 2024-04-27:
I am experiencing the same problem as well.

idogrf, 2024-05-06:
I'm having the same issue. The device came with version 1.0.6 and can't be configured properly because of `(SNZB-06P) is newer than latest firmware online.`

Is there a way to flash an earlier version?

umtauscher, 2024-05-18:
I am struggeling about the same thing. I just received two units, one with 1.03 which I could update to 1.05 working quite well now.
The other one came with 1.06, was very difficult to include and was not working. I could not downgrade it because of the above error. 
After I powered it off and on after inclusion I finally could make it work.

LeeStevens318, 2024-06-24:
I have just received two on firmware 1.0.6, these are also only exposing the occupancy and not the illumination etc. 

[image]
[image]

cjlapao, 2024-10-23:
I have several ones, all with 1.0.5 and only one of them is reporting the update. But even when I try to update it, it fails
[image]

closed 2025-01-06 by github-actions[bot]

andrei-lazarov, 2025-11-25:
@Reifen234 Yes. It's available in the [zigbee-OTA repo](https://github.com/Koenkk/zigbee-OTA/blob/master/not-in-manifest-images/Sonoff/snzb-06p_v1.0.5.ota).

schlaubstar, 2025-12-31:
@andrei-lazarov : I did use your template for that device and tried to downgrade .... took me a while to understand that I have to restart Z2M prior the index-file gets evaluated but when I do I fail with:

z2m: Update of 'mySNZB-06PDevice' failed (Update failed with reason: INVALID_IMAGE)

No matter if I store it locally or on a public accessible webserver.

andrei-lazarov, 2026-01-03:
@schlaubstar sorry, I forgot to reply.
Indeed the 'restart required' indicator is subtle with the new UI.

Apply this index and restart Z2M. Here I just added the force flag.
https://raw.githubusercontent.com/andrei-lazarov/zigbee-OTA/refs/heads/sonoff/sonoff_downgrade_force.json
Check for updates from Z2M UI (no need to use downgrade)

If it doesn't work, try again with this index. I edited the OTA file so the device thinks it's updating to v1.0.7
https://raw.githubusercontent.com/andrei-lazarov/zigbee-OTA/refs/heads/sonoff/sonoff_downgrade_trick.json

If it still doesn't work, give me some **debug** logs with those operations.
> The 'log level config' setting enables generation of debug logs, but only **info** logs are sent to the UI (to avoid MQTT traffic).
You have to view **debug** logs on the host machine: HA > Add-ons > Z2M > Log, or `docker container logs`.
Disable debug logs when you are done.

Also you can let SONOFF know if v1.0.6 is buggy. They've been active here lately.