Koenkk/zigbee2mqtt #22073: Philips Huhe dimmer switch RWL021 fail to bind on/off after upgrade to 1.36.1-1

aervig, 2024-04-04:
### What happened?

Philips Huhe dimmer switch RWL021 fail to bind on/off after upgrade to 1.36.1-1.
After upgrading is it not logner possible to turn lights on and off with the remote, so it seems like something with the binding has gone wrong.

It still works to dim the light up and down.

### Zigbee2MQTT version

1.36.1-1 

### Adapter firmware version

 0x26580700

### Adapter

ConBee2/RaspBee2

### Setup

Home Assistant OS

### Debug log

Info 2024-04-04 10:34:53MQTT publish: topic 'zigbee2mqtt/Remote Soverom/action', payload 'off_press'
Info 2024-04-04 10:34:53MQTT publish: topic 'zigbee2mqtt/Remote Soverom', payload '{"action":"off_press_release","action_duration":null,"battery":100,"brightness":255,"counter":1,"linkquality":255,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Info 2024-04-04 10:34:53MQTT publish: topic 'zigbee2mqtt/Remote Soverom', payload '{"action":"","action_duration":null,"battery":100,"brightness":255,"counter":1,"linkquality":255,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Info 2024-04-04 10:34:53MQTT publish: topic 'zigbee2mqtt/Remote Soverom/action', payload 'off_press_release'

GSzabados, 2024-04-04:
I am.havung the same issue with the RWL021, but dimming is not working either.

GSzabados, 2024-04-04:
My guess the nested group naming issue has something to do with this issue:

https://github.com/Koenkk/zigbee2mqtt/issues/21452

It was causing issues in dev already, but might be the whole issue is there when you don't have any nesting.

mUbsta, 2024-04-04:
Just to add I'm experiencing the same issue since the same update, however it (so far) it seems to be only affecting 3 out of 6 of my switches.

GSzabados, 2024-04-04:
> 3 out of 6 of my switches.

What is the difference between the working and not working ones?

mUbsta, 2024-04-04:
> > 3 out of 6 of my switches.
> 
> What is the difference between the working and not working ones?

I'm trying to work this out.  I will update here if I find anything.

aervig, 2024-04-04:
I also have one that still works, but the rest of them has stopped to work. I can not find any difference.. The firmware are the same, all configuration are the same, and the naming standard are the same. I have deleted the non working and tried to add them again, but I still get the same problem.

GSzabados, 2024-04-04:
What are you trying to control with the not working and  working ones? Hue bulbs? Ikea bulbs? Any other?

aervig, 2024-04-04:
Hue bulbs on all of them. I have eaven tried to bind the non working remote to the bulb group that the working one are binding to.

xoatrash, 2024-04-04:
I have the exact same problem with the Hue wall switch module since 1.36.1-1. :(

mUbsta, 2024-04-04:
> What are you trying to control with the not working and working ones? Hue bulbs? Ikea bulbs? Any other?

I have a group with only Linkind ZL00030014 (https://www.zigbee2mqtt.io/devices/ZL00030014.html#linkind-zl00030014) bulbs showing the same symptoms.

All of the other groups showing the same issues are all wholly hue based.

GSzabados, 2024-04-04:
I use the dimmer switch with IKEA bulbs. No control over the bulbs.

mUbsta, 2024-04-04:
> I use the dimmer switch with IKEA bulbs. No control over the bulbs. White tunable bulbs. Any chance that the working ones are color bulbs?

All white hue - Working
All white linkind - Not Working
All colour hue - Not working
another all colour hue - Not working
White & Colour hue - Working
another white & Colour hue - Not working

I have another 2 groups I've not tested that are all colour but..   I don't think white/colour is a thing here.

sagacity, 2024-04-04:
The second button of the Hue Wall Switch has also stopped working due to this change.

hvorragend, 2024-04-04:
I have noticed that all my Hue wall switches (model: **RDM001**) are behaving strangely. The events are all sent twice. After a reset, it seems to work again. 

My **RWL022** dimmer switch had no problems. 
But I had to reset the **RWL021** models and create the bindings again.

xoatrash, 2024-04-04:
> But I had to reset the **RWL021** models and create the bindings again.

I also did, the it worked, but just for a short time.

aervig, 2024-04-04:
I tried a factory reset, but that made the diming also to stop working.

hvorragend, 2024-04-04:
> I also did, the it worked, but just for a short time.

I had the same experience. It didn't last very long. I'm really glad that I'm not alone with this problem. I couldn't find any messages for days.

mUbsta, 2024-04-04:
I've tried rolling back to 1.36.0 however this issue persists :(

hvorragend, 2024-04-04:
> I've tried rolling back to 1.36.0 however this issue persists :(

Same here

aervig, 2024-04-04:
> I've tried rolling back to 1.36.0 however this issue persists :(

Same here, it's a nightmare to don't have light switches anymore... Hope someone find a fix for this very soon...

grafalex82, 2024-04-04:
I do not have any Philips or HUE devices. Can someone post me a log from a working Z2M version so that I could compare?

Also, what device is generating these actions?

aervig, 2024-04-04:
> I do not have any Philips or HUE devices. Can someone post me a log from a working Z2M version so that I could compare?
> 
> Also, what device is generating these actions?

I don't have a working Z2M version any more, as it don't help to downgrade. However one of the remote devices is still working, so maybe it will help with logfiles from that and one of the non working devices?

What logfile do you want me to grab?

aervig, 2024-04-04:
[log_hue.txt](https://github.com/Koenkk/zigbee2mqtt/files/14876217/log_hue.txt)

Remote Gjesterom is still working
Remote Stuetak is not working

`Debug 2024-04-04 21:18:30Received Zigbee message from 'Gjesterom_3', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":222}' from endpoint 11 with groupID null
Info 2024-04-04 21:18:30MQTT publish: topic 'zigbee2mqtt/Gruppe_Gjesterom', payload '{"brightness":222,"color":{"h":36,"hue":36,"s":67,"saturation":67,"x":0.4164,"y":0.396},"color_mode":"color_temp","color_temp":302,"state":"ON"}'
Info 2024-04-04 21:18:30MQTT publish: topic 'zigbee2mqtt/Gjesterom_3', payload '{"brightness":222,"color":{"h":36,"hue":36,"s":67,"saturation":67,"x":0.4164,"y":0.396},"color_mode":"color_temp","color_temp":302,"linkquality":255,"power_on_behavior":null,"state":"OFF","update":{"installed_version":16784922,"latest_version":16785162,"state":"available"},"update_available":true}'
Debug 2024-04-04 21:18:30Received Zigbee message from 'Remote Gjesterom', type 'commandHueNotification', cluster 'manuSpecificPhilips', data '{"button":4,"time":2,"type":2,"unknown1":3145728,"unknown2":0}' from endpoint 2 with groupID null
Info 2024-04-04 21:18:30MQTT publish: topic 'zigbee2mqtt/Remote Gjesterom', payload '{"action":"off-press","action_duration":null,"battery":100,"brightness":255,"counter":1,"duration":0,"linkquality":255,"update":{"installed_version":1107324829,"latest_version":1107324829,"state":"idle"},"update_available":false}'
Info 2024-04-04 21:18:30MQTT publish: topic 'zigbee2mqtt/Remote Gjesterom', payload '{"action":"","action_duration":null,"battery":100,"brightness":255,"counter":1,"linkquality":255,"update":{"installed_version":1107324829,"latest_version":1107324829,"state":"idle"},"update_available":false}'
Info 2024-04-04 21:18:30MQTT publish: topic 'zigbee2mqtt/Remote Gjesterom/action', payload 'off-press'
Debug 2024-04-04 21:18:30Received Zigbee message from 'Gjesterom_5', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 11 with groupID null
Info 2024-04-04 21:18:30MQTT publish: topic 'zigbee2mqtt/Gjesterom_5', payload '{"brightness":253,"color":{"h":36,"hue":36,"s":67,"saturation":67,"x":0.4164,"y":0.396},"color_mode":"color_temp","color_temp":302,"linkquality":255,"power_on_behavior":null,"state":"OFF","update":{"installed_version":16785162,"latest_version":16785162,"state":"idle"},"update_available":false}'
Debug 2024-04-04 21:18:30Received Zigbee message from 'Gjesterom_5', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":114}' from endpoint 11 with groupID null
Info 2024-04-04 21:18:30MQTT publish: topic 'zigbee2mqtt/Gruppe_Gjesterom', payload '{"brightness":114,"color":{"h":36,"hue":36,"s":67,"saturation":67,"x":0.4164,"y":0.396},"color_mode":"color_temp","color_temp":302,"state":"ON"}'
Info 2024-04-04 21:18:30MQTT publish: topic 'zigbee2mqtt/Gjesterom_5', payload '{"brightness":114,"color":{"h":36,"hue":36,"s":67,"saturation":67,"x":0.4164,"y":0.396},"color_mode":"color_temp","color_temp":302,"linkquality":255,"power_on_behavior":null,"state":"OFF","update":{"installed_version":16785162,"latest_version":16785162,"state":"idle"},"update_available":false}'
Debug 2024-04-04 21:18:31Received Zigbee message from 'Gjesterom_2', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 11 with groupID null
Info 2024-04-04 21:18:31MQTT publish: topic 'zigbee2mqtt/Gjesterom_2', payload '{"brightness":253,"color":{"h":265,"hue":265,"s":99,"saturation":99,"x":0.2778,"y":0.1105},"color_mode":"xy","color_temp":438,"color_temp_startup":500,"linkquality":255,"power_on_behavior":null,"state":"OFF","update":{"installed_version":16785162,"latest_version":16785162,"state":"idle"},"update_available":false}'
Debug 2024-04-04 21:18:31Received Zigbee message from 'Gjesterom_2', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":7}' from endpoint 11 with groupID null
Info 2024-04-04 21:18:31MQTT publish: topic 'zigbee2mqtt/Gruppe_Gjesterom', payload '{"brightness":7,"color":{"h":36,"hue":36,"s":67,"saturation":67,"x":0.4164,"y":0.396},"color_mode":"color_temp","color_temp":302,"state":"ON"}'
Info 2024-04-04 21:18:31MQTT publish: topic 'zigbee2mqtt/Gjesterom_2', payload '{"brightness":7,"color":{"h":265,"hue":265,"s":99,"saturation":99,"x":0.2778,"y":0.1105},"color_mode":"xy","color_temp":438,"color_temp_startup":500,"linkquality":255,"power_on_behavior":null,"state":"OFF","update":{"installed_version":16785162,"latest_version":16785162,"state":"idle"},"update_available":false}'
Debug 2024-04-04 21:18:31Received Zigbee message from 'Gjesterom_4', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 11 with groupID null
Info 2024-04-04 21:18:31MQTT publish: topic 'zigbee2mqtt/Gjesterom_4', payload '{"brightness":253,"color":{"h":36,"hue":36,"s":67,"saturation":67,"x":0.4164,"y":0.396},"color_mode":"color_temp","color_temp":302,"linkquality":255,"power_on_behavior":null,"state":"OFF","update":{"installed_version":16785162,"latest_version":16785162,"state":"idle"},"update_available":false}'
Debug 2024-04-04 21:18:31Received Zigbee message from 'Gjesterom_1', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 11 with groupID null
Info 2024-04-04 21:18:31MQTT publish: topic 'zigbee2mqtt/Gruppe_Gjesterom', payload '{"brightness":7,"color":{"h":36,"hue":36,"s":67,"saturation":67,"x":0.4164,"y":0.396},"color_mode":"color_temp","color_t [... cut ...]
Debug 2024-04-04 21:18:42Received Zigbee message from 'Stuetak 7', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":16784922,"imageType":276,"manufacturerCode":4107}' from endpoint 11 with groupID null
Debug 2024-04-04 21:18:42Device 'Stuetak 7' requested OTA
Error 2024-04-04 21:18:45Failed to poll onOff from Gjesterom_5
Debug 2024-04-04 21:18:52Received Zigbee message from 'Stuetak 7', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":16784922,"imageType":276,"manufacturerCode":4107}' from endpoint 11 with groupID null
Debug 2024-04-04 21:18:52`

aervig, 2024-04-04:
I can see in the log file that the remote that works get groupID 1, and the remotes that don't work get groupID null

`Debug 2024-04-04 21:27:05Received Zigbee message from 'Remote Stuetak', type 'commandOffWithEffect', cluster 'genOnOff', data '{"effectid":0,"effectvariant":0}' from endpoint 1 with groupID null
Debug 2024-04-04 21:27:37Received Zigbee message from 'Remote Gjesterom', type 'commandOffWithEffect', cluster 'genOnOff', data '{"effectid":0,"effectvariant":0}' from endpoint 1 with groupID 1`

xoatrash, 2024-04-04:
This is what I get when I try to bind or unbind:

`2024-04-04 21:44:12Failed to bind cluster 'genOnOff' from 'Philips Hue Wall Switch Badezimmer' to 'Badezimmer' (Error: Bind 0x001788010cc43b67/1 genOnOff from '2' failed ({"address":37721,"clusterId":32801,"sequence":144} after 10000ms))
Error 2024-04-04 21:44:22Failed to unbind cluster 'genPowerCfg' from 'Philips Hue Wall Switch Badezimmer' to 'Coordinator' (Error: Unbind 0x001788010cc43b67/1 genPowerCfg from '0xe0798dfffe999e0c/1' failed ({"address":37721,"clusterId":32802,"sequence":146} after 10000ms))
Error 2024-04-04 21:44:32Failed to bind cluster 'genLevelCtrl' from 'Philips Hue Wall Switch Badezimmer' to 'Badezimmer' (Error: Bind 0x001788010cc43b67/1 genLevelCtrl from '2' failed ({"address":37721,"clusterId":32801,"sequence":149} after 10000ms))
Error 2024-04-04 21:44:32Failed to bind
Error 2024-04-04 21:44:42Failed to unbind cluster 'genOnOff' from 'Philips Hue Wall Switch Badezimmer' to 'Coordinator' (Error: Unbind 0x001788010cc43b67/1 genOnOff from '0xe0798dfffe999e0c/1' failed ({"address":37721,"clusterId":32802,"sequence":151} after 10000ms))
Error 2024-04-04 21:44:52Failed to unbind cluster 'manuSpecificPhilips' from 'Philips Hue Wall Switch Badezimmer' to 'Coordinator' (Error: Unbind 0x001788010cc43b67/1 manuSpecificPhilips from '0xe0798dfffe999e0c/1' failed ({"address":37721,"clusterId":32802,"sequence":153} after 10000ms))`

mUbsta, 2024-04-04:
> This is what I get when I try to bind or unbind

This is what I see too.

xoatrash, 2024-04-04:
Glad we are all having the same issue. My wife is already very angry, because the bathroom light wall switch isn‘t working anymore. 😬

grafalex82, 2024-04-04:
Guys, I am really willing to help, but from the logs above it does not look like my changes could lead to these effects. My changes were related to parsing an MQTT message, while logs above do not mention this at all.

What is worrying me in these logs is "Received Zigbee message ... with groupID null" statement. For me this looks like incorrect processing of incoming Zigbee message.

aervig, 2024-04-04:
> changes were related to parsing an MQTT message, wh

I also think it is really strange and scary that it didn't help to downgrade Zigbee2mqtt to solve the problem.
It seems to me that the new update wrote some new configuration to the ConbeeII stick, that don't reverse to the old configuration again.

aervig, 2024-04-04:
Here is also a log from after I have factory reset the Hue Remote and joining it to Z2M again:

`Debug 2024-04-04 23:04:26Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":null,"time":254,"transaction":"vnyi2-2","value":true}'
Info 2024-04-04 23:04:26Zigbee: allowing new devices to join.
Info 2024-04-04 23:04:26MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"vnyi2-2"}'
Debug 2024-04-04 23:04:31Received MQTT message on 'homeassistant/sensor/znet/73_1_0/state' with data '{"lastUpdated": 1712264670, "value": "-1.7"}'
Debug 2024-04-04 23:04:31Received MQTT message on 'homeassistant/sensor/znet/73_2_0/state' with data '{"lastUpdated": 1712264670, "value": "81"}'
Info 2024-04-04 23:04:33Device '0x00178801086c3cf4' joined
Info 2024-04-04 23:04:33MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00178801086c3cf4","ieee_address":"0x00178801086c3cf4"},"type":"device_joined"}'
Info 2024-04-04 23:04:33MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x00178801086c3cf4"},"type":"device_connected"}'
Info 2024-04-04 23:04:33Starting interview of '0x00178801086c3cf4'
Info 2024-04-04 23:04:33MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00178801086c3cf4","ieee_address":"0x00178801086c3cf4","status":"started"},"type":"device_interview"}'
Info 2024-04-04 23:04:33MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x00178801086c3cf4"},"type":"pairing"}'
Info 2024-04-04 23:04:35Successfully interviewed '0x00178801086c3cf4', device has successfully been paired
Info 2024-04-04 23:04:35Device '0x00178801086c3cf4' is supported, identified as: Philips Hue dimmer switch (324131092621)
Info 2024-04-04 23:04:35MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Hue dimmer switch","exposes":[{"access":1,"category":"diagnostic","description":"Remaining battery in %, can take up to 24 hours before reported","label":"Battery","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"description":"Triggered action (e.g. a button click)","label":"Action","name":"action","property":"action","type":"enum","values":["on_press","on_press_release","on_hold","on_hold_release","up_press","up_press_release","up_hold","up_hold_release","down_press","down_press_release","down_hold","down_hold_release","off_press","off_press_release","off_hold","off_hold_release"]},{"access":1,"category":"diagnostic","description":"Triggered action duration in seconds","label":"Action duration","name":"action_duration","property":"action_duration","type":"numeric","unit":"s"},{"access":1,"category":"diagnostic","description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"324131092621","options":[{"access":2,"description":"Set to false to disable the legacy integration (highly recommended), will change structure of the published payload (default true).","label":"Legacy","name":"legacy","property":"legacy","type":"binary","value_off":false,"value_on":true},{"access":2,"description":"Simulate a brightness value. If this device provides a brightness_move_up or brightness_move_down action it is possible to specify the update interval and delta. The action_brightness_delta indicates the delta for each interval. Only works when legacy is false.","features":[{"access":2,"description":"Delta per interval, 20 by default","label":"Delta","name":"delta","property":"delta","type":"numeric","value_min":0},{"access":2,"description":"Interval duration","label":"Interval","name":"interval","property":"interval","type":"numeric","unit":"ms","value_min":0}],"label":"Simulated brightness","name":"simulated_brightness","property":"simulated_brightness","type":"composite"}],"supports_ota":true,"vendor":"Philips"},"friendly_name":"0x00178801086c3cf4","ieee_address":"0x00178801086c3cf4","status":"successful","supported":true},"type":"device_interview"}'
Info 2024-04-04 23:04:35Configuring '0x00178801086c3cf4'
Info 2024-04-04 23:04:35MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Hue dimmer switch","friendly_name":"0x00178801086c3cf4","model":"324131092621","supported":true,"vendor":"Philips"},"type":"pairing"}'
Info 2024-04-04 23:04:35MQTT publish: topic 'homeassistant/sensor/0x00178801086c3cf4/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00178801086c3cf4"],"manufacturer":"Philips","model":"Hue dimmer switch (324131092621)","name":"0x00178801086c3cf4","sw_version":"6.1.1.28573","via_device":"zigbee2mqtt_bridge_0x00212effff080360"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/0x00178801086c3cf4","object_id":"0x00178801086c3cf4_battery","origin":{"name":"Zigbee2MQTT","sw":"1.36.0","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/0x00178801086c3cf4","unique_id":"0x00178801086c3cf4_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
[... cut ...]
Info 2024-04-04 23:05:24MQTT publish: topic 'zigbee2mqtt/0x00178801086c3cf4', payload '{"action":"on-press","action_duration":null,"battery":100,"counter":1,"duration":0,"linkquality":231,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Info 2024-04-04 23:05:24MQTT publish: topic 'zigbee2mqtt/0x00178801086c3cf4', payload '{"action":"","action_duration":null,"battery":100,"counter":1,"linkquality":231,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Info 2024-04-04 23:05:24MQTT publish: topic 'zigbee2mqtt/0x00178801086c3cf4/action', payload 'on-press'`

mUbsta, 2024-04-04:
> I also think it is really strange and scary that it didn't help to downgrade Zigbee2mqtt to solve the problem. It seems to me that the new update wrote some new configuration to the ConbeeII stick, [...]

I'm not using ConbeeII.  https://smartlight.me/smart-home-devices/zigbee-devices/zigbee-coordinator-v4-cc2652p is what I'm using here.

aervig, 2024-04-05:
I went out and bought a new IKEA STYRBAR remote today to get a lightswitch, but I have the same problem with that one. It looks like it's either Zigbee2mqtt or me that have to move out of the house if we don't get any fix for this.

```
Debug 2024-04-05 13:10:26Received Zigbee message from '0x287681fffe26ef8b', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID null
Info 2024-04-05 13:10:26MQTT publish: topic 'zigbee2mqtt/0x287681fffe26ef8b', payload '{"action":"off","battery":null,"linkquality":255,"update":{"installed_version":33816581,"latest_version":581,"state":"idle"},"update_available":false}'
Info 2024-04-05 13:10:26MQTT publish: topic 'zigbee2mqtt/0x287681fffe26ef8b', payload '{"action":"","battery":null,"linkquality":255,"update":{"installed_version":33816581,"latest_version":581,"state":"idle"},"update_available":false}'
Info 2024-04-05 13:10:26MQTT publish: topic 'zigbee2mqtt/0x287681fffe26ef8b/action', payload 'off'
Debug 2024-04-05 13:10:28Received Zigbee message from '0x287681fffe26ef8b', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID null
Info 2024-04-05 13:10:28MQTT publish: topic 'zigbee2mqtt/0x287681fffe26ef8b', payload '{"action":"on","battery":null,"linkquality":255,"update":{"installed_version":33816581,"latest_version":581,"state":"idle"},"update_available":false}'
Info 2024-04-05 13:10:28MQTT publish: topic 'zigbee2mqtt/0x287681fffe26ef8b', payload '{"action":"","battery":null,"linkquality":255,"update":{"installed_version":33816581,"latest_version":581,"state":"idle"},"update_available":false}'
Info 2024-04-05 13:10:28MQTT publish: topic 'zigbee2mqtt/0x287681fffe26ef8b/action', payload 'on'
```

aervig, 2024-04-05:
@Koenkk Are you able to take a look at this issue?

a8ree, 2024-04-05:
I'm not sure if this is the same issue however when I press any button I am getting action as null

```
Info 2024-04-05 20:34:27MQTT publish: topic 'zigbeehouse/Hall Upstairs Dimmer', payload '{"action":null,"action_duration":null,"battery":75.5,"last_seen":"2024-04-05T19:34:27.912Z","linkquality":51,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Debug 2024-04-05 20:34:27Device 'Hall Upstairs Dimmer' announced itself
Info 2024-04-05 20:34:28MQTT publish: topic 'zigbeehouse/bridge/event', payload '{"data":{"friendly_name":"Hall Upstairs Dimmer","ieee_address":"0x0017880106363c2c"},"type":"device_announce"}'
```

mUbsta, 2024-04-05:
So, 

Can other people verify that if you remove the binding with source endpoint 1 destination coordinator eg:-

[image]

That the binding now works direct to the lights?  Unfortunately anything other that the binding to the light i.e. HA will get nothing from Z2m sent from the light switch.

aervig, 2024-04-05:
> So,
> 
> Can other people verify that if you remove the binding with source endpoint 1 destination coordinator eg:-
> 
> <img alt="image" width="1424" src="https://private-user-images.githubusercontent.com/16826659/320144106-910d4479-9d5a-44df-a384-b94256ca26d5.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTIzNTM2MDUsIm5iZiI6MTcxMjM1MzMwNSwicGF0aCI6Ii8xNjgyNjY1OS8zMjAxNDQxMDYtOTEwZDQ0NzktOWQ1YS00NGRmLWEzODQtYjk0MjU2Y2EyNmQ1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MDUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDA1VDIxNDE0NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWI4N2VkZTY0MWRmZWIzNTcyZTc4ZWM5Y2YzNDI3ZTE4MWIzODU1MDBlYWYyYjhhNGQzYWE4NDc4MGExZWFkOTkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.wJ1F9gscVhffxFFCC9jS5BBKiH5Rvgpojjm5ae5RqUA">
> That the binding now works direct to the lights? Unfortunately anything other that the binding to the light i.e. HA will get nothing from Z2m sent from the light switch.

It fails to unbind from the coordinator for me

mUbsta, 2024-04-05:
> It fails to unbind from the coordinator for me

What if you press unbind and then press a button on the switch @aervig ?

aervig, 2024-04-05:
> > It fails to unbind from the coordinator for me
> 
> What if you press unbind and then press a button on the switch straight after @aervig ?

Thanks, that worked, and I can confirm that I can control the light with it again now.

mUbsta, 2024-04-05:
> Thanks, that worked, and I can confirm that I can control the light with it again now.

So, happy wives all round then? :)

It looks like there's possibly a problem with multiple bindings against a single source endpoint here.

GSzabados, 2024-04-05:
@mUbsta, no, not really. I have remotes which should operate the bulb and meanwhile communicate with Z2M and HA to control other things, but I have a guess where things went side way.

Looking at the recent changes, the new modern extent changes messed things up as seems to.  So tagging @mrskycriper.

I tried the trick mentioned above, and removed the coordinator from the binding, then the remote worked with the group. But once I added back the coordinator, then the group binding was not working anymore.

Koenkk, 2024-04-06:
I think this issue can be fixed by force removing the device from z2m, restarting z2m and re-pairing the device. Probably this happened due to an auto reconfigure which I've already disabled [for the new release](https://github.com/Koenkk/zigbee2mqtt/pull/22088)

mUbsta, 2024-04-06:
> I think this issue can be fixed by force removing the device from z2m, restarting z2m and re-pairing the device. Probably this happened due to an auto reconfigure which I've already disabled [for the new release](https://github.com/Koenkk/zigbee2mqtt/pull/22088)

I've tried this and still having the same issue with multiple bindings I'm afraid.  Has anyone else had any luck?

GSzabados, 2024-04-06:
I have unbinded `OnOff` and `LevelCtrl` from the Coordinator and binded to the group and to something called `default_bind_group` which has 0 members. The group works with the remote and messages are passed through to Z2M as well.

xoatrash, 2024-04-06:
> > > It fails to unbind from the coordinator for me
> > 
> > 
> > What if you press unbind and then press a button on the switch straight after @aervig ?
> 
> Thanks, that worked, and I can confirm that I can control the light with it again now.

I can confirm this worked also for me. Let‘s see for how long…

ste-fle, 2024-04-07:
> > > > It fails to unbind from the coordinator for me
> > > 
> > > 
> > > What if you press unbind and then press a button on the switch straight after @aervig ?
> > 
> > 
> > Thanks, that worked, and I can confirm that I can control the light with it again now.
> 
> I can confirm this worked also for me. Let‘s see for how long…

What exactly did you do to get it to work again? I successfully (at least according to the popups in Z2M) unbound the dimmer switch from the light group and then afterwards tried binding it again but this failed and I can't control the lights with the switch.

aervig, 2024-04-13:
> I think this issue can be fixed by force removing the device from z2m, restarting z2m and re-pairing the device. Probably this happened due to an auto reconfigure which I've already disabled [for the new release](https://github.com/Koenkk/zigbee2mqtt/pull/22088)

Still no permanent solution for this?

stigvi, 2024-04-15:
I find it very strange that this issue still exists after 2 weeks. Makes me wonder about zigbee2mqtt .....

Anyway, I am holding off upgrading to this version and I can recommend to **not** update zigbee2mqtt in the beginning of the month, but wait to the last day instead.

Tonio16, 2024-04-20:
> So,
>
> Can other people verify that if you remove the binding with source endpoint 1 destination coordinator eg:-
>
> <img alt="image" width="1424" [...]

Hello

For me it's working.

Regards

Antoine

chris777c, 2024-05-01:
Hello, 

I had remove source endpoint 1 with coordinator and clusters : levelCtrl , OnOff. The hue dimmer switch control directly the light but no information remote to z2m.

GSzabados, 2024-05-01:
> The hue dimmer switch control directly the light but no information remote to z2m.

Add binding to `default_bind_group` and there will be info in Z2M as well.

closed 2024-11-28 by github-actions[bot]