Koenkk/zigbee2mqtt #22075: 1.36.1 causes "Error: Cluster with name '64529' does not exist" for SONOFF TRVZB

Cloudworks99, 2024-04-04:
### What happened?

Since updating to 1.36.1 I get an error showing "Error: Cluster with name '64529' does not exist" for my SONOFF TRVZB thermostats.

After the first start of 1.36.1 it shows in the logs:
```
debug 2024-04-04 09:41:17: Received MQTT message on 'zigbee2mqtt/WoonkamerRadiatorThermostaatTrap/get' with data '{"child_lock":""}'
debug 2024-04-04 09:41:17: Publishing get 'get' 'child_lock' to 'WoonkamerRadiatorThermostaatTrap'
error 2024-04-04 09:41:17: Publish 'get' 'child_lock' to 'WoonkamerRadiatorThermostaatTrap' failed: 'Error: Cluster with name '64529' does not exist'
debug 2024-04-04 09:41:17: Error: Cluster with name '64529' does not exist
    at Object.getCluster (/app/node_modules/zigbee-herdsman/src/zcl/utils.ts:61:19)
    at Function.create (/app/node_modules/zigbee-herdsman/src/zcl/zclFrame.ts:43:31)
    at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:749:36)
    at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:404:35)
    at Object.toZigbee.convertGet (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:1435:26)
    at Publish.onMQTTMessage (/app/lib/extension/publish.ts:286:37)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:23)
    at EventEmitter.emit (node:events:529:35)
    at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:115:22)
    at MQTT.onMessage (/app/lib/mqtt.ts:141:27)
info  2024-04-04 09:41:17: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'get' 'child_lock' to 'WoonkamerRadiatorThermostaatTrap' failed: 'Error: Cluster with name '64529' does not exist'","meta":{"friendly_name":"WoonkamerRadiatorThermostaatTrap"},"type":"zigbee_publish_error"}'
```

Looking through the release notes I found for zigbee-herdsman
> [#963](https://github.com/Koenkk/zigbee-herdsman/issues/963) Update manufaturer codes (@Koenkk)

Which in turn lead me to pull 955 for zigbee-herdsman
[Add eWeLink Manufacturer Specific Cluster #955](https://github.com/Koenkk/zigbee-herdsman/pull/955)

The database contains the following values for a TRVZB:
```
{"id":10,"type":"EndDevice","ieeeAddr":"0x0ceff6fffedc6cae","nwkAddr":21096,"manufId":4742,"manufName":"SONOFF","powerSource":"Battery","modelId":"TRVZB","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":769,"inClusterList":[0,1,3,6,32,513,64599,64529],"outClusterList":[10,25],"clusters":{"64529":{"attributes":{"0":0,"16":513,"24576":0,"24577":1500,"24578":700,"24579":203,"24580":299,"24581":1351,"24582":2527,"24583":1369,"24584":127,"24585":1650,"24586":1950}},"hvacThermostat":{"attributes":{"systemMode":4,"localTemp":1930,"occupiedHeatingSetpoint":1950,"runningState":0,"localTemperatureCalibration":-15}},"genOnOff":{"attributes":{"onOff":1}},"genBasic":{"attributes":{"modelId":"TRVZB","manufacturerName":"SONOFF","powerSource":3,"zclVersion":8,"appVersion":0,"dateCode":"20230811","swBuildId":"1.1.1"}},"genPollCtrl":{"attributes":{"checkinInterval":14400}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x00124b0029dae4b7","endpointID":1},{"cluster":513,"type":"endpoint","deviceIeeeAddress":"0x00124b0029dae4b7","endpointID":1}],"configuredReportings":[{"cluster":513,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":10,"manufacturerCode":null},{"cluster":513,"attrId":18,"minRepIntval":0,"maxRepIntval":3600,"repChange":10,"manufacturerCode":null},{"cluster":513,"attrId":28,"minRepIntval":10,"maxRepIntval":3600,"repChange":null,"manufacturerCode":null}],"meta":{}}},"appVersion":0,"dateCode":"20230811","swBuildId":"1.1.1","zclVersion":8,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1712220071535,"checkinInterval":3600}
```
including the cluster name '64529'

How can I fix this?

### What did you expect to happen?

No error

### Zigbee2MQTT version

1.36.1

### Adapter firmware version

20230507

### Adapter

sonoff-zigbee-3-0-usb-dongle-plus-p

### Setup

Plain on docker container

### Debug log

```
debug 2024-04-04 09:41:17: Received MQTT message on 'zigbee2mqtt/WoonkamerRadiatorThermostaatTrap/get' with data '{"child_lock":""}'
debug 2024-04-04 09:41:17: Publishing get 'get' 'child_lock' to 'WoonkamerRadiatorThermostaatTrap'
error 2024-04-04 09:41:17: Publish 'get' 'child_lock' to 'WoonkamerRadiatorThermostaatTrap' failed: 'Error: Cluster with name '64529' does not exist'
debug 2024-04-04 09:41:17: Error: Cluster with name '64529' does not exist
    at Object.getCluster (/app/node_modules/zigbee-herdsman/src/zcl/utils.ts:61:19)
    at Function.create (/app/node_modules/zigbee-herdsman/src/zcl/zclFrame.ts:43:31)
    at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:749:36)
    at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:404:35)
    at Object.toZigbee.convertGet (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:1435:26)
    at Publish.onMQTTMessage (/app/lib/extension/publish.ts:286:37)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:23)
    at EventEmitter.emit (node:events:529:35)
    at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:115:22)
    at MQTT.onMessage (/app/lib/mqtt.ts:141:27)
info  2024-04-04 09:41:17: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'get' 'child_lock' to 'WoonkamerRadiatorThermostaatTrap' failed: 'Error: Cluster with name '64529' does not exist'","meta":{"friendly_name":"WoonkamerRadiatorThermostaatTrap"},"type":"zigbee_publish_error"}'
```

r0bb10, 2024-04-06:
Same here on Z2M 1.36.1-1 and TVRZB on 1.1.1, Sonoff-P as dongle.

Gus-Fring, 2024-04-07:
Same here, Sonoff-E dongle and HAOS 12.1 and HA Core 2024.4.1

tomadan1968, 2024-04-08:
Same here on Z2M 1.36.1-1 and TVRZB on 1.1.5, Sonoff-P as dongle.

ketzerkadaver, 2024-04-09:
Same here: Z2M 1.36.1-1, Sonoff TVRZB with 1.1.5 and Sonoff-P 20230922 as dongle.

revil-O, 2024-04-09:
Same here too, on Z2M 1.36.1-1 and TVRZB on 1.1.1 & 1.1.5, Sonoff-Zigbee-3.0-USB as dongle.

fholtgrefe, 2024-04-12:
Same here:
- Z2M 1.36.1-1
- Sonoff TRVZB 1.1.1 and 1.1.5
- Error: Cluster with name '64529' does not exist.
- HA 2024.4.2 on RPi 5

It has something to do with "open window".
After setting the option, I get:
`Error 2024-04-12 14:14:02 Publish 'set' 'open_window' to 'Heizung Bad' failed: 'Error: Cluster with name '64529' does not exist'`

Conne-X, 2024-04-12:
the same error occurs with "Child lock"

CrowMaster1, 2024-04-13:
Similar problem with cluster 

[image]

CannyGent, 2024-04-16:
+1 Same problem here - getting random fails on all of my Sonoff TRVs with this error message

Koenkk, 2024-04-16:
I'm working to fix this in https://github.com/Koenkk/zigbee-herdsman/pull/1019

bggsolar, 2024-04-21:
Same error within my installation. Seems to freeze the whole HA system as soon as some settings shall be done to the Sonoff TRV devices. Error is replicable.

Koenkk, 2024-04-23:
Could you check if this is fixed now?

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

bggsolar, 2024-04-23:
> Could you check if this is fixed now?
> 
> 
> 
> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

Hi
I just switched to edge version. Installation doesn't seem to freeze right now when transmitting new temperature settings to the TRV. 

I just had a look into the Z2M logs and found the following. I'm unsure for the moment if that is of help for you. 

[image]

I'll just follow your further proceedings, thanks for keeping up with that issue and your quick responses on that.

bggsolar, 2024-04-23:
> > Could you check if this is fixed now?
> > 
> > 
> > 
> > Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.
> 
> Hi
> I just switched to edge version. Installation doesn't seem to freeze right now when transmitting new temperature settings to the TRV. 
> 
> I just had a look into the Z2M and cannot find entries regarding the TRV' yet, what is sounding good to me. 
> 
> I'll just keep an eye on it and follow your further proceedings, thanks for keeping up with that issue and your quick responses on that. 

Hmm. I'll check it tomorrow again. So far I had to freezes again since my last post. 
Temperature settings will not be transmitted to TRV while changing in HA will lead to another freeze. 😕

bggsolar, 2024-04-25:
Reinstall and Setting up the TrV's with Better Thermostat in combination with Z2M Edge seems to work finally. Will still keep an eye on it.

bggsolar, 2024-04-28:
Communication with the TRV's seems to work reliable since my last post and your update in the edge branch last Tuesday. 

It was necessary to disconnect and reconnect also TRV's within better thermostat as well as reset the device entities in the BT Extension for controlling via schedules. 

Are there any plans for when the main branch will be updated?

Koenkk, 2024-04-28:
> Are there any plans for when the main branch will be updated?

Next release = 1 May. Assuming this can be closed now.

closed 2024-04-28 by Koenkk