Koenkk/zigbee2mqtt #22078: > 1.36.1 issue with lights turning on

ArnoutD, 2024-04-04:
### What happened?

Publish 'set' 'state' to 'SlaapkamerStrip' failed: 'RangeError [ERR_OUT_OF_RANGE]: ZCL command 0x84fd27fffe63603a/1 genLevelCtrl.moveToLevelWithOnOff({"level":417,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (The value of "value" is out of range. It must be >= 0 and <= 255. Received 417)'

### What did you expect to happen?

Light to turn on but fails due to backend error on TS0502B / _TZ3210_frm6149r Miboxer 
FUT035Z+

Error seems to be in 1.36.1... does not occur in 1.36.0 and below

perhaps related to [#7251](https://github.com/Koenkk/zigbee-herdsman-converters/pull/7251/commits/61e3b98c4b62497c21920616474f870d363f1c80)

### How to reproduce it (minimal and precise)

Turn on light ... fails with error due to odd (unspecified) payload level
ZCL command 0x84fd27fffe63603a/1 genLevelCtrl.moveToLevelWithOnOff({"level":744,"transtime":5}

HA light toggle / Z2MQTT On gives error in 1.36.1 

### Zigbee2MQTT version

1.36.1

### Adapter firmware version

20221226

### Adapter

zStack3x0 on  Sonoff Dongle Plus-E

### Setup

Add-on HASSOS

Seger85, 2024-04-04:
+1

`Publish 'set' 'state' to 'Schlafzimmer_Lampe_1' failed: 'Error: ZCL command 0xa4c1381596a3bfd8/1 genLevelCtrl.moveToLevelWithOnOff({"level":26,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":8,"sourceEndpoint":1,"destinationEndpoint":1,"options":4160,"groupId":0,"sequence":220})'`

`Geräte-Typ
Router
Zigbee-Modell
EGLO_ZM_TW
Zigbee-Hersteller
AwoX
Beschreibung
LED light with color temperature
Unterstützungsstatus
Unterstützt

IEEE-Addresse
0xa4c1381596a3bfd8
Netzwerk-Adresse
0xB810
Firmware-Version
2.7.3_1387
Hersteller
[AwoX](https://www.zigbee2mqtt.io/supported-devices/#v=AwoX)
Modell
`

Adapter
ember

land88, 2024-04-05:
+1 after zigbee2mqtt_1.36.0-1 with Home assistant

Publish 'get' 'state' to 'Lidl_prise_e092ede' failed: 'Error: ZCL command 0x60a423fffe092ede/1 genOnOff.read(["onOff"],
 {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,
"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,
"writeUndiv":false}) failed (Data request failed with error: 'MAC channel access failure' (225))'

ArnoutD, 2024-04-07:
looks similar / related to  #22033

Koenkk, 2024-04-07:
could you provide the debug log of this?

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable debug logging.

ArnoutD, 2024-04-08:
Will generate this evening

land88, 2024-04-08:
I have installed Home assistant 2024.4.1 and Now I don't know why ?, the problem has disappeared

ArnoutD, 2024-04-08:
Received MQTT message on 'zigbee2mqtt/SlaapkamerStrip/set' with data '{"state":"ON"}'
Debug 2024-04-08 21:59:21Publishing 'set' 'state' to 'SlaapkamerStrip'
Error 2024-04-08 21:59:21Publish 'set' 'state' to 'SlaapkamerStrip' failed: 'RangeError [ERR_OUT_OF_RANGE]: ZCL command 0x84fd27fffe63603a/1 genLevelCtrl.moveToLevelWithOnOff({"level":508,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (The value of "value" is out of range. It must be >= 0 and <= 255. Received 508)'
Debug 2024-04-08 21:59:21RangeError: ZCL command 0x84fd27fffe63603a/1 genLevelCtrl.moveToLevelWithOnOff({"level":508,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (The value of "value" is out of range. It must be >= 0 and <= 255. Received 508) at new NodeError (node:internal/errors:405:5) at writeU_Int8 (node:internal/buffer:740:11) at Buffer.writeUInt8 (node:internal/buffer:750:10) at BuffaloZcl.writeUInt8 (/app/node_modules/zigbee-herdsman/src/buffalo/buffalo.ts:41:21) at BuffaloZcl.write (/app/node_modules/zigbee-herdsman/src/buffalo/buffalo.ts:249:18) at BuffaloZcl.write (/app/node_modules/zigbee-herdsman/src/zcl/buffaloZcl.ts:645:26) at ZclFrame.writePayloadCluster (/app/node_modules/zigbee-herdsman/src/zcl/zclFrame.ts:153:21) at ZclFrame.toBuffer (/app/node_modules/zigbee-herdsman/src/zcl/zclFrame.ts:72:18) at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:389:22) at /app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:357:25
Info 2024-04-08 21:59:21MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'SlaapkamerStrip' failed: 'RangeError [ERR_OUT_OF_RANGE]: ZCL command 0x84fd27fffe63603a/1 genLevelCtrl.moveToLevelWithOnOff({\"level\":508,\"transtime\":0}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (The value of \"value\" is out of range. It must be >= 0 and <= 255. Received 508)'","meta":{"friendly_name":"SlaapkamerStrip"},"type":"zigbee_publish_error"}'

land88, 2024-04-08:
No excuse me, Now the problem is present, not this afternoon...I don't understand !

2024-04-08 22:09:35Received MQTT message on 'homeassistant/sensor/0x142d41fffe189b48/update_state/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x142d41fffe189b48"],"manufacturer":"IKEA","model":"TRADFRI motion sensor (E1525/E1745)","name":"Detecteur_Ikea_Salon","sw_version":"2.0.022","via_device":"zigbee2mqtt_bridge_0x00124b0024c8391a"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:update","name":"Update state","object_id":"detecteur_ikea_salon_update_state","origin":{"name":"Zigbee2MQTT","sw":"1.36.1","url":"https://www.zigbee2mqtt.io"},"state_topic":"zigbee2mqtt/Detecteur_Ikea_Salon","unique_id":"0x142d41fffe189b48_update_state_zigbee2mqtt","value_template":"{{ value_json['update']['state'] }}"}'
Debug 2024-04-08 22:09:35Received MQTT message on 'homeassistant/binary_sensor/0x142d41fffe189b48/update_available/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x142d41fffe189b48"],"manufacturer":"IKEA","model":"TRADFRI motion sensor (E1525/E1745)","name":"Detecteur_Ikea_Salon","sw_version":"2.0.022","via_device":"zigbee2mqtt_bridge_0x00124b0024c8391a"},"device_class":"update","enabled_by_default":false,"entity_category":"diagnostic","name":null,"object_id":"detecteur_ikea_salon_update_available","origin":{"name":"Zigbee2MQTT","sw":"1.36.1","url":"https://www.zigbee2mqtt.io"},"payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/Detecteur_Ikea_Salon","unique_id":"0x142d41fffe189b48_update_available_zigbee2mqtt","value_template":"{{ value_json['update']['state'] == \"available\" }}"}'
Debug 2024-04-08 22:09:35Received MQTT message on 'homeassistant/update/0x142d41fffe189b48/update/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/bridge/request/device/ota_update/update","device":{"identifiers":["zigbee2mqtt_0x142d41fffe189b48"],"manufacturer":"IKEA","model":"TRADFRI motion sensor (E1525/E1745)","name":"Detecteur_Ikea_Salon","sw_version":"2.0.022","via_device":"zigbee2mqtt_bridge_0x00124b0024c8391a"},"device_class":"firmware","entity_category":"config","entity_picture":"https://github.com/Koenkk/zigbee2mqtt/raw/master/images/logo.png","json_attributes_template":"{\"in_progress\": {{ iif(value_json['update']['state'] == 'updating', 'true', 'false') }} }","json_attributes_topic":"zigbee2mqtt/Detecteur_Ikea_Salon","latest_version_template":"{{ value_json['update']['latest_version'] }}","latest_version_topic":"zigbee2mqtt/Detecteur_Ikea_Salon","name":null,"object_id":"detecteur_ikea_salon","origin":{"name":"Zigbee2MQTT","sw":"1.36.1","url":"https://www.zigbee2mqtt.io"},"payload_install":"{\"id\": \"0x142d41fffe189b48\"}","state_topic":"zigbee2mqtt/Detecteur_Ikea_Salon","unique_id":"0x142d41fffe189b48_update_zigbee2mqtt","value_template":"{{ value_json['update']['installed_version'] }}"}'
Debug 2024-04-08 22:09:35Received MQTT message on 'homeassistant/switch/0xa4c13866aa296579/switch/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/Prise_lidl_cellier_0xa4c13866/set","device":{"identifiers":["zigbee2mqtt_0xa4c13866aa296579"],"manufacturer":"Lidl","model":"Silvercrest smart plug with power monitoring (EU, FR) (HG08673)","name":"Prise_lidl_cellier_0xa4c13866","via_device":"zigbee2mqtt_bridge_0x00124b0024c8391a"},"name":null,"object_id":"prise_lidl_cellier_0xa4c13866","origin":{"name":"Zigbee2MQTT","sw":"1.36.1","url":"https://www.zigbee2mqtt.io"},"payload_off":"OFF","payload_on":"ON","state_topic":"zigbee2mqtt/Prise_lidl_cellier_0xa4c13866","unique_id":"0xa4c13866aa296579_switch_zigbee2mqtt","value_template":"{{ value_json.state }}"}'
Debug 2024-04-08 22:09:35Received MQTT message on 'homeassistant/sensor/0xa4c13866aa296579/power_outage_memory/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0xa4c13866aa296579"],"manufacturer":"Lidl","model":"Silvercrest smart plug with power monitoring (EU, FR) (HG08673)","name":"Prise_lidl_cellier_0xa4c13866","via_device":"zigbee2mqtt_bridge_0x00124b0024c8391a"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:power-settings","name":"Power outage memory","object_id":"prise_lidl_cellier_0xa4c13866_power_outage_memory","origin":{"name":"Zigbee2MQTT","sw":"1.36.1","url":"https://www.zigbee2mqtt.io"},"state_topic":"zigbee2mqtt/Prise_lidl_cellier_0xa4c13866","unique_id":"0xa4c13866aa296579_power_outage_memory_zigbee2mqtt","value_template":"{{ value_json.power_outage_memory }}"}'
Debug 2024-04-08 22:09:35Received MQTT message on 'homeassistant/select/0xa4c13866aa296579/power_outage_memory/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/Prise_lidl_cellier_0xa4c13866/set/power_outage_memory","device":{"identifiers":["zigbee2mqtt_0xa4c13866aa296579"],"manufacturer":"Lidl","model":"Silvercrest smart plug with power monitoring (EU, FR) (HG08673)","name":"Prise_lidl_cellier_0xa4c13866","via_device":"zigbee2mqtt_bridge_0x00124b0024c8391a"},"enabled_by_ [... cut ...] .send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20) at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:760:28) at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:404:24) at Object.convertGet (/app/node_modules/zigbee-herdsman-converters/src/converters/toZigbee.ts:56:13) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:286:21) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17)

ArnoutD, 2024-04-09:
When resetting the brightness the off -> on call works but ... after a "have you tried turning it off and on again" the on uses an "uninitialized" brightness value which is out of range

Koenkk, 2024-04-09:
@ArnoutD 
- when stopping z2m, what is the the `brightness` for this device in the `data/state.json` file?
- do you have a `transition` specified for this device in the z2m `configuration.yaml`?

DOC-HOC, 2024-04-10:
Hi @Koenkk , having the same issue. 
This is the state.json entries for this file:
```
 "0xa4c1382be331de22": {
        "state": "ON",
        "color_mode": "color_temp",
        "brightness": 976,
        "color_temp": 500,
        "last_seen": "2024-04-10T08:43:28.858Z",
        "linkquality": 214``
```
no transition time as well in Configuration.yaml:

```
 '0xa4c1382be331de22':
    friendly_name: OG Wohnzimmer Decke
    color_sync: false
```

ArnoutD, 2024-04-10:
bad thing is I could replicate it...... starting and stopping a few times .... brightness was 1 in the state.json.... now I am unable to get the error anymore from zigbee2mqtt but only from homeassistant.... it uses a service light toggle on. I have no config in the configuration.yaml

DOC-HOC, 2024-04-10:
I can reproduce it in Z2M status tab. maximum value jumps around between 1000 and 254. I can more or less control the device in Z2M but in HA the slider acts like crazy...

[image]
[image]

Koenkk, 2024-04-11:
>  maximum value jumps around between 1000 and 254

Could you provide the debug log of that?

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable debug logging.

DOC-HOC, 2024-04-12:
Hi Koen, 

[log (1).txt](https://github.com/Koenkk/zigbee2mqtt/files/14958568/log.1.txt)
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/14958569/log.txt)
I tried to keep the log short but I have a lot of devices.  At 13:56:57 (log.txt) I started changing the brightness of OG Wohnzimmer Decke to a few values between 0 and 100%. It takes the commands therefore the dimmer should be OK, but the brightness slider goes to 100% after the click even though it should stay at 40% or so. 
By the way, this phenomena is with all my devices that are "MiBoxer FUT035Z+". I have another dimmer that is "MiBoxer FUT036Z" where currently everything appears to be normal. I will also attach a log of that one, doing the same thing. (log(1).txt, 14:11:40

Hope this helps!

Best,Chris

DOC-HOC, 2024-04-12:
Guess this is the issue?
MQTT publish: topic 'zigbee2mqtt/OG Wohnzimmer Decke', payload '{"**brightness":323**,"color_mode":"color_temp","color_temp":500,"do_not_disturb":null,"last_seen":"2024-04-12T11:56:58.359Z","linkquality":211,"state":"ON"}'

Koenkk, 2024-04-13:
Found & fixed the issue!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-04-13 by Koenkk

NuttShell, 2024-04-13:
TS0502B light controller report a brightness of 2550.

DOC-HOC, 2024-04-13:
> Found & fixed the issue!
> 
> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

Nice, thank you! :-)

Clem-, 2024-04-16:
Hi, I am still experiencing OUT OF RANGE issue on `latest_dev` (1.36.1-dev commit: [87aa18d](https://github.com/Koenkk/zigbee2mqtt/commit/87aa18d))
```
zigbee2mqtt  | [2024-04-16 13:02:10] debug:     z2m: Received MQTT message on 'zigbee2mqtt/0xa4c138d42e947426/set' with data '{"state":"ON"}'
zigbee2mqtt  | [2024-04-16 13:02:10] debug:     z2m: Publishing 'set' 'state' to '0xa4c138d42e947426'
zigbee2mqtt  | [2024-04-16 13:02:10] debug:     zh:controller:endpoint: ZCL command 0xa4c138d42e947426/1 genLevelCtrl.moveToLevelWithOnOff({"level":2550,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
zigbee2mqtt  | [2024-04-16 13:02:10] debug:     zh:zstack: sendZclFrameToEndpointInternal 0xa4c138d42e947426:1461/1 (0,0,1)
zigbee2mqtt  | [2024-04-16 13:02:10] debug:     zh:controller:endpoint: ZCL command 0xa4c138d42e947426/1 genLevelCtrl.moveToLevelWithOnOff({"level":2550,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (The value of "value" is out of range. It must be >= 0 and <= 255. Received 2550)
zigbee2mqtt  | [2024-04-16 13:02:10] error:     z2m: Publish 'set' 'state' to '0xa4c138d42e947426' failed: 'RangeError [ERR_OUT_OF_RANGE]: ZCL command 0xa4c138d42e947426/1 genLevelCtrl.moveToLevelWithOnOff({"level":2550,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (The value of "value" is out of range. It must be >= 0 and <= 255. Received 2550)'
zigbee2mqtt  | [2024-04-16 13:02:10] debug:     z2m: RangeError: ZCL command 0xa4c138d42e947426/1 genLevelCtrl.moveToLevelWithOnOff({"level":2550,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (The value of "value" is out of range. It must be >= 0 and <= 255. Received 2550)
zigbee2mqtt  | [2024-04-16 13:02:10] debug:     z2m: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0xa4c138d42e947426' failed: 'RangeError [ERR_OUT_OF_RANGE]: ZCL command 0xa4c138d42e947426/1 genLevelCtrl.moveToLevelWithOnOff({\"level\":2550,\"transtime\":0}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (The value of \"value\" is out of range. It must be >= 0 and <= 255. Received 2550)'","meta":{"friendly_name":"0xa4c138d42e947426"},"type":"zigbee_publish_error"}'
Publish 'set' 'state' to '0xa4c138d42e947426' failed: 'RangeError [ERR_OUT_OF_RANGE]: ZCL command 0xa4c138d42e947426/1 genLevelCtrl.moveToLevelWithOnOff({"level":2550,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (The value of "value" is out of range. It must be >= 0 and <= 255. Received 2550)'
```

ArnoutD, 2024-04-16:
Off topic @Clem- have we met before in a working fashion (Mx @ abn )?

Koenkk, 2024-04-16:
Could you provide the debug log when the `brightness` changed to `2550` for the first time?

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable debug logging.

Clem-, 2024-04-17:
Turning light on with brightness=200 (from HomeAssistant)
```
zigbee2mqtt  | [2024-04-17 15:49:30] debug:     z2m: Received MQTT message on 'zigbee2mqtt/0xa4c138d42e947426/set' with data '{"state":"ON","brightness":199}'
zigbee2mqtt  | [2024-04-17 15:49:30] debug:     z2m: Publishing 'set' 'brightness' to '0xa4c138d42e947426'
zigbee2mqtt  | [2024-04-17 15:49:30] debug:     zh:controller:endpoint: ZCL command 0xa4c138d42e947426/1 genLevelCtrl.moveToLevelWithOnOff({"level":199,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
zigbee2mqtt  | [2024-04-17 15:49:30] debug:     zh:zstack: sendZclFrameToEndpointInternal 0xa4c138d42e947426:1461/1 (0,0,1)
zigbee2mqtt  | [2024-04-17 15:49:31] debug:     z2m: MQTT publish: topic 'zigbee2mqtt/0xa4c138d42e947426', payload '{"brightness":199,"color":{"h":0,"hue":0},"color_mode":"color_temp","color_temp":303,"do_not_disturb":true,"linkquality":47,"state":"ON"}'
zigbee2mqtt  | [2024-04-17 15:49:31] debug:     z2m: Received Zigbee message from '0xa4c138d42e947426', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
zigbee2mqtt  | [2024-04-17 15:49:31] debug:     z2m: Received Zigbee message from '0xa4c138d42e947426', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
zigbee2mqtt  | [2024-04-17 15:49:31] debug:     z2m: MQTT publish: topic 'zigbee2mqtt/0xa4c138d42e947426', payload '{"brightness":199,"color":{"h":0,"hue":0},"color_mode":"color_temp","color_temp":303,"do_not_disturb":true,"linkquality":69,"state":"ON"}'
zigbee2mqtt  | [2024-04-17 15:49:31] debug:     z2m: MQTT publish: topic 'zigbee2mqtt/0xa4c138d42e947426', payload '{"brightness":199,"color":{"h":0,"hue":0},"color_mode":"color_temp","color_temp":303,"do_not_disturb":true,"linkquality":69,"state":"ON"}'
zigbee2mqtt  | [2024-04-17 15:49:31] debug:     z2m: Received Zigbee message from '0xa4c138d42e947426', type 'attributeReport', cluster 'genOnOff', data '{"61440":0}' from endpoint 1 with groupID 0
zigbee2mqtt  | [2024-04-17 15:49:31] debug:     z2m: Received Zigbee message from '0xa4c138d42e947426', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
zigbee2mqtt  | [2024-04-17 15:49:31] debug:     z2m: MQTT publish: topic 'zigbee2mqtt/0xa4c138d42e947426', payload '{"brightness":199,"color":{"h":0,"hue":0},"color_mode":"color_temp","color_temp":303,"do_not_disturb":true,"linkquality":69,"state":"OFF"}'
zigbee2mqtt  | [2024-04-17 15:49:31] debug:     z2m: Received Zigbee message from '0xa4c138d42e947426', type 'attributeReport', cluster 'lightingColorCtrl', data '{"tuyaRgbMode":0}' from endpoint 1 with groupID 0
zigbee2mqtt  | [2024-04-17 15:49:31] debug:     z2m: MQTT publish: topic 'zigbee2mqtt/0xa4c138d42e947426', payload '{"brightness":199,"color":{"h":0,"hue":0},"color_mode":"color_temp","color_temp":303,"do_not_disturb":true,"linkquality":72,"state":"OFF"}'
zigbee2mqtt  | [2024-04-17 15:49:31] debug:     z2m: Received Zigbee message from '0xa4c138d42e947426', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":199}' from endpoint 1 with groupID 0
zigbee2mqtt  | [2024-04-17 15:49:31] debug:     z2m: MQTT publish: topic 'zigbee2mqtt/0xa4c138d42e947426', payload '{"brightness":199,"color":{"h":0,"hue":0},"color_mode":"color_temp","color_temp":303,"do_not_disturb":true,"linkquality":69,"state":"OFF"}'
zigbee2mqtt  | [2024-04-17 15:49:31] debug:     z2m: Received Zigbee message from '0xa4c138d42e947426', type 'attributeReport', cluster 'genLevelCtrl', data '{"61440":783}' from endpoint 1 with groupID 0
zigbee2mqtt  | [2024-04-17 15:49:31] debug:     z2m: MQTT publish: topic 'zigbee2mqtt/0xa4c138d42e947426', payload '{"brightness":1997,"color":{"h":0,"hue":0},"color_mode":"color_temp","color_temp":303,"do_not_disturb":true,"linkquality":69,"state":"OFF"}'
zigbee2mqtt  | [2024-04-17 15:49:31] debug:     z2m: Received Zigbee message from '0xa4c138d42e947426', type 'attributeReport', cluster 'lightingColorCtrl', data '{"57600":[783,0]}' from endpoint 1 with groupID 0
zigbee2mqtt  | [2024-04-17 15:49:31] debug:     z2m: MQTT publish: topic 'zigbee2mqtt/0xa4c138d42e947426', payload '{"brightness":1997,"color":{"h":0,"hue":0},"color_mode":"color_temp","color_temp":303,"do_not_disturb":true,"linkquality":51,"state":"OFF"}'
zigbee2mqtt  | [2024-04-17 15:49:31] debug:     zh:controller:endpoint: ZCL command 0xa4c138d42e947426/1 genOnOff.defaultRsp({"cmdId":10,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":84,"writeUndiv":false})
zigbee2mqtt  | [2024-04-17 15:49:31] debug:     zh:zstack: sendZclFrameToEndpointInternal 0xa4c138d42e947426:1461/1 (0,0,1)
zigbee2mqtt  | [2024-04-17 15:49:31] debug:     z2m: Received Zigbee message from '0xa4c138d42e947426', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
zigbee2mqtt  | [2024-04-17 15:49:31] debug:     z2m: MQTT publish: topic 'zigbee2mqtt/0xa4c138d42e947426', payload '{"brightness":1997,"color":{"h":0,"hue":0},"color_mode":"color_temp","color_temp":303,"do_not_disturb":true,"linkquality":69,"state":"OFF"}'
[... cut ...] bleRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":85,"writeUndiv":false})
zigbee2mqtt  | [2024-04-17 15:49:31] debug:     zh:zstack: sendZclFrameToEndpointInternal 0xa4c138d42e947426:1461/1 (0,0,1)
zigbee2mqtt  | [2024-04-17 15:49:31] debug:     z2m: Received Zigbee message from '0xa4c138d42e947426', type 'attributeReport', cluster 'genOnOff', data '{"61440":0}' from endpoint 1 with groupID 0
```

jdbrookes, 2024-04-17:
Finally got around to investigating this very issue today with LED countertop lights I've had installed for years. Thought I had a config or YAML issue in Home Assistant that was causing the lights to immediately toggle off immediately after toggling on, but it only happens approx every 5 times I turn on the lights.

Zigbee Model: TS0502B
Zigbee Manufacturer: _TZ3210_rm0hthdo

`Publish 'set' 'state' to 'Kitchen Counter Tops' failed: 'RangeError [ERR_OUT_OF_RANGE]: ZCL command 0xa4c1384964116d93/1 genLevelCtrl.moveToLevelWithOnOff({"level":772,"transtime":20}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (The value of "value" is out of range. It must be >= 0 and <= 255. Received 772)'`

Clem-, 2024-04-17:
That is very interesting because it is also intermittent in my case. Not always turning off by itself. 

It seems that we have the exact same device. Please check this [issue](https://github.com/Koenkk/zigbee2mqtt/issues/22197) to add your feedbacks.

Koenkk, 2024-04-18:
Made a typo, fixed now.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.