Koenkk/zigbee2mqtt #22079: [New device support]: TS0601 _TZE204_edl8pz1k

pq16, 2024-04-04:
### Link

no information

### Database entry

{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0024cbf976","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,8,10,11,12,13,47,110,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"8":{"profId":260,"epId":8,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"10":{"profId":260,"epId":10,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[1281,10],"outClusterList":[1280,1282],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"12":{"profId":49246,"epId":12,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"13":{"profId":260,"epId":13,"devId":5,"inClusterList":[25],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"47":{"profId":260,"epId":47,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"110":{"profId":260,"epId":110,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":null}

### Comments

If possible, please share the converter. Thank you

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_edl8pz1k',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

Kastorsky, 2024-04-19:
Hi!
You can try this external converter:
```js
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE200_edl8pz1k']),
    model: 'TS0601_floor_thermostat',
    vendor: 'TuYa',
    description: 'Zigbee thermostat for electric floors',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        e.climate()
            .withSystemMode(['off', 'heat'], ea.STATE_SET)
            .withPreset(['manual', 'auto'])
            .withRunningState(['idle', 'heat'], ea.STATE)
            .withSetpoint('current_heating_setpoint', 5, 35, 0.5, ea.STATE_SET)
            .withLocalTemperature(ea.STATE).withDescription('Floor temperature')
            .withLocalTemperatureCalibration(-9, 9, 0.1, ea.STATE_SET).withDescription('Calibration floor temperature sensor'),
        e.deadzone_temperature().withValueMin(0).withValueMax(5).withValueStep(1).withDescription('Floor temperature'),
        e.child_lock(),
        ...tuya.exposes.scheduleAllDays(ea.STATE_SET, 'HH:MM/C HH:MM/C HH:MM/C HH:MM/C'),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'system_mode', tuya.valueConverterBasic.lookup({'heat': true, 'off': false})],
            [2, 'preset', tuya.valueConverter.tv02Preset()],
            [16, 'current_heating_setpoint', tuya.valueConverter.divideBy10],
            [24, 'device_temperature', tuya.valueConverter.divideBy10],
            [27, 'local_temperature_calibration', tuya.valueConverter.localTempCalibration2],
            [36, 'running_state', tuya.valueConverterBasic.lookup({'heat': tuya.enum(0), 'idle': tuya.enum(1)})],
            [40, 'child_lock', tuya.valueConverter.lockUnlock],
            [102, 'local_temperature', tuya.valueConverter.divideBy10],
            [103, 'deadzone_temperature', tuya.valueConverter.raw],
            [110, 'schedule_monday', tuya.valueConverter.thermostatScheduleDayMultiDP],
            [109, 'schedule_tuesday', tuya.valueConverter.thermostatScheduleDayMultiDP],
            [108, 'schedule_wednesday', tuya.valueConverter.thermostatScheduleDayMultiDP],
            [107, 'schedule_thursday', tuya.valueConverter.thermostatScheduleDayMultiDP],
            [106, 'schedule_friday', tuya.valueConverter.thermostatScheduleDayMultiDP],
            [105, 'schedule_saturday', tuya.valueConverter.thermostatScheduleDayMultiDP],
            [101, 'schedule_sunday', tuya.valueConverter.thermostatScheduleDayMultiDP],
        ],
    },
    whiteLabel: [{vendor: 'ELECTSMART', model: 'EST-120Z'}],
};
module.exports = definition;
```

pq16, 2024-04-19:
Hi, friend.
Thanks, it's working.

Koenkk, 2024-04-21:
`_TZE204_edl8pz1k` will be supported out-of-the-box with the next release.

closed 2024-04-21 by Koenkk

aleks-mariusz, 2024-12-13:
> You can try this external converter:

Just as an FYI, this converter also works for _TZE204_o9d1hdma - it was purchased from [here](https://www.aliexpress.com/item/1005005959326070.html).. details on the model (incl manual) are [here](https://thermostat.guide/acmelec/acmelec-e-740d-underfloor-heating-16a-thermostat-user-manual/).

Maybe we can update this convert to also support this model out of the box?

Kastorsky, 2024-12-14:
Unfortunately, I don't have this thermostat, and in order to write a converter, you need to see what commands it sends during various actions. But you can try to do it yourself, based on my converter, for this you need to:
1. enable zigbee2mqtt logging in info mode; 
2. press the buttons on the thermostat itself to see what codes are displayed in the log and specify these codes in the external converter, when you identify all the codes, the device can be controlled. 
This process is described in more detail
[here](https://www.zigbee2mqtt.io/advanced/support-new-devices/02_support_new_tuya_devices.html)

mattia9508, 2025-10-15:
Buonasera a tutti, perdonate la domanda, uso home assistant da circa 2 anni, ma sto entrando solamente ora nella parte tecnica, in particolare con zigbee2mqtt, vi chiedo perdono in anticipo per il disturbo, purtroppo mi sono messo d'impegno per risolvere da solo il problema, ma purtroppo non ne vengo a capo.
ho acquistato 6 termostati tuya 
[https://italian.alibaba.com/product-detail/Gold-Adjustable-Digital-Thermostat-Fcu-Air-1600287511784.html](url)
purtroppo mio sono accorto subito che non sono supportati da zigbe2mqtt, ho provveduto a creare un contenitore esterno, rispecchiando il lavoro da voi riportato con qualche leggera modifica in quanto non mi interessa della programmazione oraria, funziona praticamente tutto tranne quello che mi serve veramente, non riesco a leggere la temperatura rilevata e tantomeno lo stato del rele, se chiuso o aperto.
Qualcuno potrebbe per favore darmi un aiuto? 
grazie 

[image]

Convertitore esterno:
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
        fingerprint: tuya.fingerprint('TS0601', ['_TZE204_o9d1hdma', '_TZE204_o9d1hdma']),
        model: "AE-669K",
        vendor: "ACMELEC",
        description: "Compatible with Mitsubishi Electric vrf system",
        extend: [tuya.modernExtend.tuyaBase({dp: true, timeStart: "2000"})],
        exposes: [
            e.binary("state", ea.STATE_SET, "ON", "OFF").withDescription("Turn the thermostat ON/OFF"),
            e.climate()
                .withSystemMode(["cool", "heat", "fan_only", "dry"], ea.STATE_SET)
                .withSetpoint("current_heating_setpoint", 16, 32, 1, ea.STATE_SET)
                .withRunningState(['idle', 'heat'], ea.STATE)
                .withLocalTemperature(ea.STATE).withDescription('Floor temperature'),
            e.child_lock(),
    ],
    meta: {
        tuyaDatapoints: [
                [1, "state", tuya.valueConverter.onOff],
                [2,"system_mode", tuya.valueConverterBasic.lookup({cool: tuya.enum(0), heat: tuya.enum(1), fan_only: tuya.enum(2),dry: tuya.enum(3),}),],
                [16, "current_heating_setpoint", tuya.valueConverter.raw],
                [24, 'device_temperature', tuya.valueConverter.divideBy10],
                [36, 'running_state', tuya.valueConverterBasic.lookup({'heat': tuya.enum(0), 'idle': tuya.enum(1)})],
                [40, "child_lock", tuya.valueConverter.lockUnlock],
        ],
    },
    whiteLabel: [{vendor: 'ELECTSMART', model: 'EST-120Z'}],
};
module.exports = definition;

Definizione esterna:

export default {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_o9d1hdma',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

LOG in modalità debug:

[15/10/2025, 19:12:49] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c138e129050f30', payload '{"child_lock":"UNLOCK","current_heating_setpoint":28,"device_temperature":2.1,"linkquality":148,"running_state":null,"state":"ON","system_mode":"cool"}'