Koenkk/zigbee2mqtt #22080: External Convertor of Water Meter exposes missing after update to1.36-1.1

tzeryin1981, 2024-04-04:
### What happened?

Exposes become empty

### Zigbee2MQTT version

1.36.1-1

### Adapter firmware version

20220219

### Adapter

zStack3x0

### Setup

Add on Home Assistant on x86 intel pc

MrEcosse, 2024-04-04:
None of my External Converters are now being picked up and used - all devices showing as unsupported.  This using 1.36.1 and HA 2024.4.0

Great-Chart, 2024-04-04:
There may be some clues from within #22076?

tzeryin1981, 2024-04-05:
> There may be some clues from within #22076?

Tried, doesn't work.

tzeryin1981, 2024-04-05:
> There may be some clues from within #22076?

Ya, t works. However need to add a line for Tuya product:

const tuya = require('zigbee-herdsman-converters/lib/tuya');

Koenkk, 2024-04-06:
Could you create a [pull request](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/tuya.ts) to add out-of-the-box support for this device?

closed 2024-11-03 by github-actions[bot]