Koenkk/zigbee2mqtt #22081: Very unstable Zigbee network

Danixu, 2024-04-04:
### What happened?

Hello,

From a long time ago I am having an unstable zigbee network which causes to some devices to randomly disconnect without any reason, other are just irresponsible to receiving commands but Zigbee2Mqtt receives their state changes, or very big delays. The problems maybe matches with the size increase of my network because when having just 5 or 6 devices it has never failed.

Now I have 30 devices:
By model
```
TS0601: 6
TS0011: 5
TS011F: 5
TS0012: 3
TS0201: 3
lumi.relay.c2acn01: 2
TS0203: 2
lumi.weather: 2
lumi.switch.n0acn2: 1
TS0202: 1
```
The devices have random problems and is not always the same devices or brands. For example, a device can be responding perfectly to commands 7 times, and the 8th just fails or have a delay of several seconds. Or even be responding correctly for a week and then just stop working without reason. To restart the Z2M or the computer doesn't solve the problem.
I have though about interferences, but on my zone the only 2.4Ghz device is my router and because I have some devices that need it. The rest are very far away (undetectable by my phone on the coordinator zone). The distance is another problem that I have though, but one of the devices which is failing right now is at about 4 meters of the coordinator. Most of the LQI indicators are above 200...

Even I have changed the coordinator with another model with a different firmware (SONOFF dongle-P to dongle-E), but after some days working perfectly, the starts to cause troubles. On the change, I have fully reinstalled the Z2M and migrated the network (pairing all the devices again). On this change I have also changed the Zigbee channel to ensure that is not the problem. Now I am using the channel 25 (my 2.4ghz network is at channel 13).

I am desesperated... I don't know what to do. I am even thinking about to fully migrate to ZHA to see if works.

Any tip to try to find the problem?

### Zigbee2MQTT version

1.35.2

### Adapter firmware version

7.3.1.0 build 176

### Adapter

SONOFF Dongle-E

### Setup

Running on a ProxMox CT. I have HA as VM and Mosquitto as CT too, but the problem is just related with Z2M.

Nerivec, 2024-04-05:
Since this affect different types of coordinators, this is likely interferences and/or bad routers.

I assume you have gone through https://www.zigbee2mqtt.io/advanced/zigbee/02_improve_network_range_and_stability.html ?

Changing your WiFi to channel 6 and your Zigbee network to 25 was a [good first step](https://www.metageek.com/training/resources/zigbee-wifi-coexistence/).

Second step, I'd suggest you go through the [device page](https://www.zigbee2mqtt.io/supported-devices/) for each of your devices (there should be a link in Z2M UI when you open a device, under `Model`). If one is known to be a bad router, there will usually be a notice about it; if so, remove that device from the network and see if it stabilizes after a couple of hours. Since you said you had no trouble before, when you had less devices, it seems likely that one (or more) of the devices added after that is the cause of (some of) your problems; start with them.

Last, you can try [updating](https://darkxst.github.io/silabs-firmware-builder/) the [Dongle-E to 7.4.1](https://github.com/darkxst/silabs-firmware-builder/raw/main/firmware_builds/zbdonglee/ncp-uart-hw-v7.4.1.0-zbdonglee-115200.gbl) and use the `ember` driver (change `adapter: ezsp` to `adapter: ember` in your `configuration.yaml`). Even though that won't solve your problems if it's interferences and/or bad routers, `ember` logs will detail network errors, and every hour will report stats from the coordinator, which may help in figuring out the issue more specifically.

lux73, 2024-04-06:
AFAIK the Dongle-E is still experimental even there are new drivers - if you want a stable & reliable Network buy an [CC2652x based Coordinator](https://www.zigbee2mqtt.io/guide/adapters/#recommended) and flash a [stable firmware](https://github.com/Koenkk/Z-Stack-firmware/files/12706322/20230922_20230923.zip) to it

The 20230922 is the most rock stable Firmware for my CC2652RB Coordinator atm - any other even all newer is NOT stable (but i have to admit that i have not tried the [latest 20240315](https://github.com/Koenkk/Z-Stack-firmware/files/14842466/20240315.zip)

anyhow did you use an extension Cable for your dongle?  its not advised to put it directly in your USB Port...

Danixu, 2024-04-06:
> AFAIK the Dongle-E is still experimental even there are new drivers - if you want a stable & reliable Network buy an [CC2652x based [...]

Yes, I know that is experimental, but I wanted to test a different model. I was using the Sonoff Dongle-P which is a CC2652P based coordinator with the firmware 20230507, and I also had problems with the network: Some devices fully stopped working, some partially...
I don't know if the difference is the channel that I have used (even when there are no other routers near my house), the output power of the coordinator,... but the network has much higher LQI after the change. Before I had LQI of 25 to 50 on some far devices, and now the same devices has a LQI of 150+ and even some above 200. Anyway, the problems were  there again and I am still testing the changes I have done.

For now the changes of the WiFi channel and the use of an extension cord has make it more stable, but only two days have passed and is soon to be fully sure. 

Best regards.

lux73, 2024-04-06:
before proceeding pls update your CC2652P Coordinator to the most stable Firmware 20230922 😉

20230507 was NOT stable for me...

my Network:
```
End devices: 36
Router: 20
```

Danixu, 2024-04-14:
For now the network seems to be stable, only a few problems with a device but because is far away and the signal seems to be low and then it randomly disconnect, but the rest of devices seems to be there without disconnecting for now.

I'll keep an eye on it at least one week more and if still working, I'll close this ticket.

Finally I have kept the Dongle-E with the latest firmware because to move the entire network to a new coordinator is a mess... today I have enabled the ember driver after a while testing just the update and for now everything is OK.

Just one question: where did you find that updated firmwares?, because the latest available in the sislab page is the 7.3 if I am not wrong (I remember that is not the one that you sent here, it is an older version). And for the Z-stack firmware is the same, the latest version available at the repository is the 20230507, but you have provided two versions that are more updated.

Best regards.

Nerivec, 2024-04-14:
For silabs, the firmware files by darkxst for Dongle-E are here: https://github.com/darkxst/silabs-firmware-builder/tree/main/firmware_builds/zbdonglee
_The flasher still provides 7.3 by default (when not providing the file directly), but 7.4 has been available for a while now._

closed 2024-04-21 by Danixu