Koenkk/zigbee2mqtt #22082: Provide acceleration threshold and threshold multiplier for the Aeotec IM6001-MPP01 Multipurpose Sensor to configure sensitivity

Schorschemann, 2024-04-04:
### Is your feature request related to a problem? Please describe

I use some Aeotec IM6001-MPP01 Multipurpose Sensors. I planned to use them to detect vibration at my windows when someone unauthorized tries to get in.  The detected vibration will trigger my shutters to close my shutters.

Unfortunately the sensitivity of the IM6001-MPP01 is much too high. It even detects someone knocking at the window or when the paper boy brings the newspaper at night and drops it through the main door.

From the smart things forum I know that the parameters for the acceleration threshold and the threshold multiplier are available.

Mariano_Colmenarejo (https://community.smartthings.com/u/Mariano_Colmenarejo) programmed a converter for smartthings.

Unfortunately I was not able to adapt it to a Zigbee2Mqtt converter. Unfortunately Mariano could not help.

Deatils can be found here: https://community.smartthings.com/t/change-sensitivity-on-aeotec-zigbee-sensor/279249

### Describe the solution you'd like

It would be great to have the acceleration threshold and the threshold multiplier parameters configurable in Zigbee2MQTT.

### Describe alternatives you've considered

I tried to convert the SmartThings .LUA files to .JS and write my own converter, but reached my limits.

### Additional context

Details of my discussion with Mariano in the SmartThings forum can be found here:

Deatils can be found here: https://community.smartthings.com/t/change-sensitivity-on-aeotec-zigbee-sensor/279249

closed 2024-11-23 by github-actions[bot]

ptr727, 2024-12-18:
Just installed one of these, too sensitive, triggers from cars driving by.

Schorschemann, 2024-12-18:
Wow, this actually is very sensitive. It sounds like even strong winds will trigger.