Koenkk/zigbee2mqtt #22085: power strip SM-0306E-2W - not working entity "power_outage_memory"

johnfromul, 2024-04-04:
### What happened?

What happened?
Receiving regular errors in the GUI. Isn´t possible set the state of entity "power_outage_memory"
[image]
[image]
[image]

Device type: Router
Zigbee Model: TS011F
Zigbee Manufacturer: _TZ3000_cfnprab5
Description: 4 gang switch, with USB
Manufacturer: [UseeLink](https://www.zigbee2mqtt.io/supported-devices/#v=UseeLink)
Model: [SM-0306E-2W](https://www.zigbee2mqtt.io/devices/SM-0306E-2W.html#useelink-sm-0306e-2w)

### What did you expect to happen?

will be set the value to : on/off/restore

### How to reproduce it (minimal and precise)

choice a device in Z2M - exposes - and try set entity "power outage memory"

### Zigbee2MQTT version

 1.36.1

### Adapter firmware version

20210708

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus, TI CC2652P + CP2102(N)

### Setup

Synology virtual machine, 8gb ram

ymerj, 2024-04-06:
I can confirm I'm experiencing the exact same problems with mine in 1.36.1.

Same model SM-0306E-2W

sabaatworld, 2024-06-20:
Still the same problem. I see the error though the features do work.

tbgoose, 2024-08-09:
Also same issue, and not sure if related but I have the AU version [here ](https://www.zigbee2mqtt.io/devices/SM-O301-AZ.html)(SM-O301-AZ)

raphael1688, 2024-10-06:
Will these info dump from TUYA IOT site help? 

**Query Device Details**

`
{
  "result": [
    {
      "active_time": 1712632727,
      "bind_space_id": "73249562",
      "category": "pc",
      "create_time": 1712632727,
      "custom_name": "排插4+1 Zigbee CDC4",
      "icon": "smart/icon/001459991929841m52si/97e33e06a65ed0a5a5843d753097da7d.png",
      "id": "eb1c70f6f09bb7f155ts6f",
      "ip": "",
      "is_online": false,
      "lat": "24.1300",
      "local_key": "",
      "lon": "120.7200",
      "model": "SM-SO301Z",
      "name": "排插4+1 Zigbee",
      "product_id": "cfnprab5",
      "product_name": "排插4+1 Zigbee",
      "sub": true,
      "time_zone": "+08:00",
      "update_time": 1712632762,
      "uuid": "eb1c70f6f09bb7f155ts6f"
    }
  ],
  "success": true,
  "t": 1728200234657,
  "tid": "ce53cf8283b511ef8c45ae4e0fbf60d4"
}
`

**Get the specifications and properties of the device**

`
{
  "result": {
    "category": "pc",
    "functions": [
      {
        "code": "switch_1",
        "desc": "{}",
        "name": "开关1",
        "type": "Boolean",
        "values": "{}"
      },
      {
        "code": "switch_2",
        "desc": "{}",
        "name": "开关2",
        "type": "Boolean",
        "values": "{}"
      },
      {
        "code": "switch_3",
        "desc": "{}",
        "name": "开关3",
        "type": "Boolean",
        "values": "{}"
      },
      {
        "code": "switch_4",
        "desc": "{}",
        "name": "开关4",
        "type": "Boolean",
        "values": "{}"
[... 369 lines cut ...]
        "time": 1712632730521,
        "type": "bool",
        "value": false
      },
      {
        "code": "cycle_timing",
        "custom_name": "",
        "dp_id": 209,
        "time": 1712632727059,
        "type": "raw"
      },
      {
        "code": "random_timing",
        "custom_name": "",
        "dp_id": 210,
        "time": 1712632727059,
        "type": "raw"
      }
    ]
  },
  "success": true,
  "t": 1728200975305,
  "tid": "87d1f82383b711ef86b3722f08288c6c"
}
`

raphael1688, 2024-10-06:
I also dump the 'working' SM-0306E-2W which is the same as picture from TUYA IOT site.
The model is SM-SO306 and FCC ID is 2AJ5F-SM-SO306.

**Query Device Details**

`
{
  "result": {
    "active_time": 1728201683,
    "bind_space_id": "73249562",
    "category": "pc",
    "create_time": 1728201683,
    "custom_name": "排插4+1 Zigbee 27E7",
    "icon": "smart/icon/001459991929841m52si/97e33e06a65ed0a5a5843d753097da7d.png",
    "id": "eb4de13d3a73ba892fudv6",
    "ip": "",
    "is_online": true,
    "lat": "24.13",
    "local_key": "",
    "lon": "120.72",
    "model": "SM-SO301Z",
    "name": "排插4+1 Zigbee 5",
    "product_id": "cfnprab5",
    "product_name": "排插4+1 Zigbee",
    "sub": true,
    "time_zone": "+08:00",
    "update_time": 1728201726,
    "uuid": "a4c1383997b027e7"
  },
  "success": true,
  "t": 1728201795058,
  "tid": "70682c2983b911efbcbd7aea62a19b8a"
}

**Query Properties**

`{
  "result": {
    "properties": [
      {
        "code": "switch_1",
        "custom_name": "",
        "dp_id": 1,
        "name": "Switch 1",
        "time": 1728201685732,
        "type": "bool",
        "value": true
      },
      {
        "code": "switch_2",
        "custom_name": "",
        "dp_id": 2,
        "name": "Switch 2",
        "time": 1728201685747,
        "type": "bool",
        "value": true
      },
      {
        "code": "switch_3",
        "custom_name": "",
        "dp_id": 3,
        "name": "Switch 3",
        "time": 1728201685763,
        "type": "bool",
        "value": true
      },
      {
        "code": "switch_4",
        "custom_name": "",
        "dp_id": 4,
[... 368 lines cut ...]
        "values": "{\"range\":[\"power_off\",\"power_on\",\"last\"]}"
      },
      {
        "code": "light_mode",
        "desc": "light mode",
        "name": "light mode",
        "type": "Enum",
        "values": "{\"range\":[\"none\",\"relay\",\"pos\"]}"
      },
      {
        "code": "child_lock",
        "desc": "child lock",
        "name": "child lock",
        "type": "Boolean",
        "values": "{}"
      }
    ]
  },
  "success": true,
  "t": 1728201917368,
  "tid": "b94fcf3583b911ef86b3722f08288c6c"
}
`
`
`

iUnstable0, 2024-11-19:
im getting the same error! I swear it used to work but then new zigbee2mqtt updates it is no longer working. I still have the feature child lock and restore state turned on but i can no longer change it without throwing error

teskanoo, 2025-01-19:
I also have (what appears to be visually) the [SM-O301-AZ](https://www.zigbee2mqtt.io/devices/SM-O301-AZ.html) but it is being recognized as a [SM-0306E-2W](https://www.zigbee2mqtt.io/devices/SM-0306E-2W.html#useelink-sm-0306e-2w)

Experiencing the same inability to se the power_outage_memory and Child Lock

Bart-VanDerGucht, 2025-01-20:
I experience the same issue with a Zauzau (UseeLink) SM-302-F-PA20, which visually resembles the [SM-O301-AZ](https://www.zigbee2mqtt.io/devices/SM-O301-AZ.html) but it is being recognized as a [SM-0306E-2W](https://www.zigbee2mqtt.io/devices/SM-0306E-2W.html#useelink-sm-0306e-2w)

Zigbee Model : TS011F
Zigbee Manufacturer : _TZ3000_cfnprab5

state:
{
    "child_lock": "LOCK",
    "last_seen": "2025-01-20T10:50:12.001Z",
    "linkquality": 176,
    "power_outage_memory_l1": "restore",
    "state_l1": "ON",
    "state_l2": "ON",
    "state_l3": "ON",
    "state_l4": "ON",
    "state_l5": "ON",
    "action": "on_l5",
    "power_outage_memory": null
}

alexvenom, 2025-03-20:
I am having a lot of trouble trying to add this device to my Zigbee2MQTT.
Basically, every time I try to sync, I get this:

```
[2025-03-20 09:50:08] debug:     zh:controller: Received ZDO response: clusterId=END_DEVICE_ANNOUNCE, status=SUCCESS, payload={"nwkAddress":26390,"eui64":"0x00158d0003557911","capabilities":{"alternatePANCoordinator":0,"deviceType":1,"powerSource":1,"rxOnWhenIdle":1,"reserved1":0,"reserved2":0,"securityCapability":0,"allocateAddress":1}}
[2025-03-20 09:50:08] debug:     zh:controller: Device announce from '0x00158d0003557911:26390'
[2025-03-20 09:50:08] debug:     zh:controller: Device announce is from unknown device '0x00158d0003557911:26390'
````

I am using this Coordinator:
```
/dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0014B88675-if00
````

And even tried to write an external_converter for it:

````
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    // Expand model identifiers to catch potential variants
    zigbeeModel: ['SM-O301-AZ', 'SM-O301', 'SM-O301AZ', 'SiHAS outlet'],
    model: 'SM-O301-AZ',
    vendor: 'ShinaSystem',
    description: 'SiHAS multipurpose power outlet',

    // Multi endpoint support
    endpoint: (device) => {
        return {
            l1: 1,
            l2: 2,
            l3: 3,
            l4: 4,
            l5: 5,
        };
    },

    // Expanded conversion functions to catch more potential messages
    fromZigbee: [
        fz.on_off,
        fz.ignore_basic_report,
        fz.command_recall,
        fz.ignore_genOta,
        fz.ignore_zclversion_read,
        fz.ignore_basic_cluster
    ],
    toZigbee: [
        tz.on_off,
        tz.power_on_behavior,
        tz.child_lock,
        tz.on_off_timed
    ],

    // Device capabilities
    exposes: [
        e.switch().withEndpoint('l1'),
        e.switch().withEndpoint('l2'),
        e.switch().withEndpoint('l3'),
        e.switch().withEndpoint('l4'),
        e.switch().withEndpoint('l5'),
        e.power_outage_memory(),
        e.child_lock(),
    ],

    // Configuration function for setup
    configure: async (device, coordinatorEndpoint, logger) => {
        logger.debug('Configuring SM-O301-AZ');
        for (const ep of [1, 2, 3, 4, 5]) {
            try {
                const endpoint = device.getEndpoint(ep);
                if (endpoint) {
                    await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
                    await reporting.onOff(endpoint);
                    logger.debug(`Successfully configured endpoint ${ep}`);
                }
            } catch (error) {
                logger.debug(`Failed to configure endpoint ${ep}: ${error}`);
            }
        }
    },
};

module.exports = definition;
````

But all I get in the logs are the messages at the beginning of this message.
Anyone has a clue?

teskanoo, 2025-03-21:
I notice that on the dev console there are both numeric and string endpoints 1-5 listed in the dropdown

[image]

Also I notice that oddly the cluster is no named but is numeric

[image]

Selecting a numeric endpoints seems fine
Selecting an alpha-numeric one results in ...

[image]

alexvenom, 2025-04-16:
Just in case someone stumbles on this in the future, I was able to make it work.
First, I had to use Zigbee2MQTT specifically in version 1.36.0. No other version worked.
Also, I had to write this custom external converter:

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposesNew');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['SM-O301-AZ'],
    model: 'SM-O301-AZ',
    vendor: 'Tuya',
    description: 'Zigbee power strip with multiple outlets',
    fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering, fz.tuya_switch_power_outage_memory, fz.lock],
    toZigbee: [tz.on_off, tz.tuya_switch_power_outage_memory, tz.lock],
    exposes: [
        e.switch().withEndpoint('l1'),
        e.switch().withEndpoint('l2'),
        e.switch().withEndpoint('l3'),
        e.switch().withEndpoint('l4'),
        e.switch().withEndpoint('l5'),
        exposes.enum('power_outage_memory', ea.ALL, ['on', 'off', 'restore']).withDescription('Recover state after power outage'),
        exposes.binary('child_lock', ea.ALL, 'LOCK', 'UNLOCK').withDescription('Enables/disables physical input on the device'),
        e.power(),
        e.energy(),
        e.voltage(),
        e.current(),
    ],
    endpoint: (device) => {
        return { 'l1': 1, 'l2': 2, 'l3': 3, 'l4': 4, 'l5': 5 };
    },
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoints = [1, 2, 3, 4, 5].map((ep) => device.getEndpoint(ep));
        
        for (const ep of endpoints) {
            await reporting.bind(ep, coordinatorEndpoint, ['genOnOff']);
            await reporting.onOff(ep);
        }
        
        const powerEndpoint = device.getEndpoint(1);
        await reporting.bind(powerEndpoint, coordinatorEndpoint, ['haElectricalMeasurement', 'seMetering']);
        await reporting.current(powerEndpoint);
        await reporting.voltage(powerEndpoint);
        await reporting.power(powerEndpoint);
        await reporting.energy(powerEndpoint);
    },
};

module.exports = definition;`

After a restart (Zigbee2MQTT running in Docker), I got it working with all properties correctly exposed. ;)

JohnTitor177, 2025-04-29:
I seem to have an actual SM-0306E-2W and I get the same error.

z2m: Publish 'set' 'power_outage_memory' to 'DEVICE_NAME_HERE' failed: 'Error: ZCL command 0xa4c138a427dbad64/1 genOnOff.write({"moesStartUpOnOff":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'

Z2M version 2.2.1. I obscured the device name in the log.

fonso2, 2025-04-30:
@JohnTitor177 

Running Z2M 2.2.1 as well but own a Zemismart 4 outlet / 4 USB port version of this. Haven't tried pairing it to my network yet (Skyconnect + multiple Sonoff Dongle P routers).

In spite of the errors, Is the device working properly or are certain exposes disabled?

JohnTitor177, 2025-05-01:
@fonso2 

It seems to work the same way for me as it does for @detLAN in an earlier comment here.
It gives out an error when you make a change for the 'power_outage_memory' or 'child_lock', but it seems to still work as intended.

closed 2025-07-09 by github-actions[bot]

trenb, 2026-02-07:
I have the same issue as well. No chance of getting this fixed?
`z2m: Publish 'set' 'power_outage_memory' to 'Office - Power Bar' failed: 'Error: ZCL command 0xa4c13853e13XXXXX/1 genOnOff.write({"moesStartUpOnOff":2}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'`