Koenkk/zigbee2mqtt #22086: When Starting the addon Zigbeetomqtt under HAOS - Problem

SunsetGolfer, 2024-04-04:
### What happened?

[19:49:22] INFO: Preparing to start...
[19:49:22] INFO: Socat not enabled
[19:49:24] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:error 2024-04-04 19:49:31: Failed to load external converter file 'silabs.js' (Cannot find module 'zigbee-herdsman-converters/lib/extend'
Require stack:
- /app/dist/util/utils.js
- /app/dist/util/settings.js
- /app/index.js)
Zigbee2MQTT:error 2024-04-04 19:49:31: Probably there is a syntax error in the file or the external converter is not compatible with the current Zigbee2MQTT version
Zigbee2MQTT:error 2024-04-04 19:49:31: Note that external converters are not meant for long term usage, it's meant for local testing after which a pull request should be created to add out-of-the-box support for the device
Zigbee2MQTT:info  2024-04-04 19:49:31: Logging to console and directory: '/config/zigbee2mqtt/log/2024-04-04.19-49-30' filename: log.txt
Zigbee2MQTT:info  2024-04-04 19:49:31: Starting Zigbee2MQTT version 1.36.1 (commit #unknown)
Zigbee2MQTT:info  2024-04-04 19:49:31: Starting zigbee-herdsman (0.40.3)

### What did you expect to happen?

Work until I updated to latest version :-(

### How to reproduce it (minimal and precise)

Uninstall and reinstall did not help.

### Zigbee2MQTT version

1.36.1

### Adapter firmware version

Frontend not running

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus,Zigbee Gateway TI CC2652P + CP2102N 

### Setup

Docker

### Debug log

[19:49:22] INFO: Preparing to start...
[19:49:22] INFO: Socat not enabled
[19:49:24] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:error 2024-04-04 19:49:31: Failed to load external converter file 'silabs.js' (Cannot find module 'zigbee-herdsman-converters/lib/extend'
Require stack:
- /app/dist/util/utils.js
- /app/dist/util/settings.js
- /app/index.js)
Zigbee2MQTT:error 2024-04-04 19:49:31: Probably there is a syntax error in the file or the external converter is not compatible with the current Zigbee2MQTT version
Zigbee2MQTT:error 2024-04-04 19:49:31: Note that external converters are not meant for long term usage, it's meant for local testing after which a pull request should be created to add out-of-the-box support for the device
Zigbee2MQTT:info  2024-04-04 19:49:31: Logging to console and directory: '/config/zigbee2mqtt/log/2024-04-04.19-49-30' filename: log.txt
Zigbee2MQTT:info  2024-04-04 19:49:31: Starting Zigbee2MQTT version 1.36.1 (commit #unknown)
Zigbee2MQTT:info  2024-04-04 19:49:31: Starting zigbee-herdsman (0.40.3)

Chartreusito, 2024-04-05:
Please refer to release Notes : https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.36.1

SunsetGolfer, 2024-04-05:
Ok, but I have no "const extend = require('zigbee-herdsman-converters/lib/extend'); " in my config...

[image]

Chartreusito, 2024-04-05:
> Ok, but I have no "const extend = require('zigbee-herdsman-converters/lib/extend'); " in my config...
> 
> ![grafik](https://private-user-images.githubusercontent.com/11488056/319967990-ac77f097-191b-4481-8dd9-0305141a8b9c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTIzMTY0NjcsIm5iZiI6MTcxMjMxNjE2NywicGF0aCI6Ii8xMTQ4ODA1Ni8zMTk5Njc5OTAtYWM3N2YwOTctMTkxYi00NDgxLThkZDktMDMwNTE0MWE4YjljLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MDUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDA1VDExMjI0N1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTg4NDNmZTVjMjU3ZWNlODM1ZmY3MWVkZjU1ZGE3OWEzNGRkZWExYjQwYWIyOGFlYWM2YjcyMThkYzQ4OTcyMzUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.R8_DTPaN_1jnbwwILsvR7E7CGYpLr-tk40AngMV86OU)

It says you have to remove this line from your external converter file, not from your config

SunsetGolfer, 2024-04-05:
Thanks, I have a very old silabs.js - but it isn´t referenced anywhere...

[image]

lux73, 2024-04-06:
then why not store `silabs.js` anywhere & delete it from your z2m root Folder?

but is has to referenced anywhere - i doubt there is a parser which load all available *.js Files from root Directory automatic ^^

closed 2024-10-04 by SunsetGolfer