Koenkk/zigbee2mqtt #22092: 1.36.1 All Jasco Switches and Dimmers go Offline

majorsl, 2024-04-05:
### What happened?

Upgraded to 1.36.1 and all Jasco brand switches go offline but can be toggled on and off under Exposes. When done, they appear on-line again until sometime later when it happens again.

Reverting to 1.36.0 returns everything to normal again.

### What did you expect to happen?

My switches to be available as they have been for years until 1.36.1.

### How to reproduce it (minimal and precise)

Install 1.36.1 and observe.

### Zigbee2MQTT version

1.36.1

### Adapter firmware version

20230507

### Adapter

Sonoff Dongle-P

### Setup

X86 Linux host running Z2M in Docker

bill7531, 2024-04-05:
I can confirm I am experiencing the same issue with my Jasco dimmers. Though the first generation switch (45856GE) seemed to be fine.

rehanson, 2024-04-07:
This same issue is happening to me. I have two Jasco 43080 dimmers with the issue. It started happening after upgrading to 1.36.1

majorsl, 2024-04-07:
> This same issue is happening to me. I have two Jasco 43080 dimmers with the issue. It started happening after upgrading to 1.36.1

I have those too. I updated my issue with all my models, I have 4 varieties of the Jasco wall switches, all don't work with 1.36.1.

majorsl, 2024-04-07:
This appears to be fixed for the next release in [https://github.com/Koenkk/zigbee-herdsman/pull/1004](url)

quadcom, 2024-04-07:
When might this get pushed out? I have been living with a broken light automation for over a week because of this. The WAF is getting eroded slowly!

peterskajr, 2024-04-09:
I have the same issue with models 43076, 43082, 43084, 43090, 43096.  My only workaround so far is to have an automation that detects one of these devices as unavailable then restarts the Z2M addon.  Some of my Jasco products, mainly an outdoor plug (model 43100) don't seem to be affected.

closed 2024-11-06 by github-actions[bot]